FedeDP/Clight #4: Systemd service does not start (Ubuntu 17.04)

DonHugo69, 2017-04-18:
I enabled the service via `systemctl --user enable clight.service` however after reboot, `systemctl --user status clight.service` results in

``
clight.service - C daemon to adjust screen brightness to match ambient brightn
   Loaded: loaded (/usr/lib/systemd/user/clight.service; enabled; vendor preset:
   Active: failed (Result: exit-code) since Tue 2017-04-18 11:23:07 CEST; 13min 

Apr 18 11:23:07 Surface-Pro-3 systemd[1623]: Starting C daemon to adjust screen 
Apr 18 11:23:07 Surface-Pro-3 systemd[1623]: clight.service: Control process exi
Apr 18 11:23:07 Surface-Pro-3 systemd[1623]: Failed to start C daemon to adjust 
Apr 18 11:23:07 Surface-Pro-3 systemd[1623]: clight.service: Unit entered failed
Apr 18 11:23:07 Surface-Pro-3 systemd[1623]: clight.service: Failed with result 
Apr 18 11:23:08 Surface-Pro-3 systemd[1623]: [/usr/lib/systemd/user/clight.servi
Apr 18 11:23:08 Surface-Pro-3 systemd[1623]: [/usr/lib/systemd/user/clight.servi
``
Starting the service manually via `systemctl --user start clight.service` turns clight on and everything seems to work.

FedeDP, 2017-04-18:
Hi! Thanks for reporting this issue.
Can you paste me the output of: journalctl --user --unit clight? 
Thanks!

DonHugo69, 2017-04-18:
`journalctl --user --unit clight` just returns
```
No journal files were found.
-- No entries --
```
I use Unity so it should be LightDM.

FedeDP, 2017-04-18:
What does  
env | grep XDG_SESSION_TYPE
tells you?
Can you paste me the output of "killall clight && clight" too?

DonHugo69, 2017-04-18:
` env | grep XDG_SESSION_TYPE`
```
XDG_SESSION_TYPE=x11
```
`killall clight && clight`
```
clight: Kein Prozess gefunden (engl: No process found)
```
After using `systemctl --user start clight.service` & `killall clight && clight`
```
Config file /home/roland/.config/clight.conf not found.
Bus support started.
Brightness module started.
Location module started.
Gamma module started.
Signal module started.
Dpms module started.
New location received: 47.62, 7.67
Average frames brightness: 0.406966.
New brightness value: 381
Gamma temp was already 6500
Next gamma alarm due to: Tue Apr 18 19:49:00 2017
```

FedeDP, 2017-04-18:
So, clight starts without errors. Can you paste me the output of 
systemctl --user --all status clight
?

DonHugo69, 2017-04-18:
```
clight.service - C daemon to adjust screen brightness to match ambient brightness, as computed capturing frames from webcam
   Loaded: loaded (/usr/lib/systemd/user/clight.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2017-04-18 18:41:07 CEST; 56s ago

Apr 18 18:41:07 Surface-Pro-3 systemd[1619]: Starting C daemon to adjust screen brightness to match ambient brightness, as computed capturing f
Apr 18 18:41:07 Surface-Pro-3 systemd[1619]: clight.service: Control process exited, code=exited status=1
Apr 18 18:41:07 Surface-Pro-3 systemd[1619]: Failed to start C daemon to adjust screen brightness to match ambient brightness, as computed capt
Apr 18 18:41:07 Surface-Pro-3 systemd[1619]: clight.service: Unit entered failed state.
Apr 18 18:41:07 Surface-Pro-3 systemd[1619]: clight.service: Failed with result 'exit-code'.
Apr 18 18:41:07 Surface-Pro-3 systemd[1619]: [/usr/lib/systemd/user/clight.service:8] Unknown lvalue 'Wants' in section 'Service'
Apr 18 18:41:07 Surface-Pro-3 systemd[1619]: [/usr/lib/systemd/user/clight.service:9] Unknown lvalue 'After' in section 'Service'
```

It seems that clight cannot start the clightd service on startup compared to running it in the terminal after the desktop has loaded.

FedeDP, 2017-04-18:
Thanks you spotted a stupid issue!
It should be now fixed, can you re-download deb packages for clight and clightd and try again?

DonHugo69, 2017-04-18:
Unfortunately the problem is still present. After login the service fails to be active. 
What has changed is that after running `systemctl --user start clight.service` the brightness gets adjusted but the service stops.
```
systemctl --user --all status clight
● clight.service - C daemon to adjust screen brightness to match ambient brightness, as computed capturing frames from webcam
   Loaded: loaded (/usr/lib/systemd/user/clight.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Tue 2017-04-18 22:42:49 CEST; 5min ago
  Process: 3593 ExecStart=/usr/bin/clight (code=exited, status=0/SUCCESS)
  Process: 3590 ExecStartPre=/usr/bin/xhost + (code=exited, status=0/SUCCESS)
 Main PID: 3593 (code=exited, status=0/SUCCESS)

Apr 18 22:42:47 Surface-Pro-3 systemd[1621]: Starting C daemon to adjust screen brightness to match ambient brightness, as computed capturing f
Apr 18 22:42:47 Surface-Pro-3 systemd[1621]: Started C daemon to adjust screen brightness to match ambient brightness, as computed capturing fr
Apr 18 22:42:47 Surface-Pro-3 clight[3593]: Config file /home/roland/.config/clight.conf not found.
```
Here is content of the logfile:
```
Clight
Version: 0.9
Time: Tue Apr 18 22:42:47 2017

Starting options:
* Number of captures: 5
* Daily timeout: 600
* Nightly timeout: 2700
* Webcam device: 
* Backlight path: 
* Daily screen temp: 6500
* Nightly screen temp: 4000
* Smooth transitions: enabled
* Latitude: 0.00
* Longitude: 0.00
* User setted sunrise: 
* User setted sunset: 
* Gamma correction: enabled

(I) Bus support started.
(I) Brightness module started.
(I) Gamma module started.
(I) Signal module started.
(I) Dpms module started.
(I) Average frames brightness: 0.339358.
(I) New brightness value: 317
(E) Wrong sunrise/sunset time setted by a cmdline arg. Leaving.
(E) Wrong sunrise/sunset time setted by a cmdline arg. Leaving.
(E) Failed to retrieve sunrise/sunset informations.
(I) Brightness module destroyed.
(I) Gamma module destroyed.
(I) Signal module destroyed.
(I) Dpms module destroyed.
(I) Bus destroyed.
```

FedeDP, 2017-04-18:
Because i am stupid and i fucked up a small thing. You can retry now (only download clight deb file!)

DonHugo69, 2017-04-18:
But still you offer great support ;)
I am starting to feel quite bad but the service has still the same issue after login.
But manually starting it via `systemctl --now --user start clight.service` works again:
```
systemctl --user --all status clight
● clight.service - C daemon to adjust screen brightness to match ambient brightness, as computed capturing frames from webcam
   Loaded: loaded (/usr/lib/systemd/user/clight.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2017-04-18 23:01:14 CEST; 3s ago
  Process: 2872 ExecStartPre=/usr/bin/xhost + (code=exited, status=0/SUCCESS)
 Main PID: 2875 (clight)
   CGroup: /user.slice/user-1000.slice/user@1000.service/clight.service
           └─2875 /usr/bin/clight

Apr 18 23:01:14 Surface-Pro-3 systemd[1659]: Starting C daemon to adjust screen brightness to match ambient brightness, as computed capturing f
Apr 18 23:01:14 Surface-Pro-3 systemd[1659]: Started C daemon to adjust screen brightness to match ambient brightness, as computed capturing fr
Apr 18 23:01:14 Surface-Pro-3 clight[2875]: Config file /home/roland/.config/clight.conf not found.
```
Strange is however that the location does not get fetched from geoclue as before but that is probably unrelated.

FedeDP, 2017-04-18:
Have you rebooted since updating today (after my commit from 4-5hrs ago)?

DonHugo69, 2017-04-18:
Yes, every time.

FedeDP, 2017-04-18:
That is really really weird. 
Can you post again your .clight.log and "sudo journalctl -u clightd" output right after a reboot?
This way i can see what is the issue in both clight and clightd, and hope to come up with a solution!:)

DonHugo69, 2017-04-18:
`sudo journatctl -u clightd`
```
-- No entries --
```
The log file contains only entries from the last session I invoked manually (and stopped manually):
```
Clight
Version: 0.9
Time: Tue Apr 18 23:25:26 2017

Starting options:
* Number of captures: 5
* Daily timeout: 600
* Nightly timeout: 2700
* Webcam device: 
* Backlight path: 
* Daily screen temp: 6500
* Nightly screen temp: 4000
* Smooth transitions: enabled
* Latitude: 0.00
* Longitude: 0.00
* User setted sunrise: 
* User setted sunset: 
* Gamma correction: enabled

(I) Bus support started.
(I) Brightness module started.
(I) Gamma module started.
(I) Signal module started.
(I) Dpms module started.
(I) Average frames brightness: 0.328940.
(I) New brightness value: 308
(I) received signal 15. Leaving.
(I) Brightness module destroyed.
(I) Gamma module destroyed.
(I) Signal module destroyed.
(I) Dpms module destroyed.
(I) Bus destroyed.
```
If it helps you, I can try it tomorrow on a machine which runs Manjaro but has no webcam. Maybe Ubuntu does something totally different in terms of systemd services than Arch ?

DonHugo69, 2017-04-18:
A small addition, maybe its helpful: I just added clight to the startup programs and it gets activated after login (webcam LED lights briefly up and brightness changes).
Logs:
```
systemctl --user --all status clight
● clight.service - C daemon to adjust screen brightness to match ambient brightness, as computed capturing frames from webcam
   Loaded: loaded (/usr/lib/systemd/user/clight.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2017-04-18 23:42:12 CEST; 4min 13s ago

Apr 18 23:42:12 Surface-Pro-3 systemd[1642]: Starting C daemon to adjust screen brightness to match ambient brightness, as computed capturing f
Apr 18 23:42:12 Surface-Pro-3 systemd[1642]: clight.service: Control process exited, code=exited status=1
Apr 18 23:42:12 Surface-Pro-3 systemd[1642]: Failed to start C daemon to adjust screen brightness to match ambient brightness, as computed capt
Apr 18 23:42:12 Surface-Pro-3 systemd[1642]: clight.service: Unit entered failed state.
Apr 18 23:42:12 Surface-Pro-3 systemd[1642]: clight.service: Failed with result 'exit-code'.
```
```
sudo journalctl -u clightd
-- Logs begin at Tue 2017-04-18 23:41:40 CEST, end at Tue 2017-04-18 23:46:29 CEST. --
Apr 18 23:42:15 Surface-Pro-3 systemd[1]: Starting Bus service to change screen brightness...
Apr 18 23:42:15 Surface-Pro-3 systemd[1]: Started Bus service to change screen brightness.
```
```
Clight
Version: 0.9
Time: Tue Apr 18 23:42:15 2017

Starting options:
* Number of captures: 5
* Daily timeout: 600
* Nightly timeout: 2700
* Webcam device: 
* Backlight path: 
* Daily screen temp: 6500
* Nightly screen temp: 4000
* Smooth transitions: enabled
* Latitude: 0.00
* Longitude: 0.00
* User setted sunrise: 
* User setted sunset: 
* Gamma correction: enabled

(I) Bus support started.
(I) Brightness module started.
(I) Gamma module started.
(I) Signal module started.
(I) Dpms module started.
(I) Average frames brightness: 0.098664.
(I) New brightness value: 92
```

FedeDP, 2017-04-19:
As you said, it seems to me that clight does not per started.
What's the output for:
systemctl --user is-enabled clight
And
systemctl --all status display-manager
?
Two things possible here: 
1) clight unit is not enabled (well i don't think so)
2) clight unit has to be started right After display-manager service. If display-manager does not get started, clight does not get started too.

DonHugo69, 2017-04-19:
`systemctl --user is-enabled clight` returns `enabled` and 
```
systemctl --all status display-manager
● lightdm.service - Light Display Manager
   Loaded: loaded (/lib/systemd/system/lightdm.service; enabled; vendor preset: enabled)
  Drop-In: /lib/systemd/system/display-manager.service.d
           └─xdiagnose.conf
   Active: active (running) since Wed 2017-04-19 07:36:23 CEST; 3min 11s ago
     Docs: man:lightdm(1)
 Main PID: 1207 (lightdm)
    Tasks: 8 (limit: 4915)
   CGroup: /system.slice/lightdm.service
           ├─1207 /usr/sbin/lightdm
           ├─1220 /usr/lib/xorg/Xorg -core :0 -seat seat0 -auth /var/run/lightdm/root/:0 -nolisten tcp vt7 -novtswitch
           └─1639 lightdm --session-child 12 19

Apr 19 07:36:24 Surface-Pro-3 lightdm[1414]: PAM unable to dlopen(pam_kwallet5.so): /lib/security/pam_kwallet5.so: cannot open shared object fi
Apr 19 07:36:24 Surface-Pro-3 lightdm[1414]: PAM adding faulty module: pam_kwallet5.so
Apr 19 07:36:24 Surface-Pro-3 lightdm[1414]: pam_unix(lightdm-greeter:session): session opened for user lightdm by (uid=0)
Apr 19 07:36:25 Surface-Pro-3 lightdm[1584]: PAM unable to dlopen(pam_kwallet.so): /lib/security/pam_kwallet.so: cannot open shared object file
Apr 19 07:36:25 Surface-Pro-3 lightdm[1584]: PAM adding faulty module: pam_kwallet.so
Apr 19 07:36:25 Surface-Pro-3 lightdm[1584]: PAM unable to dlopen(pam_kwallet5.so): /lib/security/pam_kwallet5.so: cannot open shared object fi
Apr 19 07:36:25 Surface-Pro-3 lightdm[1584]: PAM adding faulty module: pam_kwallet5.so
Apr 19 07:36:25 Surface-Pro-3 lightdm[1584]: pam_succeed_if(lightdm:auth): requirement "user ingroup nopasswdlogin" not met by user "roland"
Apr 19 07:36:36 Surface-Pro-3 lightdm[1638]: pam_ecryptfs: Passphrase file wrapped
Apr 19 07:36:36 Surface-Pro-3 lightdm[1584]: pam_unix(lightdm:session): session opened for user roland by (uid=0)
```

FedeDP, 2017-04-19:
I don't understand how is it possible that clight.service does not get even started...
A workaround is to add "systemctl --user start clight.service" command to your session startup manager (ie: in autostart in unity settings).  
This is really weird though. 
One more thing you can try is to add a "Restart=always" in Service section of clight.service (edit /usr/lib/systemd/user/clight.service) and try to reboot...Let me know if it works, i'll add that line in clight service (or if you wish to contribute with a PR, you're welcome!)
Btw i also see that geoclue fails to retrieve your location (ie: location is not received from geoclue), that is weird too but it happened sometimes here too. It seems an issue with geoclue as far as i can tell though.
Can you paste me the output of systemctl status geoclue too?

DonHugo69, 2017-04-19:
`Restart=always` did the trick, now clight is up and running after logging in!
I quickly googled around and found https://github.com/google/cloud-print-connector/issues/140 which is kinda similar. Maybe one needs to add something more to `After=...`. However, I tried adding `graphical.target` but this did not help. 
I tried to create a pull request but I did this for the first time, so I don't know if it was done correctly...
Regarding the geoclue issue:

FedeDP, 2017-04-19:
I don't really know. From your logs it seemed clight did not get started (so it did not fail or whatever); i don't understand why adding Restart=always does the trick though. It seems like it fails to start clight immediately (and, as there is no .clight.log, i assume ExecStartPre is failing for some reasons...).

By the way, thanks! PR is fine! Do you mind to switch to "Restart=on-failure" and add "RestartSec=5" too? This way we will not continuosly restart clight if for any reason it fails to start/crashes, but only every 5 seconds!

About geoclue, i'm pretty sure it will work agin once i push!
Thank you very much for your time!

PS: about "After=": it is kinda hard because clight (with gamma support enabled) needs an X environment. But in systemd there is no way to know if X is started. So i binded my service unit to display-manager. 
While systemd user service is quite convenient because it self-manages dependencies (clightd and xhost call), it is quite easy to have a non-working clight.
For this reason, i'm thinking about switching to dear old /etc/xdg/autostart scripts instead of systemd user service.

DonHugo69, 2017-04-19:
I don't have a deep knowledge in that regard, I am just an enthusiastic linux user. But isn't that the meaning of `After=graphical.target`? I read that this corresponds to Init 5, which is a grapical X session? 
But I think having other users with other DEs like GNOME or KDE commenting here would be more helpful. Probably it is more related to Unity or LightDM and the way it handles the startup process.

FedeDP, 2017-04-19:
It seems graphical.target can be perhaps better (i already tried that, but then i noticed redshift was using display-manager.service).
Can you try without "Restart=" and using "After=graphical.target"? Just change clight.service, and issue "make deb", then install the newly created Debian file inside Debian folder.

If it fixes the original issue, add back "Restart=on-failure" and "RestartSec" and update the PR thanks!
Just remember to remove Debian folder before updating the PR :)
BTW thank you very much!

The fact is that clight does not strictly *need* an X environment; it can be started from wayland or from a tty too, it will just auto-disable gamma where it is not supported.
I am trying to figure out if it's not the case to move gamma support directly into clight (now it is offered by clightd), as it needs to interact with X session (thus the call to xhost).

DonHugo69, 2017-04-19:
Unfortunately, setting `After=graphical.target` does not solve the issue. Adding `Restart=on-failure` and `RestartSec=5` leads to the expected behavior and the service (re)starts after 5 seconds.

FedeDP, 2017-04-19:
Thanks! As i thought it makes no difference :( 
I'll leave this issue open, hopefully some other users will provide some more feedback/ideas about how to properly solve.
In the meantime, expect a commit later with some fixes and an updated deb! 
Again let me thank you for your support!

FedeDP, 2017-04-19:
@DonHugo69  In latest commit i also committed some changes to clight.service. Let me know if you encounter any issue! (deb pkg is updated!)

DonHugo69, 2017-04-19:
Haha, thank you, you're making me blush :)
Just tried the new .deb and voila, everything up and running! 
Geoclue is still making some trouble but this is not related to your program. It is most likely the shaky network connection. I restarted your service several times and watched journalctl -u geoclue.service and got
```
Apr 19 20:54:00 Surface-Pro-3 geoclue[3104]: Failed to query location: Fehler beim Auflösen von »location.services.mozilla.com«: Der Name oder 
```
However after a little bit of browsing and one more restart, my screen has a become reddish.
Also clight.log now shows:
```
Clight
Version: 0.9
Time: Wed Apr 19 20:56:02 2017

Starting options:
* Number of captures: 5
* Daily timeout: 600
* Nightly timeout: 2700
* Webcam device: 
* Backlight path: 
* Daily screen temp: 6500
* Nightly screen temp: 4000
* Smooth transitions: enabled
* Latitude: 0.00
* Longitude: 0.00
* User setted sunrise: 
* User setted sunset: 
* Gamma correction: enabled

(I) Bus support started.
(I) Brightness module started.
(I) Location module started.
(I) Gamma module started.
(I) Signal module started.
(I) Dpms module started.
(I) New location received: 47.62, 7.67
(I) Average frames brightness: 0.433259.
(I) Old brightness value: 407
(I) New brightness value: 405
(I) 4000 gamma temp setted.
(I) Next gamma alarm due to: Thu Apr 20 06:05:00 2017
```
Thus, everything seems to work! Thank you very much for writing this program and your great support!!

FedeDP, 2017-04-19:
Thank you for being such a good and enthusiast tester!:)
Happy you're finally enjoying it!

Yes, geoclue is far from perfect...

FedeDP, 2017-04-20:
Updated readme now clearly states that recommended way of starting clight is by adding:
`systemctl --user start clight.service `
in DE startup configuration.
This should fix every issue :)

commit 323cdb77 (2017-04-18): Updated deb pkg with latest fixes. Fixed stupid issue in clight.service that fixes #4
closed 2017-04-20 by FedeDP