FedeDP/Clight #5: Error when running clight

emansih, 2017-04-18:
Hi, I ran into an error while running clight. The daemon does not start due to an unknown error. 
$ systemctl --user start clight.service

$ systemctl --user status clight.service
● clight.service - C daemon to adjust screen brightness to match ambient brightness, as computed capturing frames from webcam
   Loaded: loaded (/usr/lib/systemd/user/clight.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Tue 2017-04-18 19:48:09 +08; 1s ago
  Process: 3925 ExecStart=/usr/bin/clight (code=exited, status=0/SUCCESS)
  Process: 3922 ExecStartPre=/usr/bin/xhost + (code=exited, status=0/SUCCESS)
 Main PID: 3925 (code=exited, status=0/SUCCESS)

$ clight
Failed to open lock.

It seems that the error message is not verbose enough. The below is my config 

```
## Number of frames to be captured
frames = 7;

## Timeout between captures during the day
day_timeout = 100;

## Timeout between captures during the night
night_timeout = 500;

## Gamma daily temperature
day_temp = 6500;

## Gamma nightly temperature
night_temp = 4000;

## Uncomment to disable gamma smooth transitions
# no_smooth_gamma_transition = 1;

## Uncomment to disable gamma support
# no_gamma = 1;

## Your desired latitude for gamma support (surise/sunset in this location)
latitude = **<removed>**;

## Your desired longitude for gamma support (surise/sunset in this location)
longitude = **<removed>**;

## Video device to be used
video_devname = "/dev/video0";

## Screen syspath to be used
screen_sysname = "/sys/class/backlight/intel_backlight";

## Force set a sunrise time
# sunrise = "7:00";

## Force set a sunset time
# sunset = "19:00";
```

FedeDP, 2017-04-18:
It seems to be my fault: video_devname and screen_sysname must only be last part of path (eg: video0 and intel_backlight in your case).
I'll add this information to the comments in config file asap! (i can accept a PR if you're willing to!)
Moreover,
> $ clight
> Failed to open lock.
Means that a clight instance is already running. Try a "killall clight" and re run it from cmdline to check its output (fix video_devname and screen_sysname before!)

emansih, 2017-04-18:
$ killall clight 

that doesn't work since clight isn't running in the first place. 

$ ps aux | grep clight
daniel   10528  0.0  0.0  12784   948 pts/0    S+   20:21   0:00 grep clight

FedeDP, 2017-04-18:
Remove ".clight.lock" file in your home folder and re-run clight. It seems weird to me that the lock file has not been removed from a previous run though...

emansih, 2017-04-18:
nice. removing .clight.lock works!

FedeDP, 2017-04-18:
Let me know if this issue returns :)

FedeDP, 2018-02-03:
That surely is a good idea :)
This is gonna be included in next release (i've got no eta but it will be quite soon, may be a couple of weeks if i managed to get the time to workaround some small issues).
Thanks for your feedback @Technohacker !

Technohacker, 2018-02-03:
Glad to help :D

FedeDP, 2018-03-03:
Fixed in 4f8efac.

closed 2018-03-03 by FedeDP