FedeDP/Clight #7: Disabling gamma support as X is not running.

DonHugo69, 2017-04-20:
There seems to be a glitch when clight is trying to get the location. After startup, my clight.log looks like this:
```
(I) Disabling gamma support as X is not running.
Clight
Version: 0.9
Time: Thu Apr 20 07:24:08 2017

Starting options:
* Number of captures: 5
* Daily timeout: 600
* Nightly timeout: 2700
* Webcam device: 
* Backlight path: 
* Daily screen temp: 6500
* Nightly screen temp: 4000
* Smooth transitions: enabled
* Latitude: 0.00
* Longitude: 0.00
* User setted sunrise: 
* User setted sunset: 
* Gamma correction: disabled

(I) Bus support started.
(I) Brightness module started.
(I) Signal module started.
(I) Average frames brightness: 0.118881.
(I) Old brightness value: 259
(I) New brightness value: 111
```
Thus the gamma correction/location module is disabled.
After running `systemctl --user restart clight.service` it changes to:
```
Clight
Version: 0.9
Time: Thu Apr 20 07:53:40 2017

Starting options:
* Number of captures: 5
* Daily timeout: 600
* Nightly timeout: 2700
* Webcam device: 
* Backlight path: 
* Daily screen temp: 6500
* Nightly screen temp: 4000
* Smooth transitions: enabled
* Latitude: 0.00
* Longitude: 0.00
* User setted sunrise: 
* User setted sunset: 
* Gamma correction: enabled

(I) Bus support started.
(I) Brightness module started.
(I) Location module started.
(I) Gamma module started.
(I) Signal module started.
(I) Dpms module started.
(I) New location received: 47.62, 7.67
(I) Average frames brightness: 0.369102.
(I) Old brightness value: 112
(I) New brightness value: 345
(I) Gamma temp was already 6500
(I) Next gamma alarm due to: Thu Apr 20 19:52:00 2017
```

FedeDP, 2017-04-20:
These are the changes i committed yesterday evening.  
Well, at least now we found the root of your issues: it seems that when clight starts, you're not on X (or, well, X is not loaded?).  
This is really weird though; i only see a reason for it: you're either running gdm as your DM (that nowadays runs on wayland), or lightdm runs on wayland too (i cannot find any source for this though).
This would explain the issue: as far as you login, you're still on wayland when clight starts; after you manually restart clight.service, from your X session, it does not fail anymore.
If this is the case, the best and simplest thing to do is to disable user service, and add
`systemctl --user start clight.service`
to your autostart scripts/DE autostart configuration.

FedeDP, 2017-04-20:
Another thing you can do is to delay init of clight user service with a timer service: https://wiki.archlinux.org/index.php/Systemd/Timers, using probably 
`[Unit]
Description=clight.service
[Timer]
RemainAfterElapse=no
[Timer]
OnUnitActiveSec=5s`
should fix your issue (enable the timer and disable your clight.service).

DonHugo69, 2017-04-20:
And fixed, thanks a lot again! I decided to take the autostart route directly as I also don't like the idea of an artificial service delay. 
Probably Cannonical activated Mir for LightDM at some point like GNOME for GDM. But I can't find a definitive statement about that. 
Probably its best if you update your readme and install instructions that Ubuntu (or maybe just Unity)  user should use it via autostart and ask for user reports if that happens in other DEs as well.

FedeDP, 2017-04-20:
Yes! Indeed i am really thinking about switching to that (ie manually add service start to session autostart config) instead of enabling the service. It causes much less troubles.
BTW for now i'll add that to readme (are you willing to? :D otherwise i'll do that later in the afternoon!)
Thanks for reporting this issue!

DonHugo69, 2017-04-20:
Haha, I can't make promises here, my recovery from the flue is over and I have to work again (unfortunately).

FedeDP, 2017-04-20:
No problem, i know those struggles :D

closed 2017-04-20 by FedeDP

FedeDP, 2017-04-20:
@DonHugo69 I updated both clightd and clight with some improvements (most notably, dropped xhost call as it was a small security hole).
Moreover, it is now stated clearly in readme that recommended way to start clight is through:
`systemctl --user start clight.service`
in DE startup configuration.
Again, let me know if you encounter any issue :)

DonHugo69, 2017-04-20:
Sorry for the boring news, just updated the debs and all seems to work well! 
Thank you very much!