FedeDP/Clight #8: Clight is failing to start on Ubuntu 16.04

alekcacko, 2017-04-22:
Steps to reproduce:

1. Enable clightd systemctl service:
    $ sudo systemctl enable clightd.service
    $ sudo systemctl start clightd.service

2. Try to start clight from command line:
    $ clight
    Config file /home/mate/.config/clight.conf not found.
    Bus support started.
    Message recipient disconnected from message bus without   replying
    Bus destroyed.

3. Or try to start with systemctl
    $ systemctl --user start clight.service

So clightd service just doesn't want to start

Logs:

Systemctl: https://pastebin.com/X5dKjzSS
journalctl: https://pastebin.com/5Y6vHt1p

FedeDP, 2017-04-22:
Can you paste me the output of  
`coredumpctl  dump` 
(if it fails to find clightd dump, please run it as root)

alekcacko, 2017-04-22:
Here is the log from coredumpctl dump I managed to obtain:
https://pastebin.com/rFYN87wh

FedeDP, 2017-04-22:
Wow it seems you haven't got a maxbrightness attribute for your backlight interface...i didn't think this could happen.
Can you paste me output of:
`ls /sys/class/backlight/* | grep max`
Thank you!

alekcacko, 2017-04-22:
The output of the command:

     $ ls /sys/class/backlight/* | grep max
    ls: cannot access '/sys/class/backlight/*': No such file or directory

FedeDP, 2017-04-22:
Uhm, what about:
`ls /sys/class/backlight/*`?

alekcacko, 2017-04-22:
EDIT:

All I forgot to tell you is that I am trying to run this on a PC, not laptop.

And as far as I know, we can't change brigthness on PC monitor right?

FedeDP, 2017-04-22:
It can probably be issue's cause.
Btw i don't know if you can change brightness on a pc...do you have any interface in /sys/class/backlight/?

alekcacko, 2017-04-22:
Well, no. the backlight folder is empy.

FedeDP, 2017-04-22:
Then you're out of luck :(

alekcacko, 2017-04-22:
Oh it's okay, since the monitor I am using doesn't have hardware brigthness.
The issue can be closed.

Thank you for your great support though.

closed 2017-04-22 by alekcacko

FedeDP, 2017-04-22:
You're welcome! I hope you can try it out succesfully in some laptop :)