FedeDP/Clight #11: clight hangs on no network connection

runical, 2017-05-13:
As said on the Arch boards, I found a way to test (a sock over the webcam, works great :P )

Anyway, when running clight without a network connection and no defined location, it hangs on I assume getting the location. You might want to check for a network connection and maybe define a backup location in the config.
[clight.txt](https://github.com/FedeDP/Clight/files/998508/clight.txt)

FedeDP, 2017-05-13:
Yes, i am aware of this issue.
If you check TODO, that is on my todo list for next release(caching location).
It is gonna be fixed soon:)
Thanks for your report!

FedeDP, 2017-05-13:
Fixed in  f4e278e!

closed 2017-05-13 by FedeDP

FedeDP, 2017-05-13:
EDIT: obviously if at first start ever of clight (ie when no cache file is present), clight is run without lat/lon (so, with geoclue enabled) and no internet connection, it will hang still. Sadly, that issue cannot be prevented.

runical, 2017-05-13:
Ah, my bad. I completely missed the TODO! Thanks for fixing it!