FedeDP/Clight #15: Location don't switch from hard dep to soft dep after setting manually.

tx1683, 2017-08-27:
When I set location manually in config file, or in command line (or both) location module is disabled, which is expected, but after that gamma module is also disabled because the lack of location module. 
In src/location.c there is a function that should switch location from hard dep to soft dep, but it doesn't working for some reason.

> Trying to start Gamma module as its Xorg dependency was loaded...
Trying to start GammaSmooth module as its Xorg dependency was loaded...
Trying to start Dimmer module as its Xorg dependency was loaded...
Trying to start DimmerSmooth module as its Xorg dependency was loaded...
Trying to start Dpms module as its Xorg dependency was loaded...
Trying to start Inhibit module as its Xorg dependency was loaded...
Trying to start Brightness module as its Bus dependency was loaded...
Trying to start Location module as its Bus dependency was loaded...
Location module disabled.
Disabling module Gamma as its hard dep Location was disabled...
Gamma module disabled.
Trying to start Brightness module as its Gamma soft dep was disabled...
Disabling module GammaSmooth as its hard dep Gamma was disabled...
GammaSmooth module disabled.
GammaSmooth module destroyed.
Gamma module destroyed.
Location module destroyed.

When I remove manual location setting the geoclue2 kicks in and everything is working as expected.

System: Arch Linux
Clight installed from AUR.
I would gladly provide more info when necessary.

FedeDP, 2017-08-27:
I can reproduce the problem, thank you.
I'll fix it asap :)

closed 2017-08-27 by FedeDP

FedeDP, 2017-08-27:
That was quite a stupid bug indeed. Thanks for your report!
It is now fixed in master.

commit 1becdabb (2017-08-27): Fixed #15: update dependent_m too when changing a dep type.