FedeDP/Clight #19: Allow light correction with configfile

asm0dey, 2018-01-01:
For me default clight settings are to bright. I would prefer to have way to correct settings.
i.e. right now clights sets brightness 3600 for me, but I would prefer 2200.

FedeDP, 2018-01-01:
Hi, thanks for your report. You can actually change backlight-to-ambient-brightness function to better suit your preference, have a look at config file in /etc/default/clight.conf: https://github.com/FedeDP/Clight/blob/master/Extra/clight.conf#L71.
As you can see, you can even set 2 different curves for AC state.

asm0dey, 2018-01-01:
Thank you! It can serve even better than "hard" adjustment!

closed 2018-01-01 by asm0dey

ModDM, 2019-08-14:
If you make a line chart of it in your spreadsheet program, you can display the curve graphically and play with the numbers and see the curve changes:

[image]

ModDM, 2019-08-14:
Question: Is it also allowed to add curve points?

FedeDP, 2019-08-14:
Right now it is not possible.
But there is no theoretical limit in code; it can be easily done. 

Can you open another issue/feature request?
Maybe explain why you'd think it could improve your experience!

ModDM, 2019-08-14:
Sure, after testing some days / afternoons / evenings etc. I will have more experience with the curve and then I am better able to describe why / which way etc.