FedeDP/Clight #20: Add ability for smooth backlight change

asm0dey, 2018-01-01:
I would prefer macbooklike style of brightness correction. e.g change from brightness 2000 to brightness 3000 in 5 seconds

FedeDP, 2018-01-01:
It would be nice (and it already exists for dimmer) yeah, but given that lots of people only have something like 10 backlight steps, and that usually backlight is lowered/raised up of very little, it would be useless in most cases.
It exists for dimmer where usually backlight is dimmed much more.

asm0dey, 2018-01-01:
I can only say about my laptop:
```
~/ cat /sys/class/backlight/intel_backlight/max_brightness
7500
```
I would propose to add optional parameter if that's possible. Also, it may be reasonable to look if there is more then 10 steps to change and if yes — then change backlight level smoothly.

asm0dey, 2018-01-01:
Also, it may be reasonable to change, say, 1 step per second. guys with 15 steps will have immediate results, guys with lots of steps will have a smooth adaptation of screen.

FedeDP, 2018-01-02:
I'll look into this :)
I was working on another thing in clight that right now has top priority (and as soon as i got time i'll finish this): support for desktop PCs (and external HDMI monitor) through ddcutil.
It is quite there (devel branch) but it still needs some fixes in clightd.
After that feature is in place, i'll start looking into this.

Btw any patch is really welcome :)

FedeDP, 2018-01-20:
@asm0dey On devel branch you can find backlight smooth transitions.
If you're willing to test it (and clight too for next release), you can build devel branch.
Remember that you also need clightd from "ddcutil" branch ;)

Let me know if you encounter any issue!

asm0dey, 2018-01-20:
@FedeDP will clightd from master work for me?

asm0dey, 2018-01-20:
I've got following error: Unknown property or interface without any clue of what property it doen't like

FedeDP, 2018-01-20:
You should use clightd built from ddcutil branch! 
Once some tests is done i'll merge both branch in master(for clight and clightd)

asm0dey, 2018-01-20:
@FedeDP I've installed it exactly that way.  But getting error `Unknown property or interface`

FedeDP, 2018-01-20:
This is weird.
Mind to share your /usr/share/polkit-1/actions/org.clightd.backlight.policy?
Moreover, this seems like a clight/clightd versions mismatch.
Are you really sure you're running correct versions?
Mind to share how you are building and starting both clight and clightd?
Thanks!

FedeDP, 2018-03-03:
Fixed in https://github.com/FedeDP/Clight/commit/4f8efac147dec513bde4f61379b95bc537723ad3.

closed 2018-03-03 by FedeDP