FedeDP/Clight #22: Gamma correction doesn't work on arch liunx

flovo, 2018-02-13:
Hi,
I've installed clight-git arch Linux from aur. 
Clight changes backlight correct, but no gamma-correction is pressent.

I use i3 as wm.

>   ~ % pacman -Qi clight-git
Name            : clight-git
Version         : r341.42747ad-1
Description     : A C daemon that turns your webcam into a light sensor. It can also change display gamma temperature.
Architecture    : x86_64
URL             : https://github.com/FedeDP/Clight
Licenses        : GPL
Groups          : None
Provides        : None
Depends On      : systemd  popt  libconfig  gsl  clightd-git
Optional Deps   : geoclue2: to retrieve user location through geoclue2. [installed]
                  upower: to save energy by increasing timeouts between captures while on battery. [installed]
Required By     : None
Optional For    : clightd-git
Conflicts With  : None
Replaces        : None
Installed Size  : 87.00 KiB
Packager        : Unknown Packager
Build Date      : Sat 10 Feb 2018 10:30:45 PM CET
Install Date    : Sat 10 Feb 2018 10:30:45 PM CET
Install Reason  : Explicitly installed
Install Script  : Yes
Validated By    : None

The content of the .clight.log is: (I cencored the location in logfile)

Clight
* Software version:		1.4
* Starting time:		Tue Feb 13 11:08:54 2018

Starting options:

### Generic ###
* Verbose (debugging):		Enabled
* Number of captures:		5
* Webcam device:		Unsetted
* Backlight path:		intel_backlight
* User position:		xx.xx	xx.xx
* Daily screen temp:		6500
* Nightly screen temp:		4000
* User setted sunrise:		Unsetted
* User setted sunset:		Unsetted
* Event duration:		2700
* Dimmer backlight:		20%

### Timeouts ###
* Daily timeouts:		AC 600	BATT 1200
* Nightly timeout:		AC 2700	BATT 5400
* Event timeouts:		AC 300	BATT 600
* Dimmer timeouts:		AC 300	BATT 45
* Weather timeouts:		AC 3600	BATT 10800
* Dpms timeouts:		AC 900:1200:1800	BATT 150:210:300

### Modules ###
* Gamma smooth trans:		Enabled
* Dimmer smooth trans:		Enabled
* Gamma correction:		Enabled
* Screen dpms tool:		Enabled
* Weather support:		Enabled
* Screen dimmer tool:		Enabled

(D)[11:08:54]{modules.c:59}	Signal module started.
(D)[11:08:54]{modules.c:59}	Bus module started.
(D)[11:08:54]{modules.c:96}	UserBus module being started as submodule of Bus...
(D)[11:08:54]{modules.c:59}	UserBus module started.
(D)[11:08:54]{modules.c:190}	Xorg module disabled.
(D)[11:08:54]{modules.c:200}	Disabling module Gamma as its hard dep Xorg was disabled...
(D)[11:08:54]{modules.c:190}	Gamma module disabled.
(D)[11:08:54]{modules.c:203}	Trying to start Brightness module as its Gamma soft dep was disabled...
(D)[11:08:54]{modules.c:200}	Disabling module GammaSmooth as its hard dep Gamma was disabled...
(D)[11:08:54]{modules.c:190}	GammaSmooth module disabled.
(D)[11:08:54]{modules.c:250}	GammaSmooth module destroyed.
(D)[11:08:54]{modules.c:250}	Gamma module destroyed.
(D)[11:08:54]{modules.c:200}	Disabling module Dimmer as its hard dep Xorg was disabled...
(D)[11:08:54]{modules.c:190}	Dimmer module disabled.
(D)[11:08:54]{modules.c:200}	Disabling module DimmerSmooth as its hard dep Dimmer was disabled...
(D)[11:08:54]{modules.c:190}	DimmerSmooth module disabled.
(D)[11:08:54]{modules.c:250}	DimmerSmooth module destroyed.
(D)[11:08:54]{modules.c:190}	Inhibit module disabled.
(D)[11:08:54]{modules.c:190}	UserBus module disabled.
(D)[11:08:54]{modules.c:250}	UserBus module destroyed.
(D)[11:08:54]{modules.c:250}	Inhibit module destroyed.
(D)[11:08:54]{modules.c:250}	Dimmer module destroyed.
(D)[11:08:54]{modules.c:200}	Disabling module Dpms as its hard dep Xorg was disabled...
(D)[11:08:54]{modules.c:190}	Dpms module disabled.
(D)[11:08:54]{modules.c:250}	Dpms module destroyed.
(D)[11:08:54]{modules.c:250}	Xorg module destroyed.
(D)[11:08:54]{modules.c:147}	Trying to start Brightness module as its Bus dependency was loaded...
(D)[11:08:54]{modules.c:147}	Trying to start Location module as its Bus dependency was loaded...
(D)[11:08:54]{modules.c:190}	Location module disabled.
(D)[11:08:54]{modules.c:203}	Trying to start Weather module as its Location soft dep was disabled...
(D)[11:08:54]{modules.c:250}	Location module destroyed.
(D)[11:08:54]{modules.c:147}	Trying to start Upower module as its Bus dependency was loaded...
(D)[11:08:54]{modules.c:59}	Upower module started.
(D)[11:08:54]{modules.c:147}	Trying to start Brightness module as its Upower dependency was loaded...
(D)[11:08:54]{modules.c:147}	Trying to start Weather module as its Upower dependency was loaded...
(D)[11:08:54]{modules.c:147}	Trying to start Network module as its Bus dependency was loaded...
(D)[11:08:54]{modules.c:59}	Network module started.
(D)[11:08:54]{modules.c:147}	Trying to start Weather module as its Network dependency was loaded...
(D)[11:08:54]{timer.c:34}	Setted timeout of 0s 1ns on fd 6.
(D)[11:08:54]{modules.c:59}	Weather module started.
(D)[11:08:54]{modules.c:68}	Callback added for module Weather on module Upower bus match.
(D)[11:08:54]{modules.c:68}	Callback added for module Weather on module Network bus match.
(I)[11:08:54]{weather.c:166}	Weather cloudiness: 20.
(D)[11:08:54]{timer.c:34}	Setted timeout of 10800s 0ns on fd 6.
(D)[11:08:54]{modules.c:147}	Trying to start Brightness module as its Weather dependency was loaded...
(D)[11:08:54]{brightness.c:201}	AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
(D)[11:08:54]{brightness.c:201}	BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
(D)[11:08:54]{timer.c:34}	Setted timeout of 0s 1ns on fd 7.
(D)[11:08:54]{modules.c:59}	Brightness module started.
(D)[11:08:54]{modules.c:68}	Callback added for module Brightness on module Upower bus match.
(D)[11:08:55]{brightness.c:167}	Average frames brightness: 0.547502.
(I)[11:08:55]{brightness.c:151}	New brightness value: 2299
(D)[11:08:55]{timer.c:34}	Setted timeout of 1200s 0ns on fd 7.

The status of clightd.service is running
>  systemctl status clightd.service 
● clightd.service - Bus service to manage various screen related properties (gamma, dpms, backlight)
   Loaded: loaded (/usr/lib/systemd/system/clightd.service; disabled; vendor preset: disabled)
   Active: active (running) since Tue 2018-02-13 11:08:54 CET; 13min ago
 Main PID: 16465 (clightd)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/clightd.service
           └─16465 /usr/lib/clightd/clightd
Feb 13 11:08:54 systemd[1]: Starting Bus service to manage various screen related properties (gamma, dpms, backlight)...
Feb 13 11:08:54 systemd[1]: Started Bus service to manage various screen related properties (gamma, dpms, backlight).
Feb 13 11:08:54 clightd[16465]: setting V4L2_CID_AUTOGAIN: Invalid argument

FedeDP, 2018-02-13:
Hi, thanks for your report!
As you can see:
> (D)[11:08:54]{modules.c:190}	Xorg module disabled.
> (D)[11:08:54]{modules.c:200}	Disabling module Gamma as its hard dep Xorg was disabled...

Clight fails to detect you are on a X session.
It checks 2 env variable: "DISPLAY" and "XAUTHORITY".
Can you paste here the output of:
`echo $DISPLAY`
and
`echo $XAUTHORITY`

Thanks!

flovo, 2018-02-13:
It seems, that XAUTHORITY is not set on my system:

` echo $DISPLAY
:0`
` echo $XAUTHORITY
`

FedeDP, 2018-02-13:
I see...gamma correction requires XAUTHORITY env variable (as it uses xauthority cookie to gain privileges to change your current session gamma).
Can you check if you got a .Xauthority file in your home folder?

flovo, 2018-02-13:
I've an empty .Xauthority file in my home folder.

FedeDP, 2018-02-13:
No magic cookie inside it? Weird...
Can you try if this works:
`busctl call org.clightd.backlight /org/clightd/backlight org.clightd.backlight setgamma "ssi" "$DISPLAY" "/home/$USER/.Xauthority" 4500`

flovo, 2018-02-13:
It works. Now the display is redshifed.

FedeDP, 2018-02-13:
Ok, next  Clight version will fallback to use $HOME/.Xauthority file if no XAUTHORITY env variable is found, thanks!
In the meantime you can export XAUTHORITY env variable from your .bash_profile file (or wherever you feel more convenient).

FedeDP, 2018-02-13:
@flovo  Can you paste me the output of
`loginctl show-session $YOUR_SESSION_ID -p Type`?
where $YOUR_SESSION_ID is something like "c1" (try tab-completion).
Thanks!

flovo, 2018-02-13:
```
~ % loginctl show-session c1 -p Type 
Type=x11
```

FedeDP, 2018-02-13:
Thank you again!

flovo, 2018-02-14:
Hi,
a short update.
I found that my `.xinitrc` script lacked some lines of code. Therefor the *.service didn't got the DISPLAY and XAUTHORITY variables set. 
[https://wiki.archlinux.org/index.php/Systemd/User#DISPLAY_and_XAUTHORITY](https://wiki.archlinux.org/index.php/Systemd/User#DISPLAY_and_XAUTHORITY) and [https://www.linuxsecrets.com/archlinux-wiki/wiki.archlinux.org/index.php%3Ftitle=Xinit&mobileaction=toggle_view_desktop.html#xinitrc](https://www.linuxsecrets.com/archlinux-wiki/wiki.archlinux.org/index.php%3Ftitle=Xinit&mobileaction=toggle_view_desktop.html#xinitrc)

After adding the relevant lines, clight works fine.

Thank you for your help.

FedeDP, 2018-02-14:
Thanks for taking the time to report back!
By the way on devel branch i tried to enforce some stricter/better checking to workaround issues like this :)
I'm now in a testing phase for next release, that will fix (by introducing new features) both #5 and #20 .

FedeDP, 2018-03-03:
Closed in https://github.com/FedeDP/Clight/commit/4f8efac147dec513bde4f61379b95bc537723ad3.

closed 2018-03-03 by FedeDP