FedeDP/Clight #24: Add PC support.

shpinog, 2018-04-05:
```
clight.service - C daemon to adjust screen brightness to match ambient brightness, as computed capturing frames from webcam
   Loaded: loaded (/usr/lib/systemd/user/clight.service; disabled; vendor preset: enabled)
   Active: active (running) since Fri 2018-04-06 02:28:36 MSK; 12s ago
 Main PID: 2711 (clight)
   CGroup: /user.slice/user-1000.slice/user@1000.service/clight.service
           └─2711 /usr/bin/clight

```
But in _/sys/class/backlight/_ empty. This is normal for ati \ radeon, and im can set this if used 
`xrandr --output DVI-D-0 --brightness 0.8`
can you add xrandr backend  for change  brightness ?

FedeDP, 2018-04-06:
I don't understand your issue: are you running clight on a pc (and not a laptop)?
Can you post your $USER/.clight.log ?

shpinog, 2018-04-06:
> I don't understand your issue: are you running clight on a pc (and not a laptop)?
I`m running clight on a pc.
```
Clight
* Software version:		2.0
* Starting time:		Apr  6 2018 01:55:09
Starting options:

### Generic ###
* Verbose (debugging):		Enabled
* Number of captures:		5
* Webcam device:		video0
* Backlight path:		Unsetted
* User position:		Unsetted
* Daily screen temp:		6500
* Nightly screen temp:		2000
* User setted sunrise:		Unsetted
* User setted sunset:		Unsetted
* Event duration:		1800
* Dimmer backlight pct:		0.20

### Timeouts ###
* Daily timeouts:		AC 600	BATT 1200
* Nightly timeout:		AC 2700	BATT 5400
* Event timeouts:		AC 300	BATT 600
* Dimmer timeouts:		AC 45	BATT 20
* Dpms timeouts:		AC 900:1200:1800	BATT 300:420:600

### Modules ###
* Brightness correction:		Enabled
* Gamma correction:		Enabled
* Screen dpms tool:		Enabled
* Screen dimmer tool:		Enabled

### Smooth ###
* Bright smooth trans:		Enabled
* Gamma smooth trans:		Enabled
* Dimmer smooth trans:		Enabled
* Bright smooth steps:		0.05
* Gamma smooth steps:		50
* Dimmer smooth steps:		0.05
* Bright smooth timeout:		30
* Gamma smooth timeout:		300
* Dimmer smooth timeout:		30

(D)[01:55:09]{modules.c:59}	Signal module started.
(D)[01:55:09]{modules.c:59}	Bus module started.
(D)[01:55:09]{modules.c:96}	UserBus module being started as submodule of Bus...
(D)[01:55:09]{modules.c:59}	UserBus module started.
(D)[01:55:09]{modules.c:59}	Xorg module started.
(D)[01:55:09]{modules.c:147}	Trying to start Gamma module as its Xorg dependency was loaded...
(D)[01:55:09]{modules.c:147}	Trying to start Dimmer module as its Xorg dependency was loaded...
(D)[01:55:09]{modules.c:147}	Trying to start Dpms module as its Xorg dependency was loaded...
(D)[01:55:09]{modules.c:147}	Trying to start Inhibit module as its Xorg dependency was loaded...
```

shpinog, 2018-04-06:
My request is to add support for brightness control for PCs with radeon graphics cards. As I understood in your program brightness adjustment implemented through /sys/class/backlight/, but in a PC with radeon graphics it is impossible. The brightness is adjusted via xrandr.

FedeDP, 2018-04-06:
Clight actually already supports desktop PCs in a portable way (xrandr is X only): it uses ddcutil (https://github.com/rockowitz/ddcutil).  
Have you built clightd with ddcutil support (https://github.com/FedeDP/Clightd#ddcutil-support)? On which distro are you running clight?

shpinog, 2018-04-08:
Im use archlinux.

FedeDP, 2018-04-15:
What's the output of
`pacman -Q ddcutil` ?