FedeDP/Clight #25: Kde Plasma - lower part of screen frozen

postadelmaga, 2018-04-20:
I have tried clight while using Plasma as DE  but when I start it I got the lower part of the desktop blocked:
 - 3/4 of the screen are normal while 1/4 is like frozen

FedeDP, 2018-04-20:
Well this is weird. I use plasma too. Are you running clight on a desktop pc or laptop?
Can  you share our $HOME/.clight.log?

postadelmaga, 2018-04-23:
Hi,
I am running Clight on a 
- laptop Xiaomi mi book 13'
- fresh Arch install 
- Plasma Xorg
- Clight is installed from Aur: https://aur.archlinux.org/packages/clight-git/

I just gave another try and but got the same issue, I noted that even if log out into SDDM or swirch to a terminal ( CTRL+ALT+Fx) the lower part of the screen remain the same.

Here the log 
[clight.log](https://github.com/FedeDP/Clight/files/1939091/clight.log)

FedeDP, 2018-04-23:
Mmh may I ask you to try to disable some modules (have a look at /etc/default/clight.conf)?  
I'd try to disable gamma module first, then, if issue persists, re-enable it and disable brightness module.
Let me know if any of these modules is guilty.

If none of these modules is guilty, you can try same approach with other modules, hoping to find the culprit so i can fix it.

FedeDP, 2018-07-20:
@postadelmaga  Any news on this?