FedeDP/Clight #26: Gamma correction is backwards in southern hemisphere.

blackjok3rtt, 2018-04-27:
Hey as my other report. Putting my conf file like this:
```gamma_temp = [ 5900, 3800 ] ```
Results in my daytime gamma being  set at 3800.
Settng it like this:
```gamma_temp = [ 3800, 5900 ] ```
Has it set correctly working as expected. lat long set as follows:
``` latitude = -31.4823342;
longitude = 118.281299;```

FedeDP, 2018-04-27:
Thanks for your report!  
It surely is a bug with current sunrise/sunset computation for southern emisphere. I will try to fix it asap, in the meantime you found a nice workaroud :D

blackjok3rtt, 2018-04-27:
No, its not changing at all. There is no workaround :(

FedeDP, 2018-05-01:
@blackjok3rtt I think that for now you can set fixed sunrise/sunset times in your clight.conf file.  
I'm still digging trying to understand where does your issue come from.  
Mind to share your sunrise/sunset times as identified by clight? (clight verbose will tell you when next gamma event will be).  
Thank you!

FedeDP, 2018-05-01:
@blackjok3rtt  It should be fixed now. Can you pull and rebuild to check if it is fixed for you?  
I checked with  
`TZ=Australia/Perth ./clight --lat=-31.48 --lon=118.28 --verbose`  
and it worked: you should be right now (10:42 gmt) in night mode (your current time local is 18:42, right? Well past 17.39 that is today's sunset), and tomorrow at 6.37 you should enter day mode.

blackjok3rtt, 2018-05-01:
Sorry mate, I've been horribly busy. Will reinstall this and try again for a few days. I had some problem with Xorg using 40% of a CPU core. Not sure if it was related to Clight, but as gamma wasn't working at the time I disabled it and used redshift again for a while. Thanks for fixing, will test now.

FedeDP, 2018-05-01:
No problem!  Real life always comes first :D. 
Thank you!

blackjok3rtt, 2018-05-01:
YES! its working. Will monitor and not shut down for you no problem. I will also keep any eye on my Xorg, like i said I'm not sure its related but it did't happen after I turned off Clight so I will keep a look out. Thanks.

blackjok3rtt, 2018-05-01:
One question, if I want to increase all the brightness values a little bit do I increase or decrease the numbers in the conf file?

FedeDP, 2018-05-01:
You should increase them :)

FedeDP, 2018-05-18:
Hi @blackjok3rtt !
Any news on this issue? I'd like it to be fixed before releasing next clight version and it is the only stopper right now :) 

Thank you very much!

blackjok3rtt, 2018-05-19:
Sorry mate, been snowed under last month and forgot about this. I've been using it on my laptop every day since u fixed it and haven't ran into any bugs at all. Perfect job dude.

closed 2018-05-19 by blackjok3rtt