FedeDP/Clight #27: can't build

nick87720z, 2018-06-29:
I tried trunk, 2.1 and 1.4 versions, under lubuntu 16.04. I need it for i386 laptop (only 2Gb ram).
Installed build-essentials with *-dev packages, mentioned in build deps, in README.md.
There is gcc-5.3 used (probably not best, at least gentoo was 5.4.0, right after 4.9.4, before 6.4.0.

Build log:
```
$ LANG=C make
fatal: Not a git repository (or any parent up to mount point /home/margo)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
brightness.c: In function 'callback':
brightness.c:57:5: warning: ignoring return value of 'read', declared with attribute warn_unused_result [-Wunused-result]
     read(main_p[self.idx].fd, &t, sizeof(uint64_t));
     ^
dimmer.c: In function 'callback':
dimmer.c:77:9: warning: ignoring return value of 'read', declared with attribute warn_unused_result [-Wunused-result]
         read(main_p[self.idx].fd, &t, sizeof(uint64_t));
         ^
gamma.c: In function 'callback':
gamma.c:48:5: warning: ignoring return value of 'read', declared with attribute warn_unused_result [-Wunused-result]
     read(main_p[self.idx].fd, &t, sizeof(uint64_t));
     ^
location.c: In function 'load_cache_location':
location.c:73:9: warning: ignoring return value of 'fscanf', declared with attribute warn_unused_result [-Wunused-result]
         fscanf(f, "%lf %lf", &conf.loc.lat, &conf.loc.lon);
         ^
In file included from /usr/include/fcntl.h:289:0,
                 from /usr/include/i386-linux-gnu/sys/file.h:24,
                 from lock.c:2:
In function 'open',
    inlined from 'gain_lck' at lock.c:13:14:
/usr/include/i386-linux-gnu/bits/fcntl2.h:50:4: error: call to '__open_missing_mode' declared with attribute error: open with O_CREAT or O_TMPFILE in second argument needs 3 arguments
    __open_missing_mode ();
    ^
makefile:42: recipe for target 'objects' failed
make: *** [objects] Error 1
```
(make deb doesn't go different, as it needs to build anyway)

FedeDP, 2018-06-29:
Hi, thanks for the report!
Can you try to modify src/lock.c line 13, in this way:
`lck_fd = open(lockfile, O_RDWR | O_CREAT, 0600);`
(note the added 0600 at the end)
and try to build again?

nick87720z, 2018-06-29:
Thanks, now make is ok.
Next problem is in deb generation (make deb):
dpkg-dev tells error parsing ./Clight/DEBIAN/control, because version string is empy... this is in release 2.1. It seems to be generated, becasue when i fill it, it is overwritten. I tried paste:
CLIGHT_VERSION = 2.1 to makefile, but it doesn't help.

FedeDP, 2018-06-29:
What does Version in DEBIAN/control says after a make deb?

nick87720z, 2018-06-29:
It is empty

FedeDP, 2018-06-29:
Thank you very much for your hints!
All issues is fixed as of b55aef5266e4fb8ca1bf2a8cf19baa9b1e69f11e.
Care to give it a go?
Thank you!

nick87720z, 2018-06-29:
Yes, i'm trying to get it working.
Though it is said to be designed as 0-conf, it tries with failure to find user ~/.config/clight.conf (though it has system config).

And says "Unit polkit.service" not found.
However, i have /lib/systemd/system/polkitd.service from policykit-1.

FedeDP, 2018-06-29:
> it tries with failure to find user ~/.config/clight.conf (though it has system config).
This is ok (it is just a warning).

> And says "Unit polkit.service" not found.
> However, i have /lib/systemd/system/polkitd.service from policykit-1.
This is wrong. Is polkit.service running?

FedeDP, 2018-06-30:
I see that right now policykit-1 ships polkit.service in ubuntu too.  
But on 16.04 it was named polkitd.service. I think I'll add a reference about this somewhere in my README, thanks!

FedeDP, 2018-07-04:
Fixed in clightd: https://github.com/FedeDP/Clightd/commit/975cde2ce51a9e38eac07c2308503994622910c5

closed 2018-07-04 by FedeDP