FedeDP/Clight #28: segmentation fault in backlight module

nick87720z, 2018-06-29:
I build clight deb in lubuntu 16.04 i386. Runing without no_backlight in config causes it to crash.
After doing make debug and runing in gdb from source dir, i got backtrace with 12 levels, but without function names, only two lowest levels are:
```
#11..... __libc_start_main() from libc
#12..... ?? from /lib/ld-linux.so.2
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

FedeDP, 2018-06-29:
Hi! Can you please try to build clight with debug symbols (make debug) and check the trace once more?
Does it crash immediately?
Care to paste here your conf file?

nick87720z, 2018-06-29:
I already made it - with debug symbols. Other 10 backtrace lines are useless - only addresses (they seem to be from clight itself). Without debug it gives just 2 backtrace levels. I edited makefile, adding debug options set -g3 -ggdb3 -O0.
Yet, it seems, that make deb always builds from begining, even if make or make objects was run manually. Searching way to get more useful backtrace.

nick87720z, 2018-06-29:
I forgot, i'm under ubuntu. Do i need any debub symbols for runtime dependencies, or just make debug is enough?
I ran: make clean clean-deb && rm ./clight && make debug
(it doesn't remove clight in clean), then ran again gdb -e ./clight
It gives backtrace with only two nameless levels.
Can't get even above output with 12 levels.

FedeDP, 2018-06-29:
Can you paste me your clight.log file with debug mode enabled?

nick87720z, 2018-06-29:
Clight
* Software version:		2.1
* Starting time:		Jun 30 2018 00:25:30
Starting options:

### Generic ###
* Verbose (debugging):		Disabled
* Number of captures:		5
* Webcam device:		Unsetted
* Backlight path:		Unsetted
* User position:		Unsetted
* Daily screen temp:		6500
* Nightly screen temp:		4000
* User setted sunrise:		Unsetted
* User setted sunset:		Unsetted
* Event duration:		1800
* Dimmer backlight pct:		0.20

### Timeouts ###
* Daily timeouts:		AC 600	BATT 1200
* Nightly timeout:		AC 2700	BATT 5400
* Event timeouts:		AC 300	BATT 600
* Dimmer timeouts:		AC 45	BATT 20
* Dpms timeouts:		AC 900:1200:1800	BATT 300:420:600

### Modules ###
* Brightness correction:		Enabled
* Gamma correction:		Disabled
* Screen dpms tool:		Disabled
* Screen dimmer tool:		Disabled

### Smooth ###
* Bright smooth trans:		Enabled
* Gamma smooth trans:		Enabled
* Dimmer smooth trans:		Enabled
* Bright smooth steps:		0.05
* Gamma smooth steps:		50
* Dimmer smooth steps:		0.05
* Bright smooth timeout:		30
* Gamma smooth timeout:		300
* Dimmer smooth timeout:		30

(I)[00:25:30]{clightd.c:54}	Clightd found, version: 2.1.

FedeDP, 2018-06-29:
>  Verbose (debugging):		Disabled

Run clight --verbose thanks!

nick87720z, 2018-06-29:
updated above

FedeDP, 2018-06-29:
Care to test a single capture too?
./clight -c

nick87720z, 2018-06-30:
Direct capture with above command gives this:
ad 5.14.173 14.173 14.173 14.17314.173

./clight --verbose -c: log is exactly same as above (i compared line by line).

FedeDP, 2018-06-30:
Nice, clightd is working fine.  
The issue happens when clight tries to start brightness module. 
Mind to use the attached patch and after a build retry ./clight --verbose?

`git apply clight.txt` from your clight root folder.

[clight.txt](https://github.com/FedeDP/Clight/files/2151529/clight.txt)

nick87720z, 2018-06-30:
Still no difference. I checked, that patch is applied.
Btw, did not notice, that my debug options had -O3 :) now i have backtrace with 12-lines, still useless.

I thought, issue is in code, but even when i wrote simplest program, which causes segmentation fault by freeing from just number, passed as pointer, and compiled with `gcc -O0 -g3 -ggdb3 -o test main.c` ,
it still unable to show function names :/ problem either in gcc or gdb

FedeDP, 2018-07-01:
Mmh i don't really understand what's going on here.
Can you try to add another debug printf in src/modules.c L 24? Right before 
`modules[module].init();`

Unfortunately i am not having much time this weekend to debug, sorry. 
Really thanks for your cooperation :)

nick87720z, 2018-07-01:
Yes, it crashes somewhere in this init() call. I surrounded it with "test" and "test2" prints. Last "test" appears before "Upower module started.".
What is interesting, it tries to start Brightness several times.

nick87720z, 2018-07-01:
I got to inc/commons.h. Doesn't look good idea, defining variable in header:
`struct module modules[MODULES_NUM]`.
Should it really be in one instance per object file, instead of one common for entire program?

nick87720z, 2018-07-01:
Can't be sure, what is expected behavior with modules array per file...
crash seem to happen even before modules[module].init() call, but in this block.

FedeDP, 2018-07-02:
> What is interesting, it tries to start Brightness several times.

It is because it tries to start brightness module after every "module"'s change.

> I got to inc/commons.h. Doesn't look good idea, defining variable in header:
struct module modules[MODULES_NUM].

I know, but commons.h is covered by "pragma once"  so it does not create any issue. It should be declared extern there and defined somewhere else.
> Looks like there are indeed several Brightness instances.

As i already said, it is ok. There is 1 instance that clight tries to start every time a change happen.  
If you disable upower, and gamma, clight tries to start brightness module right after bus module, thus that log you see.  
There is something wrong that happen when brightness module gets loaded, but i really don't understand what.  

Btw judging by  your gdb trace, it seems to me that the crash does not even happen in clight...i fear we are looking at the issue the wrong way.  

> Rewrote post, as i found a bit different place:
brightness check(): call() line (looks like call to dbus method webcam_available).
There are indeed some issues with webcam, but not everywhere.
guvcview and snap-photobooth (Camera app from elementary os) are ok, but e.g. pidgin also doesn't detect it (could be pidgin issue though), empathy tries seems to enable cam, but fails to create connection.

Clightd has no issue with your webcam ("captureframes" gave correct results).

Btw thank for taking the time to dig into the code, i really appreciate that :)

FedeDP, 2018-07-03:
@nick87720z I may have found your issue! Care to test?  

[bus.txt](https://github.com/FedeDP/Clight/files/2157929/bus.txt)

nick87720z, 2018-07-03:
Thanks, now that place is passed. There is yet another place, causing segfault, just after another dbus call. From log i can write it as: d(bdu)
(it reports one letter from string, with message like "Wring signature in bus call: d"...)

FedeDP, 2018-07-03:
> Thanks, now that place is passed.

Nice!  

For the other error: can you paste your clight.log in verbose mode here?  
That signature is for "setallbrightness" method in clightd. It should not fail.

nick87720z, 2018-07-03:
You did not notice reply updates :)
in call() code, in bus.c;
`r = sd_bus_message_append_basic(m, "s", s);`
probably should be
`r = sd_bus_message_append_basic(m, "s", &s);`

Crash happens exactly on this line, after change it doesn't anymore.
(this is how it is done for INT variant)

FedeDP, 2018-07-03:
Yes but i forgot to mention that :D 
sd_bus_message_append_basic() wants a pointer and char *s is already a pointer.  
There is something else going on there...

nick87720z, 2018-07-03:
Hm, you are about arguments number...
it says "too few arguments".
If i append some fifth arg, it will say "b true" (same reply, no matter, is it "Hello", 0 or 0.3 for example).

nick87720z, 2018-07-03:
d-feet showes this "Method name":
`setallbrightness (Double arg_0, Struct of (Boolean, Double, UInt32) arg_1, String arg_2) -> (Boolean arg_3)`

Trying to fill input with `{0.3, 0, 0, 0}`, i get `Must be number, not set`.

nick87720z, 2018-07-03:
Looks more like it need pointer to variable to return reply. Even if it is already pointer. Is not it?
Variables are created before loop uninitialized, thus 's' points to nothing or garbage.

nick87720z, 2018-07-03:
oh shi... it is systemd api wrapper for dbus o_O

nick87720z, 2018-07-03:
All ok... i even have misread function name, which is about to append field.
After returning back to passing just 's' (as expected by this function), i don't get segfault by some reason.
Instead there is error:
`Invalid argument 's' to call <vlahblahblah>(), expecting 'd(bdu)s'

FedeDP, 2018-07-03:
> oh shi... it is systemd api wrapper for dbus o_O
Yup.

I saw where the issue is.  
Sorry but that part of code is not properly tested as since libsystemd version 234 the nice sd_bus_message_appendv() functions was added and thus it is used.  
I did never implement various signatures (eg: "d" and "b" and "u").  
I will do that asap (probably not before this weekend though...)!

FedeDP, 2018-07-03:
@nick87720z  Ok, i did not want to let you wait too long :)
Can you test this patch (please update clight from master too!)?
[bus.txt](https://github.com/FedeDP/Clight/files/2160449/bus.txt)

nick87720z, 2018-07-03:
Now it determines some brightness, but i need yet to solve permissions for write access to intel_backlight dir.
Starting as root, it can't connect to user bus.

FedeDP, 2018-07-04:
That's weird; setallbrightness method in clightd is polkit-restricted. Do you have a correctly active user session?  
What does 

`$ loginctl show-session $XDG_VTNR`

say about your session? Can you paste it here?

nick87720z, 2018-07-04:
Ok, will look for that. Another problem, in clight daemon, is that it broke shutdown sequence (can't remember exact content from syslog), so i removed it. Will look if it presents in trunk, and if yes, create proper issue.

It is basically Lubuntu 16.04 i386... probably i would not need to set permissions as for acpilight, since daemon is under root, yet wingpanel battery applet is happy (not sure, does it use acpi interface, or works like xbacklight).

nick87720z, 2018-07-04:
`Failed to get session: No session '7' known`

FedeDP, 2018-07-04:
What does 
`$ loginctl list-sessions`
say?

FedeDP, 2018-07-04:
> Ok, will look for that. Another problem, in clight daemon, is that it broke shutdown sequence (can't remember exact content from syslog), so i removed it. Will look if it presents in trunk, and if yes, create proper issue.

What do you mean with "clight daemon"? Clightd?

nick87720z, 2018-07-04:
Its name is completely different: "c2"

FedeDP, 2018-07-04:
Can you run 

$ loginctl show-session c2

And paste the output here?

nick87720z, 2018-07-04:
```
$ loginctl show-session c2
Id=c2
User=1000
Name=margo
Timestamp=Ср 2018-07-04 16:36:24 +05
TimestampMonotonic=65002682
VTNr=7
Seat=seat0
Display=:0
Remote=no
Service=lightdm
Desktop=pantheon
Scope=session-c2.scope
Leader=1402
Audit=0
Type=x11
Class=user
Active=yes
State=active
IdleHint=no
IdleSinceHint=1530705196246824
IdleSinceHintMonotonic=1076585621
```

FedeDP, 2018-07-04:
Nice, your session is active!  
Can you try a

`$ busctl call org.clightd.backlight /org/clightd/backlight org.clightd.backlight setgamma "ssi(buu)" ":0" "/home/federico/.Xauthority" 5000 0 0 0`

And check that it properly sets your screen temperature to 5000? (it is quite visible!)  
If that works it means it is not a polkit issue (btw call it again with 6500 instead of 5000 to revert to a normal temperature).

nick87720z, 2018-07-04:
yes, this works, and effect is visible

FedeDP, 2018-07-04:
Ok, then it seems to be an issue with clightd method for setting brightness (is relies on udev).  
Can you please open an issue on clightd to track this down?  

Thank you!

FedeDP, 2018-07-04:
Can you try with the following patch (starting from a clean clight checkout, without my last patch)?
[bus2.txt](https://github.com/FedeDP/Clight/files/2163435/bus2.txt)

nick87720z, 2018-07-04:
Message "Operation not permitted" is returned from sd_bus_message_enter_container() call, causing to break loop.

nick87720z, 2018-07-04:
Found:
sd_bus_message_open_container() works instead

FedeDP, 2018-07-04:
> Can you try with the following patch (starting from a clean clight checkout, without my last patch)?

https://github.com/FedeDP/Clight/issues/28#issuecomment-402485635 :D

nick87720z, 2018-07-04:
just synced local copy :)

nick87720z, 2018-07-04:
nice :)
honestly i even enjoyed the process. Not first time. Trying to take some more active part in bug solving, yet this blind debugging would take very hard and long (another system with gentoo doesn't have systemd, as it already has openrc).

nick87720z, 2018-07-04:
Another crash... i enabled other modules, besides inhibition (it says for me, that PM inhibition is not supported)

FedeDP, 2018-07-04:
Please open a new issue with verbose log and conf file, thanks!

commit 73575fd6 (2018-07-04): Fixed bus.c call() method for libsystemd < 234 systems. Fixes #28.
closed 2018-07-04 by FedeDP