FedeDP/Clight #30: segmentation fault, loading dimmer

nick87720z, 2018-07-04:


FedeDP, 2018-07-04:
I need clight verbose log and your conf.  
Is #29 related? Was it opened by mistake?

nick87720z, 2018-07-04:
Mistake. I wanted to rename it.

nick87720z, 2018-07-04:
```
Config file /etc/default/clight.conf not found.
Gamma module disabled.
Trying to start Brightness module as its Gamma soft dep was disabled...
Location module disabled.
Location module destroyed.
Gamma module destroyed.
Signal module started.
Bus module started.
UserBus module being started as submodule of Bus...
UserBus module started.
Dpms module disabled.
Dpms module destroyed.
Xorg module started.
Trying to start Dimmer module as its Xorg dependency was loaded...
Inhibit module disabled.
Trying to start Dimmer module as its Inhibit soft dep was disabled...
UserBus module disabled.
UserBus module destroyed.
Inhibit module destroyed.
Ошибка сегментирования (сделан дамп памяти)
```
last message - segmentation fault; LANG=C doesn't work here.

FedeDP, 2018-07-04:
Can you test with this patch? ( #31 too)

[test.txt](https://github.com/FedeDP/Clight/files/2163971/test.txt)

nick87720z, 2018-07-04:
Doesn't help. As with inhibit tried to loaded, I looked above in log - it says about missing *.service file.
with no_inhibit it crashes as before.

FedeDP, 2018-07-04:
This should be proper output:  

> Gamma module disabled.
Trying to start Brightness module as its Gamma soft dep was disabled...
Location module disabled.
Location module destroyed.
Gamma module destroyed.
Signal module started.
Bus module started.
UserBus module being started as submodule of Bus...
UserBus module started.
Dpms module disabled.
Dpms module destroyed.
Xorg module started.
Trying to start Dimmer module as its Xorg dependency was loaded...
Inhibit module disabled.
Trying to start Dimmer module as its Inhibit soft dep was disabled...
UserBus module disabled.
UserBus module destroyed.
Inhibit module destroyed.
Clightd found, version: 2.1.

So, it crashes while initing Clightd module.

nick87720z, 2018-07-04:
Btw, about log:
everything is default, only these 3 lines:
```
no_dpms = 1
no_inhibit = 1
verbose = 1
```
To stop segfault, it needs either uncomment `no_dimmer = 1` or comment `no_inhibit = 1`.

FedeDP, 2018-07-04:
My guess is that here: https://github.com/FedeDP/Clight/blob/master/src/bus.c#L257, it is using userbus instead of bus. But userbus in NULL there, as userbus module has been destroyed.  
Why should it use it though, that's quite a mistery.  
Can you please check adding a printf:
`if (userbus == NULL) { printf("Nice"); }`

FedeDP, 2018-07-05:
I pushed some updates. I don't think they will fix your issue though, but never say never :)

FedeDP, 2018-07-07:
I fixed some issues spotted by valgrind. Can you test once more with updated master?

nick87720z, 2018-07-07:
Confirming, no crashes anymore.

FedeDP, 2018-07-07:
Nice to hear! Thanks for your testing!

closed 2018-07-07 by FedeDP