FedeDP/Clight #31: gamma module: No such device or address

nick87720z, 2018-07-04:
This happens inside bus.c: call() without clear reason - all arguments see to be ok  - with busctl test they cause slow fading to wanted temperature.

nick87720z, 2018-07-04:
I tried following. Changed code in end of first loop in call() so:
```
            printf("arg end\n");
            if (check_err(r, &error)) {
                printf("error - finishing\n");
                goto finish;
            }
            printf("arg end2\n");
            i++;
```
So, at least one of two last lines should be printed. However, i have:
```
arg end
No such device or address
Gamma module destroyed.
Upower module destroyed.
```

FedeDP, 2018-07-04:
How did you run clight? With which options?

nick87720z, 2018-07-04:
No options, just ./clight from sources dir. As for --verbose, i found option in conf file.
When i test gamma, i have these lines uncommented:
```
no_dpms = 1; (relying to DE settings here)
no_inhibit = 1;
no_dimmer = 1;
```

FedeDP, 2018-07-04:
> EDIT: can't understand, why either "arg end2" or "error - finishing" is not printed.

In case of fatal error, ERROR() macro (that prints "No such device or address") will leave clight through longjmp() c call.  
I guess this issue and #30 are related.

nick87720z, 2018-07-04:
Looks like gamma module needs /dev/video0, which is really missing.

nick87720z, 2018-07-04:
After modprobe video /dev/video0 appeared, yet i uncommented conf line with "video0", but it still says "No such device or address".

FedeDP, 2018-07-05:
Gamma module does not need /dev/video0, brightness module needs that: it is your webcam.

FedeDP, 2018-07-05:
Are you using updated clight from master?

nick87720z, 2018-07-05:
Yes, problem doesn't appear now.

FedeDP, 2018-07-05:
> Yes, problem doesn't appear now.

I did not understand: you were using the old patch? Is this issue fixes?

nick87720z, 2018-07-05:
I synced to latest git state, now it looks fixed.

closed 2018-07-05 by FedeDP