FedeDP/Clight #32: Wrong behavior in dimmer module

nick87720z, 2018-07-07:
With last commits it behaves somehow strange. After clight start it immediately changes brightness to minimum, then does nothing (i have timeout 5s for test). I expected, it will darken it after specified time if no activity and restore to origin on input event.

nick87720z, 2018-07-07:
It appears to be more complicated. Last time i started clight with all modules disabled and screen brightness began oscillate around some very slowly moving point. Another case, when it may appear - if both dimmer and brightness are enabled. But whe only one - it did not (may be i missed some conditions).

FedeDP, 2018-07-08:
Hi! Thanks for opening this issue.  
The issue happened to me too sometimes, but i never had the time (aka: motivation) to check it out.  
I'll surely do it this time :)  
I'm pretty sure it has something to do with a rework that went into clightd 2.0, in Clightd/src/backlight.c (https://github.com/FedeDP/Clightd/blob/master/src/backlight.c#L199).

FedeDP, 2018-07-14:
@nick87720z  Can you test with backlight smooth transitioning off?

FedeDP, 2018-07-20:
As i'm no more experiencing the bug with latest master, i'll consider this issue fixed.
Please feel free to re-open the issue in case you are still affected.

Moreover, i finally managed to release clight and clightd 2.2 :)

closed 2018-07-20 by FedeDP