FedeDP/Clight #33: clight doesn't find dependencies Ubuntu 18.04

nhejazi, 2018-07-12:
I just came across clight and was excited to try it out on Ubuntu 18.04 LTS but it seems like it runs into an error when trying to start up. In case it matters, this is a fresh Ubuntu 18.04 install with all of the dependencies for clight installed via a combination of `apt` and `dpkg`. It looks like the install of clight completes just fine, as I noticed it in the Ubuntu applications menu; however, I didn't notice it having any effect. Also, I mainly use i3 and am not sure that clight is even starting properly in this setting (I didn't try to start it from my i3 config).

When just trying to invoke clight, I get the following (even though `libconfig9` is installed properly):
```bash
❯ clight
clight: error while loading shared libraries: libconfig.so.11: cannot open shared object file: No such file or directory
```
I'm assuming that there's something odd going on with finding dependencies based on [this SO comment](https://askubuntu.com/questions/409310/error-while-loading-shared-libraries-cannot-open-shared-object-file-no-such-fi) and related posts.

Has anyone run into a similar problem on Ubuntu systems? Any help would be much appreciated.

FedeDP, 2018-07-12:
The issue is that i created the deb package from my Archlinux; it has libconfig.so.11, while ubuntu ships libconfig.so.9.  
You can fallback at building and installing yourself for now: https://github.com/FedeDP/Clight#build-instructions
Obviously you need "-dev" packages to build clight.  

I'll have to figure out a way to let deb package be built from current ubuntu version (may be using a docker image).  
Sorry for the inconvenience and thanks for the report!

FedeDP, 2018-07-12:
PS: a dirty-but-fast solution would be to link /usr/lib/libconfig.so.9 to /usr/lib/libconfig.so.11 :)

nhejazi, 2018-07-12:
thanks for the quick reply! hmm, it looks like I'm missing something --- for some reason neither `libconfig.so.9` nor `libconfig.so.11` appears under `/usr/lib/` (even after a successful `apt-get install libconfig9`). Is there something obvious that I'm just missing here?

FedeDP, 2018-07-12:
Ubuntu installs shared objects in /usr/lib/x86_64-linux-gnu/libconfig.so.9. Unfortunately I think you'll need to build and install clight manually.

nhejazi, 2018-07-12:
Based on your comments, I found that the easiest (and maybe dirtiest) way was to link /usr/lib/x86_64-linux-gnu/libconfig.so.9 to /usr/lib/libconfig.so.11, which seems to resolve the error I originally reported. It seems like clight expects to find the config file in /root/.config whereas it appears in ~/.config in my system. Is there a way to rectify this without a manual build?

FedeDP, 2018-07-13:
> It seems like clight expects to find the config file in /root/.config whereas it appears in ~/.config in my system. Is there a way to rectify this without a manual build?

It tries to find an /etc/default/clight.conf and then a $HOME/.config.  
Are you running clight with sudo or as root user? It does not need that.

nhejazi, 2018-07-20:
@FedeDP — thanks for all your help with this. I’m going to close this issue because this doesn’t seem to report any actual deficiency in clight itself, really only the fact that installation on Ubuntu can be an annoyance (since the only way appears to be to build from source — e.g., even the use of dpkg causes a lock from sudo). If the project has time at some point going forward, it might be worth making Ubuntu installs easier via setting up a PPA, etc., but that’s conditional on interest from the clight authors/maintainers and, again, not a problem with clight itself. Thanks again!

closed 2018-07-20 by nhejazi

FedeDP, 2018-07-20:
I really hope you enjoy clight!  
I'd like to add that i am thinking about entirely dropping deb packages as i don't have much time anymore for my side projects and i don't want to waste time on such a thing.  
It would be great if anyone using ubuntu could step up and creating these debs (both clight and clightd) for me, or if clight/clightd was included in debian/ubuntu official repositories.