FedeDP/Clight #35: Clight freezes part of screen

jghauser, 2018-08-27:
When I run `clight -c` the bottom ~2/3 of my screen freeze (a sharp horizontal line separates the two areas). That is, the bottom part continues to display what has been there before and does not update anymore. The only way to solve this is a reboot (as far as I know). I'm on Arch Linux, using X and i3, everything fully updated.

Here's the output of `clight -c --verbose`:

    % clight -c --verbose
    Config file /home/julian/.config/clight.conf not found.
    Signal module started.
    Bus module started.
    UserBus module being started as submodule of Bus...
    UserBus module disabled.
    Disabling module Inhibit as its hard dep UserBus was disabled...
    Inhibit module disabled.
    Trying to start Dimmer module as its Inhibit soft dep was disabled...
    Inhibit module destroyed.
    UserBus module destroyed.
    Xorg module disabled.
    Disabling module Gamma as its hard dep Xorg was disabled...
    Gamma module disabled.
    Trying to start Brightness module as its Gamma soft dep was disabled...
    Location module disabled.
    Location module destroyed.
    Gamma module destroyed.
    Disabling module Dimmer as its hard dep Xorg was disabled...
    Dimmer module disabled.
    Dimmer module destroyed.
    Disabling module Dpms as its hard dep Xorg was disabled...
    Dpms module disabled.
    Dpms module destroyed.
    Xorg module destroyed.
    Clightd found, version: 2.2.
    Clightd module started.
    Trying to start Brightness module as its Clightd dependency was loaded...
    Trying to start Brightness module as its Bus dependency was loaded...
    Trying to start Upower module as its Bus dependency was loaded...
    Upower module disabled.
    Trying to start Brightness module as its Upower soft dep was disabled...
    AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
    BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
    Set timeout of 0s 1ns on fd 6.
    Brightness module started.
    Upower module destroyed.
    Average frames brightness: 0.053657.
    New brightness pct value: 0.075434
    Brightness module destroyed.
    Clightd module destroyed.
    Bus module destroyed.
    Signal module destroyed.

Any idea what could be up here? Thanks a lot for developing this very promising tool!

FedeDP, 2018-08-27:
Hi! Thanks for taking the time to open the issue.
Your output seems fine. 
A very similar issue was already reported, but reporter did not answer back so i closed it without further investigation.
Does the same thing happen even with a normal clight run (ie: without "-c")?

jghauser, 2018-08-27:
Just tested it, yes, the same happens.

FedeDP, 2018-08-27:
Ok. Single capture ("-c" switch) does only capture 5 frames from webcam, compute their mean value, map this value to a backlight level through curve specified in clight conf file.
Can you test stopping clight and issuing a call to clightd daemon:
`busctl call org.clightd.backlight /org/clightd/backlight org.clightd.backlight setallbrightness "d(bdu)s" 0.07 0 0 0 ""`

jghauser, 2018-08-27:
The freezing also occurs with that command. It outputs `b true`.

FedeDP, 2018-08-27:
Can you paste here the output of
`ls /sys/class/backlight/`
?

jghauser, 2018-08-27:
`intel_backlight`
I have a Dell XPS 13 (9360).

FedeDP, 2018-08-27:
Very weird.
What's the output of 
`cat /sys/class/backlight/intel_backlight/max_brightness`
?
Does the freeze happen if you run (this time from root)
`echo 5 > /sys/class/backlight/intel_backlight/brightness`
?

jghauser, 2018-08-27:
Output of the first: `7500`

The second command turned my screen black. I could then increase brightness again (I use [light](https://github.com/haikarainen/light) for that). No freezing at all.

I also just noticed a rather scary thing. The last time I tried `clight` I left the computer on for a couple of minutes (usually I restart immediately). When I then restarted there were light leaks all around the edge of the screen in the area that was previously frozen. I turned my computer back off and waited for a while. It has luckily disappeared by now.

FedeDP, 2018-08-27:
This is even more weird :( 

> The second command turned my screen black. I could then increase brightness again (I use light for that). No freezing at all.

This is nice to hear.  
It seems the issue stands in the way clightd changes backlight level. It does not use "echo" command (its C equivalent) but uses udev to change the value.  
As far as i can tell, it should work exactly the same way though.  

Can you test with this example: http://dpaste.com? Save it as "test.c" and build it with 
`gcc test.c -o test -ludev`
Finally, run it with `sudo ./test 5000` (or any other number between 0 and 7500 that is your max backlight level).
It is same code as clightd but isolated to simplify the test.

Thanks!

jghauser, 2018-08-27:
I think you forgot part of the link. I'll be happy to test of course!

FedeDP, 2018-08-27:
Yes, i edited comment above :)

jghauser, 2018-08-27:
hurray! it works!

FedeDP, 2018-08-27:
It does not freeze your screen?

jghauser, 2018-08-27:
nope, it does not :)

FedeDP, 2018-08-27:
Ok. Then i have no clues on what's going on in clightd (the test.c code is **exactly** same code that is run inside clightd...).
Are you using an external monitor too? If not, can you test building clightd with this PKGBUILD (http://dpaste.com/2SK9JEZ) and check if
`busctl call org.clightd.backlight /org/clightd/backlight org.clightd.backlight setallbrightness "d(bdu)s" 0.07 0 0 0 ""`
Freezes again?

jghauser, 2018-08-27:
Same issue as before. This time I checked the systemd logs and noticed an error that appeared. I've checked the rest of the logs and that has happened approx. around the times I tried clight today, so I'm pretty sure it's related. (logs are reversed, top three are the errors)

    Aug 27 21:24:09 tiny kernel: i2c_designware i2c_designware.0: controller timed out
    Aug 27 21:24:08 tiny kernel: i2c_designware i2c_designware.0: controller timed out
    Aug 27 21:24:07 tiny kernel: i2c_designware i2c_designware.0: controller timed out
    Aug 27 21:24:06 tiny systemd[1]: Started Bus service to manage various screen related properties (gamma, dpms, backlight).
    Aug 27 21:24:06 tiny dbus-daemon[399]: [system] Successfully activated service 'org.clightd.backlight'
    Aug 27 21:24:06 tiny systemd[1]: Starting Bus service to manage various screen related properties (gamma, dpms, backlight)...
    Aug 27 21:24:06 tiny dbus-daemon[399]: [system] Activating via systemd: service name='org.clightd.backlight' unit='clightd.service' requested by ':1.82' (uid=1000 pid=3910 comm="busctl call org.clightd.backlight >

FedeDP, 2018-08-27:
Great find, thanks!
This will surely help. Is this issue still present in latest clightd (with modified PKGBUILD)?
It seems to come from ddcutil, i'd guess. But in my modified PKGBUILD ddcutil support should be disabled.

FedeDP, 2018-08-27:
PS: what's the output of
`lsmod | grep i2c`
?

jghauser, 2018-08-27:
Sorry, my bad, I seem to have made a mistake when installing clightd with your modified PKGBUILD. I've now removed clightd and reinstalled it (with your PKGBUILD). It now works (both the command you gave me earlier as well as `clight -c`. The errors have disappeared from the logs, too. So I guess ddcutil might really be to blame.

`lsmod | grep i2c` outputs:

    i2c_algo_bit           16384  1 i915
    i2c_i801               32768  0
    i2c_hid                24576  0
    hid                   139264  4 i2c_hid,usbhid,hid_multitouch,hid_generic

FedeDP, 2018-08-27:
Wonderful news! I'll dig into clightd ddcutil code tomorrow :) 
Thank you very much for your time though! Hopefully i'll find the bug and fix it.

jghauser, 2018-08-27:
Thank you for the super quick help! I'm very happy it now works at least with the modified PKGBUILD. Let me know if/when you need more testing.

FedeDP, 2018-08-28:
Can you share output of
`sudo ddcutil detect`
? Thanks!

jghauser, 2018-08-28:
`No displays found`

FedeDP, 2018-08-28:
Ok; i'm really thinking this has something to do with ddcutil right now. I'll contact its developer for support.
Unfortunately i could not reproduce it on my netbook (recently i switched from a laptop to a desktop PC so i have an external-monitor-only configuration, and haven't got any issue with ddcutil. My old netbook instead has an internal-monitor-only config, like you).

FedeDP, 2018-08-28:
Does `ddcutil detect` reproduce the issue for you?

jghauser, 2018-08-28:
Ok, thanks for following this up! For the time being I'll simply use clightd with disabled ddcutil. As I said, happy to test when you find out something.

No, the freeze does not happen with `ddcutil detect`.

FedeDP, 2018-08-28:
What about `sudo ddcutil setvcp 0x10 5` ? Does it freeze anything?

jghauser, 2018-08-28:
No it doesn't freeze the screen. It just outputs `Display not found`.

FedeDP, 2018-08-28:
Ok, i need a more in-depth test.
Can you clone clightd repository, then in src/backlight.c, at L184 comment out
`DDCUTIL_LOOP({
               add_backlight_sn(dinfo->sn);
});`
and then build it with "make" (with ddcutil support enabled).
Now you can stop both clight and clightd services: 
`systemctl --user stop clight
sudo systemctl stop clightd`
Finally run your built clightd with `sudo ./clightd` and issue from another terminal:
`busctl call org.clightd.backlight /org/clightd/backlight org.clightd.backlight setallbrightness "d(bdu)s" 0.07 0 0 0 ""`
Let me know if it freezes anything.

Sorry, i know it is quite a long operation (4-5 minutes i'd say)...if only i could reproduce it... :(

jghauser, 2018-08-28:
It worked perfectly (dimming my screen) without freezing.

FedeDP, 2018-08-28:
Thanks, we finally now know where is the issue! My code seems ok though, so i'll probably open an issue on ddcutil.
I'll keep you updated!

FedeDP, 2018-09-18:
@jghauser Hi!
As suggested on ddcutil issue https://github.com/rockowitz/ddcutil/issues/60, can you try to run 
`ddcutil interrogate`
and paste here output?
Thanks :)

jghauser, 2018-09-18:
Here you go: http://dpaste.com/0MEDHWW
Let me know if I should run the command as root.

FedeDP, 2018-09-18:
Can you try once more running with sudo, and after loading i2c-dev module?
`sudo modprobe i2c-dev`
`sudo ddcutil interrogate`

jghauser, 2018-09-18:
can't seem to modprobe i2c-dev: `modprobe: FATAL: Module i2c-dev not found in directory /lib/modules/4.18.6-arch1-1-ARCH`

FedeDP, 2018-09-19:
That's a huge news, thanks!

FedeDP, 2018-09-22:
> In your DDCUTIL_LOOP macro, replace the "if (dlist)" block with "ddca_report_display_info_list(dlist, 1);"

Care to try? Just remove from L13 to L27 in DDCUTIL_LOOP macro, and instead add a `ddca_report_display_info_list(dlist, 1); \`
Obviously you'll have to build clightd with ddcutil support enabled.

To build, please have a look at previous comment: https://github.com/FedeDP/Clight/issues/35#issuecomment-416504903.

Thank you very much!

Dr-Ongo, 2018-09-25:
Hello, I'm having the same issue, and using the same laptop (Dell XPS13 9360)
I attempted to build clightd with the suggested change and make failed with the following output:
```Frames capturing support enabled.
Gamma support enabled.
DPMS support enabled.
idle support enabled.
DDCutil support enabled.
backlight.c:180:5: error: expected identifier or ‘(’ before ‘return’
     return r;
     ^~~~~~
backlight.c:181:1: error: expected identifier or ‘(’ before ‘}’ token
 }
 ^
backlight.c:365:5: error: expected identifier or ‘(’ before ‘return’
     return 0;
     ^~~~~~
backlight.c:366:1: error: expected identifier or ‘(’ before ‘}’ token
 }
 ^
make: *** [makefile:98: objects] Error 1
```

Dr-Ongo, 2018-09-25:
Hello, I am having the same issue and using the same laptop (Dell XPS13 9630).
I tried to build Clightd with the suggested change and make failed with the following output:
```
Frames capturing support enabled.
Gamma support enabled.
DPMS support enabled.
idle support enabled.
DDCutil support enabled.
backlight.c:178:5: error: expected identifier or ‘(’ before ‘return’
     return r;
     ^~~~~~
backlight.c:179:1: error: expected identifier or ‘(’ before ‘}’ token
 }
 ^
backlight.c:363:5: error: expected identifier or ‘(’ before ‘return’
     return 0;
     ^~~~~~
backlight.c:364:1: error: expected identifier or ‘(’ before ‘}’ token
 }
 ^
make: *** [makefile:98: objects] Error 1
```

jghauser, 2018-09-25:
This is just to let you know that I'm currently travelling and won't be able to test anything until the beginning of next year. But I see that another tester has already been found :).

jghauser, 2018-09-25:
> Yeah, fortunately!
Let's hope to fix this issue before next year though :)

Dammit, I shouldn't rush messages! I meant next week, not next year.

Dr-Ongo, 2018-09-26:
Thanks for the patch.
After running the resulting clightd and then entering: 

`busctl call org.clightd.backlight /org/clightd/backlight org.clightd.backlight setallbrightness "d(bdu)s" 0.07 0 0 0 ""` in another terminal, the resulting output was `Found 0 displays`

FedeDP, 2018-09-26:
Thanks! I will report back as soon as ddcutil dev answers :)

FedeDP, 2018-09-26:
@Dr-Ongo  Did it freeze your screen though?

Dr-Ongo, 2018-09-26:
No, the screen didn't freeze

FedeDP, 2018-09-29:
Can you test with this patch applied to current master?
[clightd.txt](https://github.com/FedeDP/Clight/files/2430368/clightd.txt)

Then paste here full output of clightd (and tell me if it frozed your screen!)
Thanks!

Dr-Ongo, 2018-09-30:
```
Got display info list: 0 displays.
Freeing info list.
```
Dimmed the screen, but didn't freeze it

FedeDP, 2018-09-30:
@Dr-Ongo Are you sure you are modprobing i2c-dev before running the tests?
Moreover, does `ddcutil interrogate` freeze the screen for you too?

FedeDP, 2018-09-30:
> I didn't realise I was supposed to

Sorry, my fault!

> FATAL: Module i2c-dev not found in directory /lib/modules/4.18.9-arch1-1-ARCH when I try.

Did you by chance recently upgrade your kernel without rebooting? (I'm on arch too and i see core repo is on 4.18.10).
I think you need a reboot :)

So, reboot, then modprobe i2c-dev and try once more to run latest patched clightd.

Thank you very much!

Dr-Ongo, 2018-09-30:
Okay, after a reboot and running the patched clightd I got the same output as before, but my screen did freeze.

FedeDP, 2018-09-30:
Great! Thanks, we now know that is the function in ddcutil that is causing the issue!

FedeDP, 2018-10-01:
Can I ask you to join me and @rockowitz on related [ddcutil issue](https://github.com/rockowitz/ddcutil/issues/60)?

Btw can you try, with i2c-dev module loaded, to issue this commands and let us know (on the aforementioned ddcutil issue) which one triggers the freeze (if any)?
ddcutil detect   
ddcutil detect --verbose   
ddcutil environment   
ddcutil environment --verbose

Thank you very much for all the time you're spending helping us debug this issue! @Dr-Ongo @jghauser

FedeDP, 2018-10-12:
As it is now certain that this is not a Clight issue, i'm going to close this.

Let me thank you both (@jghauser  @Dr-Ongo) once more for the great help you provided (and are still providing!) to me and to ddcutil main developer.
Let's hope to get this issue fixed in next ddcutil release!

closed 2018-10-12 by FedeDP

rockowitz, 2018-10-12:
I'd like to extend my thanks as well to @jghauser and especially 
@Dr-Ongo, who has diligently tested ddcutil builds as we've diagnosed 
this problem.

Sanford