FedeDP/Clight #36: clight argument sunset not affected

vansoest, 2018-09-13:
by executing as user:
```/usr/bin/clight --sunset=22:00``` the sunset argument doesn't change the sunset time, my screen will shift to red when the sun goes down at 20:00.

how can I avoid the red shifting at 20:00?

FedeDP, 2018-09-13:
Btw you need both "--sunrise=" and "--sunset=" arguments! (I may change this requirement indeed, it seems a pretty stupid requirement now that i think about that).

vansoest, 2018-09-19:
You can document the requirement of both arguments or fix your code.
I will try it now with both sun arguments. thank you!

~/.clight.log
--
```
Clight
* Software version:		2.1
* Starting time:		Jul  9 2018 12:30:30
Starting options:

### Generic ###
* Verbose (debugging):		Disabled
* Number of captures:		5
* Webcam device:		Unsetted
* Backlight path:		Unsetted
* User position:		51.25	6.47
* Daily screen temp:		6500
* Nightly screen temp:		4000
* User setted sunrise:		Unsetted
* User setted sunset:		22:00
* Event duration:		1800
* Dimmer backlight pct:		0.40

### Timeouts ###
* Daily timeouts:		AC 600	BATT 1200
* Nightly timeout:		AC 2700	BATT 5400
* Event timeouts:		AC 300	BATT 600
* Dimmer timeouts:		AC 45	BATT 20
* Dpms timeouts:		AC 900:1200:1800	BATT 300:420:600

### Modules ###
* Brightness correction:		Enabled
* Gamma correction:		Enabled
* Screen dpms tool:		Enabled
* Screen dimmer tool:		Disabled

### Smooth ###
* Bright smooth trans:		Enabled
* Gamma smooth trans:		Enabled
* Dimmer smooth trans:		Enabled
* Bright smooth steps:		0.05
* Gamma smooth steps:		50
* Dimmer smooth steps:		0.05
* Bright smooth timeout:		30
* Gamma smooth timeout:		300
* Dimmer smooth timeout:		30

(I)[12:30:30]{clightd.c:54}	Clightd found, version: 2.1.
(I)[12:30:30]{gamma.c:199}	6500 gamma temp setted.
(I)[12:30:30]{gamma.c:87}	Next gamma alarm due to: Wed Sep 19 19:10:00 2018
(I)[12:30:30]{brightness.c:100}	New brightness pct value: 0.576936
(I)[12:30:30]{brightness.c:100}	New brightness pct value: 0.588322
(I)[12:30:30]{brightness.c:100}	New brightness pct value: 0.581045
(I)[12:30:30]{brightness.c:100}	New brightness pct value: 0.587505
(I)[12:30:30]{brightness.c:100}	New brightness pct value: 0.585877
(I)[12:30:30]{brightness.c:100}	New brightness pct value: 0.582181
(I)[12:30:30]{brightness.c:100}	New brightness pct value: 0.578574
(I)[12:30:30]{brightness.c:100}	New brightness pct value: 0.578959
(I)[12:30:30]{brightness.c:100}	New brightness pct value: 0.578105
(I)[12:30:30]{brightness.c:100}	New brightness pct value: 0.577484
(I)[12:30:30]{brightness.c:100}	New brightness pct value: 0.577566
(I)[12:30:30]{brightness.c:100}	New brightness pct value: 0.578256
(I)[12:30:30]{brightness.c:100}	New brightness pct value: 0.577801
(I)[12:30:30]{brightness.c:100}	New brightness pct value: 0.576994
```

FedeDP, 2018-09-19:
Let me know if it works with no issues passing both "--sunrise" and "--sunset" arguments!
> You can document the requirement of both arguments or fix your code.

I will fix it in code, by letting users to specify only one of them. For now i'll add a note in the readme though.

FedeDP, 2018-09-22:
> You can document the requirement of both arguments or fix your code.

You can now use only one of --sunrise/--sunset option in master branch.

commit 7e50d28b (2018-09-22): Avoid requiring both --sunrise and --sunset options. Fixes #36.
closed 2018-09-22 by FedeDP