FedeDP/Clight #40: Provide some way of knowing the current ambient light for easier tuning

leiserfg, 2018-10-29:


FedeDP, 2018-10-29:
Yup, this is a nice idea. 
Note that for now you can ask clightd about that, through its bus interface; a "calibration mode" for clight would be way better though.
The thing is, with latest clight (built from git master) you can already change ambient_brightness to backlight curve points through clight bus interface, and then call a "calibrate"; this way you can very quickly play around it.

FedeDP, 2018-11-17:
Latest Clight exposes a "CurrentAmbientBr" double value both through its api and in clight.log file.
I think you can play with Clight desktop file triggering consecutive captures, and checking CurrentAmbientBr value every time; then you can play with "AcCurvePoints" and "BattCurvePoints" to find the perfect match for you.
Have a look at: https://github.com/FedeDP/Clight/wiki/Api#bus-properties.

closed 2018-11-17 by FedeDP