FedeDP/Clight #42: Handle webcam shutters

vivien, 2018-11-16:
Webcam shutters are common nowadays, especially on the new Lenovo laptops with their ThinkShutter. With a shutter put on, clightd would detect a very dark ambient light and then wrongly adjust the brightness. This can be frustrating to workaround.

Here's what I get in normal use, shutter off:

````
$ busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "si" "" 5
sad "/dev/video0" 5 0.367066 0.367066 0.367066 0.367066 0.367066
````

And here's what I get with the shutter on:
````
$ busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "si" "" 5
sad "/dev/video0" 5 0.0745098 0.0745098 0.0745098 0.0745098 0.0745098
````

I don't think it makes much sense to consider a brightness value being this low as very dark room, because the screen will always provide some light, unless you're in the dark with your screen completely powered off, which a corner case ;-)

Here's my suggestion:

clightd can consider a threshold under which it will ignore the captures and log a message eventually. The package can ship with a default sensible value (e.g. 0.09) which would cover most setup (since clightd intends to be zero-conf). Users can override the threshold, setting it to 0 would naturally disable this behavior.

Would this make sense?

FedeDP, 2018-11-16:
I understand your use case, and i've a better solution for you: you can run clight with BACKLIGHT module disabled.
Then, using clight's provided desktop file, you should be able to quickly calibrate screen when you press your desired shortcut.  
Note that right now "Calibrate" bus interface method requires BACKLIGHT module to be running; but i think that can be changed if this small change is going to satisfy your needs :)
Have a look at: https://github.com/FedeDP/Clight/blob/master/src/modules/interface.c#L164.

FedeDP, 2018-11-16:
Note though that you still need at least one "functional module" running, ie: one between DPMS, BACKLIGHT, DIMMER and GAMMA.  
This is another obsolete requirement that i may drop in the near future (as it makes no more sense now that clight also offers a bus api).

FedeDP, 2018-11-18:
Commit 3acda5079ab5203d360351e6c163d8ade688e29d added a "Passive Mode" to clight: clight starts with GAMMA and BACKLIGHT in "idle" state, ie: they won't do anything on their own but they will just answer to bus API calls.

Passive mode can be enabled by config file or by "--passive" cmdline option.

This should fullfill your needs :)
Let me know!

vivien, 2018-11-19:
I'm sure some users might want manual control, in which case, this passive mode with e.g. `dex /usr/share/applications/clightc.desktop` might do the job.

However I still want clight to do its magic most of the time, I just would like it to be privacy friendly :-)

The issue still remains that if for some reason I want to put the shutter on, the backlight will be wrongly adjusted in the meantime.

While I still think a sensible threshold would fix this without any intervention, would it make sense to allow turning the auto-calibration on and off at runtime? So that you can either:
* start clight in passive mode
* enable camera or open shutter when needed
* trigger a oneshot calibration or enable auto-calibration (passive mode off)

or:
* start clight normally with auto-calibration
* disable webcam or close shutter
* enable passive mode

Thinking further, calling such method with an optional delay during which the auto-calibration is turned off would allow nice configurations in combination to #45: you can imagine turning on passive mode as soon as a special brightness key is released; or you can set a delay like 20s after temporarily adjusting the brightness when viewing a dark picture or a white screen.

FedeDP, 2018-11-19:
> While I still think a sensible threshold would fix this without any intervention, would it make sense to allow turning the auto-calibration on and off at runtime? So that you can either:

I think a sensible threshold is more sensible, ie: something like a "conf.lowest_bl_value"; if computed backlight is lower of that value, then the capture is rejected.
I prefer this one instead of changing modules' state at runtime.

vivien, 2018-11-21:
> I think a sensible threshold is more sensible, ie: something like a "conf.lowest_bl_value"; if computed backlight is lower of that value, then the capture is rejected.
I prefer this one instead of changing modules' state at runtime.

This was my initial comment indeed ;-) Glad to see that it makes sense. With a few tests with a sticker on the webcam (if we don't have a shutter) we can find a sensible value for the default config. For sure this can be overriden by the user.

Even though it is a bit unrelated, the passive mode you implemented still makes sense IMO. Would you consider renaming it to `AutoCalibration` (defaulting to true) though? I think it would be more intuitive than a "mode".

FedeDP, 2018-11-21:
> This was my initial comment indeed ;-) Glad to see that it makes sense. With a few tests with a sticker on the webcam (if we don't have a shutter) we can find a sensible value for the default config. For sure this can be overriden by the user.

I'd say that by default we should leave it to 0.0 value, to avoid any super-weird issue with anyone having strange camera results.
Moreover, I think this is somewhat an advanced customization level that should be disabled by default.

> Even though it is a bit unrelated, the passive mode you implemented still makes sense IMO. Would you consider renaming it to AutoCalibration (defaulting to true) though? I think it would be more intuitive than a "mode".

It's ok for me, AutoCalibrate to false is much more easy to understand than passive mode true :)

vivien, 2018-11-21:
I agree with you, the default threshold being 0.0 will provide a
deterministic behaviour, let’s keep it that way.

FedeDP, 2018-11-21:
Latest clight offers a conf.shutter_threshold config option for that purpose. Moreover, "Passive" mode was renamed to "NoAutoCalibrate".

Let me know if this works fine for you :)

FedeDP, 2018-11-21:
Please note that a "ShutterThreshold" writable property is exposed through bus API too; this makes it simple to try and test lots of time (ie: change ShutterThreshold and then trigger a Calibrate).

closed 2018-11-23 by FedeDP