FedeDP/Clight #43: Different brightness for different monitors

VorpalBlade, 2018-11-18:
I have a laptop (Thinkapd T480) with two external monitors connected: a Dell U2412M and an old Asus VW225 monitor. Currently when using clight, they end up at very different brightness levels (I suspect this is because they all have very different maximum brightness). In addition, the old Asus monitor can't handle too dark values (about 15% or lower) without starting to flicker (low PWM frequency).

Maybe I missed it, but there doesn't seem to be a way to set monitor-specific gamma curves and other settings. As an example, to be comfortable right now in a reasonably well lit room and of similar brightness I need the laptop monitor to be at about 40%, the Dell to be about 25% and the Asus at about 60%.

So for this to work I believe the ac_backlight_regression_points would have to be per monitor, preferably using some sort of monitor name (as opposed to connected output).

FedeDP, 2018-11-19:
> Currently when using clight, they end up at very different brightness levels (I suspect this is because they all have very different maximum brightness).

Unfortunately, that is not the case given Clight uses percentage values.
The fact is that I made a very nice and developing-friendly assumption: that all monitors have the same linear backlight growth, ie: that 30% backlight level is the same on every monitor.

>  So for this to work I believe the ac_backlight_regression_points would have to be per monitor, preferably using some sort of monitor name (as opposed to connected output).

Yeah; but there are some issues: 
1) Clightd does not support per-monitor settings (but that can be easily enabled; but you'll have to know each monitor name/edid)
2) It makes configuration very weird: how can I specify that i want to use same curve for every monitor? How can user know which identifier to use for each monitor? 

There is another assumption to simplify this though: we don't want to deal with monitor hotplug, as [ddcutil](https://github.com/rockowitz/ddcutil) does not support it.

VorpalBlade, 2018-11-19:
As for configuration file format, I'm not familiar enough with libconfig that I believe you use for it (I generally use YAML or JSON for my own stuff), so I'm not sure how easy this is: Why not add a default-curve for all monitors that don't have a specific override-curve?

FedeDP, 2018-11-19:
> or is it more fundamental (requiring Linux reboot)

I'd say this. ddcutil does not support monitor hotplug; I thought about using  [ddcci-driver-linux](https://gitlab.com/ddcci-driver-linux/ddcci-driver-linux) that would make it easier and, above all, it would drop a build time dep in favour of a runtime one.
But ddcci-driver-linux does not support monitor hotplug neither.
Both need a reboot before managing hotplugged monitor...

FedeDP, 2018-11-19:
> Why not add a default-curve for all monitors that don't have a specific override-curve?

Yeah, config part is quite easy to implement (i only care about it being simple for the end-user), so this is not my biggest concern.

VorpalBlade, 2018-11-19:
> Both need a reboot before managing hotplugged monitor...

Sounds to me like this is somehow related to the Linux kernel then, rather than the user space tools? Might also be related to the graphics driver? E.g. nvidia vs amd vs intel.

FedeDP, 2018-11-19:
> Sounds to me like this is somehow related to the Linux kernel then, rather than the user space tools? Might also be related to the graphics driver? E.g. nvidia vs amd vs intel.

I think it is, unfortunately.

VorpalBlade, 2018-11-19:
I experimented a bit with unplugging and re-plugging my monitors. Seems to work fine. I have switchable nvidia + intel graphics, with intel driving the all the output. The nvidia GPU uses the noveau driver. So maybe you should report a bug to the graphics driver you are using about this?

So in the end I don't think this is a general roadblock for this feature, plus it would also be nice for desktops with multiple monitors.

FedeDP, 2018-11-19:
That is weird; as far as I know ddcutil developer stated multiple times that it does not support monitor hotplug; it may be a specific behaviour of your vendor/graphic card? 
Btw clight part of this is quite a large work, and it requires some more thinking (eg: when I attach a new monitor, how can I detect it? And more...).
Clightd side is instead quite simple (methods to set/get backlight for a single monitor were dropped quite recently so i can restore them from git repo) .

FedeDP, 2018-12-24:
As of today, Clightd master branch supports "Set/Get/Raise/Lower" on individual monitor serial.

VorpalBlade, 2019-01-02:
Great. I'd love to work on this to make it work with the clight program, but unfortunately life has a habit of interfering and I won't be able to work on this for at least several weeks.

FedeDP, 2019-01-02:
There is no hurry indeed; i'm going to take some time to fix some small API flaws in libmodule, and to port clight to libmodule too.
I'd like to port another project of mine too, but it is bottom priority right now.

I'd say it will take at least a couple of months!

FedeDP, 2021-04-15:
Latest commit adds support for monitor specific adjustment curves: the idea is that you can specify a curve that maps `default_backlight_level(from default curve) -> different backlight level for monitor`.

EG: you can say that backlight level 0.5 obtained from the `ambient_brightness -> default backlight` curve, should be a 0.7 value instead on your monitor.

@VorpalBlade i don't know if you are still using Clight or interested in it, but i finally managed to fix this :)

yochananmarqos, 2021-04-15:
That's great! However, it doesn't seem to find a monitor ID, it just prints my backlight:
```
❯ busctl call org.clightd.clightd /org/clightd/clightd/Backlight org.clightd.clightd.Backlight GetAll "s" ""
a(sd) 1 "nvidia_0" 0.28
```

```
❯ ls /sys/class/backlight/
nvidia_0
```

```
❯ xrandr
Screen 0: minimum 8 x 8, current 3840 x 1080, maximum 32767 x 32767
DP-0 disconnected (normal left inverted right x axis y axis)
DP-1 disconnected (normal left inverted right x axis y axis)
HDMI-0 connected 1920x1080+1920+0 (normal left inverted right x axis y axis) 0mm x 0mm
  # unused modes removed
DP-2 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 344mm x 194mm
   1920x1080    144.00*+  60.00  
DP-3 disconnected (normal left inverted right x axis y axis)
DP-4 disconnected (normal left inverted right x axis y axis)
```

FedeDP, 2021-04-16:
@yochananmarqos  do you have clightd built with ddc support?

VorpalBlade, 2021-04-16:
> @VorpalBlade i don't know if you are still using Clight or interested in it, but i finally managed to fix this :)

I haven't been for quite some time, but I might try it again soonish (very busy currently) now that this is supported.

yochananmarqos, 2021-04-16:
@FedeDP: Yes, I'm the `clightd-git` / `clight-git` AUR package maintainer. I recently rebuilt Clightd with DDCUtil 1.1.0 if that makes a difference.

FedeDP, 2021-04-16:
Mmh and your external monitor supports DDC?
If yes, does external monitors get attached at runtime?

yochananmarqos, 2021-04-16:
I don't know. It's an ASUS VS238H-B I've had for several years. I don't see anything about it in the [specs](https://www.asus.com/ca-en/Commercial-Monitors/VS238HP/).

FedeDP, 2021-04-16:
What's the output of 'sudo ddcutil detect'?

yochananmarqos, 2021-04-16:
It's only detecting my laptop display:

```
Invalid display
   I2C bus:  /dev/i2c-4
   EDID synopsis:
      Mfg id:               NCP
      Model:                
      Product code:         77
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2019,  Week: 51
   DDC communication failed
   This appears to be a laptop display. Laptop displays do not support DDC/CI.
```

FedeDP, 2021-04-16:
It seems like you are out of luck then :(
You can try to open an issue on ddcutil project; but first you should do some research to understand if the monitor supports ddc/ci protocol.

Bonnee, 2021-04-17:
How do I escape characters in the config file? Apparently, one of my monitor's id is `/dev/i2c-5` and the config parser doesn't seem to like that name (gives syntax error).

```
$ busctl call org.clightd.clightd /org/clightd/clightd/Backlight org.clightd.clightd.Backlight GetAll "s" ""
a(sd) 2 "DL4X523" 0.08 "/dev/i2c-5" 0.08
```

FedeDP, 2021-04-18:
@Bonnee  Thank you, that's right!
I may have to change the new option format to fix this.
I will have a look!

FedeDP, 2021-04-18:
@Bonnee  it should now be properly fixed. 
The new option format has changed: https://github.com/FedeDP/Clight/blob/master/Extra/clight.conf#L129 and monitor_id is read from a string element instead of parsing settings group name, giving full freedom.

Bonnee, 2021-04-18:
Indeed now the config file gets parsed correctly, but I cannot get the feature to work. All monitors always have the same backlight no matter the overridden regression curves

FedeDP, 2021-04-18:
If you enable verbose logging, you should be able to see a "Using specific curve for %s monitor" log if everything is working fine.

Bonnee, 2021-04-18:
Yeah I don't see those lines in the logs. I will test further next weekend

NICHOLAS85, 2021-04-19:
Hello! I'd like to chime in, glad to see you were able to add this feature! For now I'm having an issue where I can't find my monitor ID using the given busctl but ddcutil is able to detect another monitor. Heres the output:
```sh
$ busctl call org.clightd.clightd /org/clightd/clightd/Backlight org.clightd.clightd.Backlight GetAll "s" ""
a(sd) 1 "intel_backlight" 0.95

$ sudo ddcutil detect
Invalid display
   I2C bus:             /dev/i2c-4
   EDID synopsis:
      Mfg id:           LGD
      Model:
      Serial number:
      Manufacture year: 2016
      EDID version:     1.4
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI.

Display 1
   I2C bus:             /dev/i2c-5
   EDID synopsis:
      Mfg id:           ACR
      Model:            XF251Q
      Serial number:    TAUAA0018521
      Manufacture year: 2018
      EDID version:     1.3
   VCP version:         2.2
```

FedeDP, 2021-04-19:
Hi! Thanks for testing this!
Is your clightd built with ddc support?
If yes, you should see same monitors in both commands.
Note, also, that ddcutil library does not support monitor hotplugging: it means that if clightd is started without a connected external monitor, it won't be able to detect it when it gets plugged.
This is a libddcutil limitation.
Instead, ddc command works because it is a oneshot command, ie it does a refresh of list of monitors when started.
Library instead does not support reacting to hotplug events right now.

NICHOLAS85, 2021-04-19:
Ahh I see, I had to restart clightd in order to get my monitor to show up. It's working now! If clightd didn't detect a monitor like in the situation above would Clight still be able to perform per-monitor backlight curves as long as the correct monitor ID was in the config file? Or would I need to restart clightd and have the monitor be detected in order to have it be manipulated?

FedeDP, 2021-04-19:
> It's working now! If clightd didn't detect a monitor like in the situation above would Clight still be able to perform per-monitor backlight curves as long as the correct monitor ID was in the config file? 

Unfortunately not :( You need to restart both Clightd and Clight (in this order!).
Only restarting clightd works, but dimming and dpms would stop working in clight.
Hopefully ddcutil will support hotplug in its library sooner or later!

FedeDP, 2021-05-06:
Closing; feel free to reopen if any issue appears.

closed 2021-05-06 by FedeDP

yochananmarqos, 2021-05-14:
What about every time the primary monitor brightness changes, it uses Xrandr to change the brightness to match on the secondary monitor? For example, Clight takes a light reading and it's 0.98. Then the secondary can be changed via `xrandr --output HDMI-0 --brightness 0.98`.