FedeDP/Clight #44: Support keyboard backlit

vivien, 2018-11-18:


vivien, 2018-11-19:
For info, upower allows to set the 3-value keyboard backlight for ThinkPads where:
````
dbus-send --system --type=method_call  --dest="org.freedesktop.UPower" "/org/freedesktop/UPower/KbdBacklight" "org.freedesktop.UPower.KbdBacklight.SetBrightness" int32:1
````
sets the backlight to low,
````
dbus-send --system --type=method_call  --dest="org.freedesktop.UPower" "/org/freedesktop/UPower/KbdBacklight" "org.freedesktop.UPower.KbdBacklight.SetBrightness" int32:2
````
sets the backlight to high,
````
dbus-send --system --type=method_call  --dest="org.freedesktop.UPower" "/org/freedesktop/UPower/KbdBacklight" "org.freedesktop.UPower.KbdBacklight.SetBrightness" int32:0
````
turns off the keyboard backlight.

FedeDP, 2018-11-19:
Thanks for the documentation!
I'll leave this here just for reference: https://upower.freedesktop.org/docs/KbdBacklight.html.

Are you by chance willing to do the job? It is a small implementation that can be very useful to dive into clight code.
I can direct you on how it should be implemented obviously.
Let me know!

FedeDP, 2018-11-21:
Latest Clight offers UPower integration to auto calibrate keyboard backlight.
It is enabled by default (where supported), and it can be disabled through a new config option.

Let me know if this works fine!

vivien, 2018-11-23:
It works well thanks! The backlit is enabled for an ambient brightness of 0.435 which doesn't seem necessary (I hardly see the keyboard light in fact, only very useful when it's dark at night), but I guess we can work on the triggers later. I'm closing this, thanks!

FedeDP, 2018-11-23:
Fixed in 1d0b5c4740c5d67c18478b1c8caaa5308f4db59b.

closed 2018-11-23 by FedeDP

vivien, 2018-11-23:
Ha ha indeed, the keyboard backlit is useful when using the laptop in a room without light, where the keys are hard or impossible to distinguish. That's when it must be at its maximum. Another valid example is on a plane or train by night. The backlit doesn't provide much light, only makes the keys visible. Thank you for the quick fixup!