FedeDP/Clight #47: Valgrind errors in geoclue code.

VorpalBlade, 2018-11-19:
Just running under valgrind without clightd running and clight not installed caused the following:

```
$ valgrind --track-origins=yes ./clight
==29667== Memcheck, a memory error detector
==29667== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==29667== Using Valgrind-3.14.0 and LibVEX; rerun with -h for copyright info
==29667== Command: ./clight
==29667== 
Config file /etc/default/clight.conf not found.
Config file /home/arvid16/.config/clight.conf not found.
==29667== Invalid read of size 2
==29667==    at 0x4C33080: memcpy@@GLIBC_2.14 (vg_replace_strmem.c:1034)
==29667==    by 0x407AD2: call (bus.c:175)
==29667==    by 0x405400: geoclue_get_client (location.c:143)
==29667==    by 0x405270: geoclue_init (location.c:93)
==29667==    by 0x405006: init (location.c:33)
==29667==    by 0x40C5C4: init_modules (modules.c:24)
==29667==    by 0x40CE61: started_cb (modules.c:155)
==29667==    by 0x40CEDC: poll_cb (modules.c:170)
==29667==    by 0x404830: main_poll (main.c:109)
==29667==    by 0x404727: main (main.c:42)
==29667==  Address 0x7187d56 is 102 bytes inside a block of size 103 alloc'd
==29667==    at 0x4C3074F: realloc (vg_replace_malloc.c:826)
==29667==    by 0x40B038D: bus_socket_read_message (bus-socket.c:944)
==29667==    by 0x40BCD5E: bus_read_message.lto_priv.204 (sd-bus.c:1689)
==29667==    by 0x40BC9B6: sd_bus_call (sd-bus.c:2087)
==29667==    by 0x4079D5: call (bus.c:157)
==29667==    by 0x405400: geoclue_get_client (location.c:143)
==29667==    by 0x405270: geoclue_init (location.c:93)
==29667==    by 0x405006: init (location.c:33)
==29667==    by 0x40C5C4: init_modules (modules.c:24)
==29667==    by 0x40CE61: started_cb (modules.c:155)
==29667==    by 0x40CEDC: poll_cb (modules.c:170)
==29667==    by 0x404830: main_poll (main.c:109)
==29667== 
==29667== Invalid read of size 2
==29667==    at 0x4C3308E: memcpy@@GLIBC_2.14 (vg_replace_strmem.c:1034)
==29667==    by 0x407AD2: call (bus.c:175)
==29667==    by 0x405400: geoclue_get_client (location.c:143)
==29667==    by 0x405270: geoclue_init (location.c:93)
==29667==    by 0x405006: init (location.c:33)
==29667==    by 0x40C5C4: init_modules (modules.c:24)
==29667==    by 0x40CE61: started_cb (modules.c:155)
==29667==    by 0x40CEDC: poll_cb (modules.c:170)
==29667==    by 0x404830: main_poll (main.c:109)
==29667==    by 0x404727: main (main.c:42)
==29667==  Address 0x7187d58 is 1 bytes after a block of size 103 alloc'd
==29667==    at 0x4C3074F: realloc (vg_replace_malloc.c:826)
==29667==    by 0x40B038D: bus_socket_read_message (bus-socket.c:944)
==29667==    by 0x40BCD5E: bus_read_message.lto_priv.204 (sd-bus.c:1689)
==29667==    by 0x40BC9B6: sd_bus_call (sd-bus.c:2087)
==29667==    by 0x4079D5: call (bus.c:157)
==29667==    by 0x405400: geoclue_get_client (location.c:143)
==29667==    by 0x405270: geoclue_init (location.c:93)
==29667==    by 0x405006: init (location.c:33)
==29667==    by 0x40C5C4: init_modules (modules.c:24)
==29667==    by 0x40CE61: started_cb (modules.c:155)
==29667==    by 0x40CEDC: poll_cb (modules.c:170)
==29667==    by 0x404830: main_poll (main.c:109)
==29667== 
check_clightd_version(): The name org.clightd.clightd was not provided by any .service files
No clightd found. Clightd is a mandatory dep.
No functional module is running. Nothing to do...leaving.
==29667== 
==29667== HEAP SUMMARY:
==29667==     in use at exit: 8,192 bytes in 2 blocks
==29667==   total heap usage: 448 allocs, 446 frees, 94,220 bytes allocated
==29667== 
==29667== LEAK SUMMARY:
==29667==    definitely lost: 0 bytes in 0 blocks
==29667==    indirectly lost: 0 bytes in 0 blocks
==29667==      possibly lost: 0 bytes in 0 blocks
==29667==    still reachable: 8,192 bytes in 2 blocks
==29667==         suppressed: 0 bytes in 0 blocks
==29667== Rerun with --leak-check=full to see details of leaked memory
==29667== 
==29667== For counts of detected and suppressed errors, rerun with: -v
==29667== ERROR SUMMARY: 1566 errors from 2 contexts (suppressed: 0 from 0)
```

FedeDP, 2018-11-19:
Thanks for the report!
Can you try reverting 600f9fafba5f8c97ae96d0ee688021099a1d5151?

VorpalBlade, 2018-11-19:
Yeah that removed the error. I guess you depended on the trailing null byte that strncpy provides most of the time?

FedeDP, 2018-11-19:
Yes that commit should not have been committed :)

commit 7ff8642c (2018-11-19): Fixed #47.
closed 2018-11-19 by FedeDP