FedeDP/Clight #48: CMake Error

yochananmarqos, 2018-11-24:
Manjaro Linux
`pamac` 7.3.0-5

```
Building clight-git...
==> Making package: clight-git r488.59886b0-1 (Sat 24 Nov 2018 08:35:53 AM MST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating Clight git repo...
Fetching origin
==> Validating source files with sha256sums...
    Clight ... Skipped
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Creating working copy of Clight git repo...
Cloning into 'Clight'...
done.
==> Starting prepare()...
==> Starting pkgver()...
==> Starting build()...
-- The C compiler identification is GNU 8.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.5.4") 
-- Checking for modules 'popt;gsl;libconfig'
--   Found popt, version 1.16
--   Found gsl, version 2.5
--   Found libconfig, version 1.7.2
-- Checking for one of the modules 'libelogind;libsystemd>=221'
CMake Error at CMakeLists.txt:84 (install):
  install FILES given no DESTINATION!

-- Configuring incomplete, errors occurred!
See also "/tmp/makepkg/clight-git/src/Clight/build/CMakeFiles/CMakeOutput.log".
==> ERROR: A failure occurred in build().
    Aborting...
```
[CMakeOutput.log](https://pastebin.com/Mv2zW4uB)

FedeDP, 2018-11-24:
@VorpalBlade  Any idea? It seems COMPLETIONS_DIR is empty there:
`install(FILES ${EXTRA_DIR}/clight
        DESTINATION ${COMPLETIONS_DIR})`

FedeDP, 2018-11-25:
@yochananmarqos  It seems you haven't got bash-completion package installed.
My fault, as it should be a build-time dependency.

I will push a new PKGBUILD with the new dep. Moreover, i will gladly make it an optional build dep (ie: you are free to modify PKGBUILD to avoid the new dep, if you wish).
Let me know!

FedeDP, 2018-11-25:
> Hm, pkg-config should resolve the completions directory just above that:
pkg_get_variable(COMPLETIONS_DIR bash-completion completionsdir)

Yup, the issue is bash-completion package is needed for that to work. So, i added bash-completion to compile-time (optional) dep, and fixed CMakeLists to avoid installing bash completion script if COMPLETIONS_DIR is empty.

yochananmarqos, 2018-11-25:
It builds now, thanks. Shouldn't `bash-completion` be in `makedepends` then?

closed 2018-11-25 by yochananmarqos

FedeDP, 2018-11-25:
> Shouldn't bash-completion be in makedepends then?

Nice catch, thanks! Fixed in latest clight.