FedeDP/Clight #49: After SystemT suspend, time/values aren’t updated

vansoest, 2018-11-30:
Running clight as SystemD user service and clightd as normal SystemD service.

If I suspend(suspend to RAM) my workstation at night and e.g. the colortemperature is very warm. On the next morning when I resume my workstation the colortemperature isn’t updated, again with a warm colortemperature. So every resume at Daytime I have to restart the clight user service, that is my workaround at the moment.

Using clight on Arch Linux:
Version: efb0f9 - 2.5

FedeDP, 2018-11-30:
Thanks for opening the bug report.
I think i should go with CLOCK_BOOTTIME for GAMMA module, as stated here: https://linux.die.net/man/2/clock_gettime 
> Identical to CLOCK_MONOTONIC, except it also includes any time that the system is suspended.

FedeDP, 2018-11-30:
Can you try once more after latest commit? (rebuild clight-git and clightd-git)

vansoest, 2018-12-01:
thank you!
so now I’ve rebuilt the packages and restarted the systemd services for both. I will write to you tomorrow.

FedeDP, 2018-12-01:
> * Starting time:		Sat Dec  1 11:11:42 2018

That's weird: it seems clight was started this morning...

> set_temp(): Operation not permitted
> capture_frames_brightness(): Operation not permitted

It seems you haven't got permissions to call these exposed method. Is your session active?

FedeDP, 2018-12-01:
Fixed after latest commit.

FedeDP, 2018-12-09:
Mmh i was able to reproduce your issue and it was fixed for me after that commit.
Are you on clight 3.0? Can you attach a new clight.log?

vansoest, 2018-12-09:
$ cat .clight.log 
```
(W)[21:33:56]{log.c:17}	Resource temporarily unavailable
(E)[21:33:56]{log.c:18}	A lock is present on /home/sophie/.clight.log. Another clight instance running?
(D)[21:38:55]{my_math.c:42}	Average frames brightness: 0.155208.
(I)[21:38:55]{backlight.c:106}	Ambient brightness: 0.155 -> Backlight pct: 0.524
(D)[21:38:55]{timer.c:41}	Set timeout of 300s 0ns on fd 9.
(D)[21:43:56]{my_math.c:42}	Average frames brightness: 0.241789.
(I)[21:43:57]{backlight.c:106}	Ambient brightness: 0.242 -> Backlight pct: 0.585
(D)[21:43:57]{timer.c:41}	Set timeout of 300s 0ns on fd 9.
(D)[21:48:57]{my_math.c:42}	Average frames brightness: 0.213904.
(I)[21:48:58]{backlight.c:106}	Ambient brightness: 0.214 -> Backlight pct: 0.566
(D)[21:48:58]{timer.c:41}	Set timeout of 300s 0ns on fd 9.
(D)[21:53:58]{my_math.c:42}	Average frames brightness: 0.327802.
(I)[21:53:59]{backlight.c:106}	Ambient brightness: 0.328 -> Backlight pct: 0.643
(D)[21:53:59]{timer.c:41}	Set timeout of 300s 0ns on fd 9.
(D)[21:58:59]{my_math.c:42}	Average frames brightness: 0.212729.
(I)[21:59:00]{backlight.c:106}	Ambient brightness: 0.213 -> Backlight pct: 0.565
(D)[21:59:00]{timer.c:41}	Set timeout of 300s 0ns on fd 9.
(D)[22:00:00]{gamma.c:187}	Currently inside an event.
(I)[22:00:00]{gamma.c:200}	4000 gamma temp set.
(I)[22:00:00]{gamma.c:78}	Next gamma alarm due to: Sun Dec  9 22:30:00 2018
(D)[22:00:00]{timer.c:41}	Set timeout of 1800s 0ns on fd 8.
(D)[22:04:00]{my_math.c:42}	Average frames brightness: 0.212520.
(I)[22:04:01]{backlight.c:106}	Ambient brightness: 0.213 -> Backlight pct: 0.565
(D)[22:04:01]{timer.c:41}	Set timeout of 300s 0ns on fd 9.
(D)[22:09:01]{my_math.c:42}	Average frames brightness: 0.211567.
(I)[22:09:02]{backlight.c:106}	Ambient brightness: 0.212 -> Backlight pct: 0.564
(D)[22:09:02]{timer.c:41}	Set timeout of 300s 0ns on fd 9.
(D)[22:14:02]{my_math.c:42}	Average frames brightness: 0.212423.
(I)[22:14:03]{backlight.c:106}	Ambient brightness: 0.212 -> Backlight pct: 0.565
(D)[22:14:03]{timer.c:41}	Set timeout of 300s 0ns on fd 9.
(D)[22:19:04]{my_math.c:42}	Average frames brightness: 0.211609.
(I)[22:19:04]{backlight.c:106}	Ambient brightness: 0.212 -> Backlight pct: 0.564
(D)[22:19:04]{timer.c:41}	Set timeout of 300s 0ns on fd 9.
(D)[22:24:05]{my_math.c:42}	Average frames brightness: 0.326524.
(I)[22:24:05]{backlight.c:106}	Ambient brightness: 0.327 -> Backlight pct: 0.643
(D)[22:24:05]{timer.c:41}	Set timeout of 300s 0ns on fd 9.
(D)[22:29:06]{my_math.c:42}	Average frames brightness: 0.211869.
(I)[22:29:06]{backlight.c:106}	Ambient brightness: 0.212 -> Backlight pct: 0.564
(D)[22:29:06]{timer.c:41}	Set timeout of 300s 0ns on fd 9.
(I)[22:30:00]{gamma.c:200}	4000 gamma temp set.
(I)[22:30:00]{gamma.c:78}	Next gamma alarm due to: Mon Dec 10 07:55:00 2018
(D)[22:30:00]{timer.c:41}	Set timeout of 33900s 0ns on fd 8.
(D)[22:30:00]{timer.c:41}	Set timeout of 2646s 0ns on fd 9.
(D)[23:14:06]{my_math.c:42}	Average frames brightness: 0.212165.
(I)[23:14:07]{backlight.c:106}	Ambient brightness: 0.212 -> Backlight pct: 0.565
(D)[23:14:07]{timer.ca:41}	Set timeout of 2700s 0ns on fd 9.
(D)[23:59:08]{my_math.c:42}	Average frames brightness: 0.211246.
(I)[23:59:08]{backlight.c:106}	Ambient brightness: 0.211 -> Backlight pct: 0.564
(D)[23:59:08]{timer.c:41}	Set timeout of 2700s 0ns on fd 9.
```

$ ps aux | grep '[c]light' 
```
root     21599  0.0  0.0 250300  5220 ?        Ss   Dez02   0:06 /usr/lib/clightd/clightd
sophie   27616  0.0  0.0  19644  2536 pts/6    S+   Dez08   0:00 clight
```

FedeDP, 2018-12-10:
This log does not make sense to me :( can you give me some background?
I see:
> (W)[21:33:56]{log.c:17}	Resource temporarily unavailable
(E)[21:33:56]{log.c:18}	A lock is present on /home/sophie/.clight.log. Another clight instance running?

It should've quit here. (E)rrors messages are fatal ones, ie: they quit clight.
Moreover, i cannot see your issue here: when did you suspend your system?

vansoest, 2018-12-11:
You are right, the log file wasn‘t helpful, you can ignore it, here is a _fresh_ one.

Started yesterday clight manually at 11:37.
Suspended my workstation at 0:00 and resumed it at 9:38.

```$ cat .clight.log 
Clight
* Software version:		3.0
* Starting time:		Mon Dec 10 11:37:41 2018

Starting options:

### Generic ###
* Verbose (debugging):		Enabled
* Automatic calibration:		Enabled
* Keyboard backlight:		Disabled
* Number of captures:		5
* Webcam device:		Unset
* Backlight path:		Unset
* Shutter threshold:		0.00
* User position:		Unset
* Daily screen temp:		6500
* Nightly screen temp:		4000
* User set sunrise:		Unset
* User set sunset:		22:00
* Event duration:		1800
* Dimmer backlight pct:		0.10

### Timeouts ###
* Daily timeouts:		AC 600	BATT 1200
* Nightly timeout:		AC 2700	BATT 5400
* Event timeouts:		AC 300	BATT 600
* Dimmer timeouts:		AC 45	BATT 20
* Dpms timeouts:		AC 900:1200:1800	BATT 300:420:600

### Modules ###
* BACKLIGHT:		Enabled
* GAMMA:		Enabled
* DIMMER:		Disabled
* DPMS:		Enabled

### Smooth ###
* Bright smooth trans:		Enabled
* Gamma smooth trans:		Enabled
* Dimmer smooth trans:		Enabled
* Bright smooth steps:		0.05
* Gamma smooth steps:		50
* Dimmer smooth steps:		0.05
* Bright smooth timeout:		30
* Gamma smooth timeout:		300
* Dimmer smooth timeout:		30

(D)[11:37:41]{modules.c:68}	SIGNAL module started.
(D)[11:37:41]{modules.c:68}	BUS module started.
(D)[11:37:41]{modules.c:105}	USERBUS module being started as submodule of BUS...
(D)[11:37:41]{modules.c:68}	USERBUS module started.
(D)[11:37:41]{modules.c:105}	INTERFACE module being started as submodule of USERBUS...
(D)[11:37:41]{modules.c:68}	INTERFACE module started.
(D)[11:37:41]{modules.c:156}	Trying to start BACKLIGHT module as its INTERFACE dependency was loaded...
(D)[11:37:41]{modules.c:156}	Trying to start GAMMA module as its INTERFACE dependency was loaded...
(D)[11:37:41]{modules.c:204}	DIMMER module disabled.
(D)[11:37:41]{modules.c:204}	INHIBIT module disabled.
(D)[11:37:41]{modules.c:270}	INHIBIT module destroyed.
(D)[11:37:41]{modules.c:270}	DIMMER module destroyed.
(D)[11:37:41]{modules.c:68}	XORG module started.
(D)[11:37:41]{modules.c:156}	Trying to start GAMMA module as its XORG dependency was loaded...
(D)[11:37:41]{modules.c:156}	Trying to start DPMS module as its XORG dependency was loaded...
(D)[11:37:41]{modules.c:156}	Trying to start LOCATION module as its BUS dependency was loaded...
(D)[11:37:41]{timer.c:41}	Set timeout of 3s 0ns on fd 7.
(D)[11:37:41]{modules.c:68}	LOCATION module started.
[... 256 lines cut ...]
(I)[22:29:01]{backlight.c:106}	Ambient brightness: 0.208 -> Backlight pct: 0.562
(D)[22:29:01]{timer.c:41}	Set timeout of 300s 0ns on fd 9.
(I)[22:30:00]{gamma.c:200}	4000 gamma temp set.
(I)[22:30:00]{gamma.c:78}	Next gamma alarm due to: Tue Dec 11 07:59:00 2018
(D)[22:30:00]{timer.c:41}	Set timeout of 34140s 0ns on fd 8.
(D)[22:30:00]{timer.c:41}	Set timeout of 2641s 0ns on fd 9.
(D)[23:14:01]{my_math.c:42}	Average frames brightness: 0.207800.
(I)[23:14:02]{backlight.c:106}	Ambient brightness: 0.208 -> Backlight pct: 0.562
(D)[23:14:02]{timer.c:41}	Set timeout of 2700s 0ns on fd 9.
(D)[23:59:02]{my_math.c:42}	Average frames brightness: 0.204388.
(I)[23:59:03]{backlight.c:106}	Ambient brightness: 0.204 -> Backlight pct: 0.559
(D)[23:59:03]{timer.c:41}	Set timeout of 2700s 0ns on fd 9.
(W)[09:38:52]{bus.c:308}	capture_frames_brightness(): Operation not permitted
(D)[09:38:52]{timer.c:41}	Set timeout of 2700s 0ns on fd 9.
(W)[09:38:52]{bus.c:308}	set_temp(): Operation not permitted
(I)[09:38:52]{gamma.c:78}	Next gamma alarm due to: Tue Dec 11 21:30:00 2018
(D)[09:38:52]{timer.c:41}	Set timeout of 42668s 0ns on fd 8.
(D)[09:38:52]{timer.c:41}	Set timeout of 599s 0ns on fd 9.
(D)[09:48:51]{my_math.c:42}	Average frames brightness: 0.160374.
(I)[09:48:52]{backlight.c:106}	Ambient brightness: 0.160 -> Backlight pct: 0.528
(D)[09:48:52]{timer.c:41}	Set timeout of 600s 0ns on fd 9.
(D)[09:58:52]{my_math.c:42}	Average frames brightness: 0.231282.
(I)[09:58:53]{backlight.c:106}	Ambient brightness: 0.231 -> Backlight pct: 0.578
(D)[09:58:53]{timer.c:41}	Set timeout of 600s 0ns on fd 9.
```

FedeDP, 2018-12-11:
I can see a couple of ways to fix this for you: 
* Avoid locking session (ok this is surely not what you want, but it's worth listing it here)
* The best method, ie: write a polkit authorization rule to satisfy your requirements: https://wiki.archlinux.org/index.php/Polkit#Authorization_rules. It is kinda easy. 
I think something like:
```
cat /etc/polkit-1/rules.d/80-clightd.rules
/* Allow any user to call clightd without authentication */
polkit.addRule(function(action, subject) {
    if (action.id.indexOf("org.clightd.clightd.") == 0) {
        return polkit.Result.YES;
    }
});
```
should do the trick.

For better security, you can even create a "clightd" group, add your user to it, and then use a rule like:
```
cat /etc/polkit-1/rules.d/80-clightd.rules
/* Allow any user belonging to "clightd" group to call clightd without authentication */
polkit.addRule(function(action, subject) {
    if (action.id.indexOf("org.clightd.clightd.") == 0) {
         if (subject.isInGroup("clightd")) {
            return polkit.Result.YES;
        } else {
            return polkit.Result.NO;
        }
    }
});
```

This comment will likely go into clightd wiki :)

vansoest, 2018-12-11:
Thank you for the support! I’ll test it now with the polkit rule.
If it works, you can add it to the README(for special session users).

vansoest, 2018-12-12:
Fixed with the first Polkit rule.

commit 1927dbd6 (2018-12-01): TFD_TIMER_ABSTIME does not play well with suspended systems. Fixes #49.
closed 2018-12-12 by vansoest