FedeDP/Clight #50: Turn off auto-calibration temporarily?

bryango, 2018-12-04:
In older versions I can temporarily disable auto-calibration by:
```sh
systemctl --user stop clight.timer clight.service
```
In [`v3.0`](https://github.com/FedeDP/Clight/tree/3.0) there seems to be no graceful way to _pause_ auto-calibration except killing the `clight` process... Am I missing something? Perhaps a new way to terminate / restart `clight`? Many thanks!

FedeDP, 2018-12-04:
Hi!
I see 3 ways to achieve that:
* killall clight
* faking dimmed state through bus API: `busctl --user set-property org.clight.clight /org/clight/clight org.clight.clight Dimmed "b" true`
* Setting a very long captures timeout through bus API: https://github.com/FedeDP/Clight/wiki/Api#orgclightclightconftimeouts

Can I ask you what's your use-case for this? :) Thanks!

bryango, 2018-12-04:
Hi! Thank you so much for your quick reply! `killall` works fine, I was just wondering if it's unhealthy to just terminate the process. :stuck_out_tongue_winking_eye: 

Auto-calibration ususally suits me well, but sometimes I need to crank the brightness all the way up and maintain that high value for some period of time (e.g. when watching a movie with dark background). In this situation I have to temporarily disable clight so that it does not change the brightness against my will.

FedeDP, 2018-12-04:
I see! Yup `killall clight` is just fine as it will result in a clean exit :)
(`systemctl --user stop clight` did exactly the same!)

I see; I could add a method to suspend backlight auto calibration, may be, as it seems like a useful use-case; there is already a `NoAutoCalib` exposed (bus API) option but it is const and read-only. I may let user disable automatic calibration at runtime through it.

FedeDP, 2019-05-13:
Hi @bryango!
Just pinging you to let you know that this request should now be fulfilled in master!
There are 2 more actions to desktop file now: "Pause backlight calibration", and "Resume backlight calibration".

You can do that from bash/script, through clight bus api:
`busctl --user set-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf NoAutoCalib "b" true`
(true here will pause backlight; false will resume it).

bryango, 2019-05-13:
Awesome! Thank you very much!

closed 2019-05-13 by bryango