FedeDP/Clight #52: clight too much cpu usage

RezaSh13, 2018-12-07:
Hi, clight too much cpu usage (After turning on the computer)
When I close the program and run it again, the problem is resolved
The version of the package I installed:
clight-git r503.9463fcd-1
clightd-git r206.f6c5c46-1

OS: Manjaro Linux x86_64 
Kernel: 4.19.4-1-MANJARO 
Resolution: 1360x768 
DE: KDE 
WM: KWin 
CPU: Pentium E5700 (2) @ 3.000GHz 
GPU: Intel 4 Series Chipset 
Memory: 1249MiB / 3908MiB 

`  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                             
  789 neoman    20       21560   3488   3144 R  99.7   0.1   2:30.78 /bin/clight                                                                                         
  731 neoman    20     1376668 192780  99804 S   8.3   4.8   0:18.62 /bin/plasmashell                                                                                    
 1355 neoman    20      469332  65072  53852 S   3.3   1.6   0:00.67 /bin/konsole                                                                                        
  480 root      20      191164  38952  26808 S   3.0   1.0   0:04.87 /usr/lib/Xorg -nolisten tcp -auth /var/run/sddm/{739ddbac-c44c-4471-97e9-a5c4912a5775} -background+ 
  707 neoman    20     3013016  97764  68932 S   3.0   2.4   0:06.14 /bin/kwin_x11                                                                                       
  899 root      20                           I   2.0         0:02.28 [kworker/1:3-events]                                                                                
   43 root      20                           I   1.7         0:01.48 [kworker/u8:1-i915]                                                                                 
   22 root      -2                           S   0.3         0:00.61 [rcuc/1]                                                                                            
   23 root      20                           S   0.3         0:00.52 [ksoftirqd/1]                                                                                       
  957 neoman    20      361296  24240  19076 S   0.3   0.6   0:00.65 http.so [kdeinit5] https local:/run/user/1000/klauncherPYGUBJ.1.slave-socket local:/run/user/1000/+ 
 1378 neoman    20       15940   4384   3616 R   0.3   0.1   0:00.04 top   `

` neo  ~  clight
Clightd found, version: 3.0.
Setted DPMS timeouts: Standby -> 900s, Suspend -> 1200s, Off -> 1800s.
6500 gamma temp set.
Next gamma alarm due to: Fri Dec  7 18:30:00 2018
init_kbd_backlight(): No such interface “org.freedesktop.UPower.KbdBacklight” on object at path /org/freedesktop/UPower/KbdBacklight
Keyboard backlight calibration unsupported.
Ambient brightness: 0.414 -> Backlight pct: 0.616`

`####################
# Clight conf file #
####################

############
# TIMEOUTS #
############

## Timeouts between captures during day/night/event on AC
 ac_capture_timeouts = [ 60, 60, 60 ];

## Timeouts between captures during day/night/event on BATT
# batt_capture_timeouts = [ 1200, 5400, 600 ];

## Dimmer timeouts on AC/on BATT. 
## Set any of these to <= 0 to disable dimmer
## in the corresponding AC state.
# dimmer_timeouts = [ 45, 20 ];

## Timeouts for dpms on AC
## Set any of these to <= 0 to disable dpms on AC
# ac_dpms_timeouts = [ 900, 1200, 1800 ];

## Timeout for dpms on BATT
## Set any of these to <= 0 to disable dpms on BATT
# batt_dpms_timeouts = [ 300, 420, 600 ];

#########
# GAMMA #
#########

## Gamma temperature during day and night
# gamma_temp = [ 6500, 4000 ]

###################
# DISABLE MODULES #
###################

## Uncomment to disable brightness support
# no_brightness = 1;

## Uncomment to disable gamma support
 no_gamma = 1;

## Uncomment to disable dpms tool
# no_dpms = 1;

## Uncomment to disable org.freedesktop.PowerManagement.Inhibit 
## interface support
 no_inhibit = 1;

## Uncomment to disable dimmer tool
 no_dimmer = 1;

##################
# SMOOTH OPTIONS #
##################

## Uncomment to disable backlight smooth transitions
# no_smooth_backlight_transition = 1;

## Uncomment to disable gamma smooth transitions
# no_smooth_gamma_transition = 1;

## Uncomment to disable dimmer smooth transitions
# no_smooth_dimmer_transition = 1;

## Backlight transition step in percentage
# backlight_trans_step = 0.05;

## Backlight transition step
# gamma_trans_step = 50;

## Dimmer transition step in percentage
# dimmer_trans_step = 0.05;

## Backlight transition timeout in ms
# backlight_trans_timeout = 30;

## Backlight transition timeout in ms
# gamma_trans_timeout = 300;

## Dimmer transition timeout in ms
# dimmer_trans_timeout = 30;

####################
# BACKLIGHT CURVES #
####################

## Y points used to compute ambient_brightness while ON AC -> screen backlight curve best-fit parameters 
## through polynomial regression. X values are simply array's indexes (from 0 to 10 included).
 ac_brightness_regression_points = [ 0.0, 0.0, 0.20, 0.70, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0 ];
 #ac_brightness_regression_points = [ 0.0, 0.15, 0.29, 0.45, 0.61, 0.74, 0.81, 0.88, 0.93, 0.97, 1.0 ];
## Y points used to compute ambient_brightness while ON BATTERY -> screen backlight curve best-fit parameters 
## through polynomial regression. X values are simply array's indexes (from 0 to 10 included).
 #batt_brightness_regression_points = [ 0.0, 0.15, 0.23, 0.36, 0.52, 0.59, 0.65, 0.71, 0.75, 0.78, 0.80 ];

###########
# GENERIC #
###########

## Number of frames to be captured
 frames = 5;

## Your desired latitude for gamma support (surise/sunset in this location)
 latitude = 38.4;

## Your desired longitude for gamma support (surise/sunset in this location)
 longitude = 48.4;

## Video device to be used (only last part of syspath)
 video_devname = "video0";

## Screen syspath to be used (only last part of syspath)
# screen_sysname = "intel_backlight";

## Force set a sunrise time
 sunrise = "7:00";

## Force set a sunset time
 sunset = "19:00";

## Duration of an "event". Clight will enter "event" mode (more frequent screen recalibrations)
## from event_duration seconds before a sunrise/sunset, until event_duration seconds after.
# event_duration = 1800

## Change dimmer backlight level, in percentage
# dimmer_pct = 0.2;

## Verbose mode, useful in case of bugs:
## run clight in verbose mode,
## then open issue on github attaching log
 verbose = 1;
`

FedeDP, 2018-12-07:
Thanks for reporting your issue. It seems weird; 
Moreover, i see
> 6500 gamma temp set.
Next gamma alarm due to: Fri Dec  7 18:30:00 2018

But gamma should be disabled given your config, right?  
It seems clight is stuck on something...i will try to reproduce it here with your config.

RezaSh13, 2018-12-08:
> Btw it seems you are using an old config file (recently-release clight 3.0 updated some configuration keys); can you try once more after updating both clight and clightd to release 3.0 (there are now non-git packages on AUR) and using new configuration file?

I deleted clight-git r499.bdf2d17-1 and installed clight 3.0-1 and  also updated the configuration, But the problem did not resolve( When I close the program and run it again, the problem is resolved)

`####################
# Clight conf file #
####################

############
# TIMEOUTS #
############

## Timeouts between captures during day/night/event on AC
 ac_capture_timeouts = [ 60, 60, 60 ];

## Timeouts between captures during day/night/event on BATT
# batt_capture_timeouts = [ 1200, 5400, 600 ];

## Dimmer timeouts on AC/on BATT. 
## Set any of these to <= 0 to disable dimmer
## in the corresponding AC state.
# dimmer_timeouts = [ 45, 20 ];

## Timeouts for dpms on AC
## Set any of these to <= 0 to disable dpms on AC
# ac_dpms_timeouts = [ 900, 1200, 1800 ];

## Timeout for dpms on BATT
## Set any of these to <= 0 to disable dpms on BATT
# batt_dpms_timeouts = [ 300, 420, 600 ];

#########
# GAMMA #
#########

## Gamma temperature during day and night
# gamma_temp = [ 6500, 4000 ]

## Your desired latitude for gamma support (surise/sunset in this location)
# latitude = 40.9;

## Your desired longitude for gamma support (surise/sunset in this location)
# longitude = 7.16;

## Force set a sunrise time
# sunrise = "7:00";

## Force set a sunset time
# sunset = "19:00";

## Duration of an "event". Clight will enter "event" mode (more frequent screen recalibrations)
## from event_duration seconds before a sunrise/sunset, until event_duration seconds after.
# event_duration = 1800

###################
# DISABLE MODULES #
###################

## Uncomment to disable brightness support
# no_backlight = true;

## Uncomment to disable gamma support
 no_gamma = true;

## Uncomment to disable dpms tool
# no_dpms = true;

## Uncomment to disable org.freedesktop.PowerManagement.Inhibit 
## interface support
# no_inhibit = true;
[... 163 lines cut ...]
Set timeout of 60s 0ns on fd 7.
Average frames brightness: 0.000360.
Ambient brightness: 0.000 -> Backlight pct: 0.000
Set timeout of 60s 0ns on fd 7.
Average frames brightness: 0.006991.
Ambient brightness: 0.007 -> Backlight pct: 0.000
Set timeout of 60s 0ns on fd 7.
Average frames brightness: 0.000730.
Ambient brightness: 0.001 -> Backlight pct: 0.000
Set timeout of 60s 0ns on fd 7.
Average frames brightness: 0.000233.
Ambient brightness: 0.000 -> Backlight pct: 0.000
Set timeout of 60s 0ns on fd 7.
Average frames brightness: 0.001351.
Ambient brightness: 0.001 -> Backlight pct: 0.000
Set timeout of 60s 0ns on fd 7.
Average frames brightness: 0.016512.
Ambient brightness: 0.017 -> Backlight pct: 0.000
Set timeout of 60s 0ns on fd 7.
Average frames brightness: 0.010348.
Ambient brightness: 0.010 -> Backlight pct: 0.000
Set timeout of 60s 0ns on fd 7.
Average frames brightness: 0.064190.
Ambient brightness: 0.064 -> Backlight pct: 0.050
Set timeout of 60s 0ns on fd 7.`

FedeDP, 2018-12-08:
There is nothing wrong from your logs, except that Ambient brightness is always very low.
I can't see any sign of weird loops.
I will try to reproduce your issue.

RezaSh13, 2018-12-08:
> There is nothing wrong from your logs, except that Ambient brightness is always very low.
> I can't see any sign of weird loops.
> I will try to reproduce your issue.

Need more information?

FedeDP, 2018-12-10:
Great! I was able to reproduce it with your config. Thanks! Will work on it asap!

FedeDP, 2018-12-10:
It seems caused by `no_gamma = true;` setting. 
I still have to understand what's causing that CPU usage as Clight is not stuck in any loop (it responds on both bus API and external signals).

RezaSh13, 2018-12-18:
> Fixed in latest commit. Try a rebuild from master :)

Thanks for the new update