FedeDP/Clight #53: External monitor not dimming

glassdimly, 2018-12-08:
Thanks for this great utility!

I'm working to get my external monitor connected through an HDMI port on mu xubuntu installation to dim with clight, but it doesn't dim.

Here is the result of `clight --verbose`: 
```
clight  --verbose
SIGNAL module started.
BUS module started.
USERBUS module being started as submodule of BUS...
USERBUS module started.
INTERFACE module being started as submodule of USERBUS...
INTERFACE module started.
Trying to start BACKLIGHT module as its INTERFACE dependency was loaded...
Trying to start GAMMA module as its INTERFACE dependency was loaded...
Trying to start DIMMER module as its INTERFACE dependency was loaded...
XORG module started.
Trying to start GAMMA module as its XORG dependency was loaded...
Trying to start DIMMER module as its XORG dependency was loaded...
Trying to start DPMS module as its XORG dependency was loaded...
Trying to start LOCATION module as its BUS dependency was loaded...
Set timeout of 3s 0ns on fd 7.
LOCATION module started.
Trying to start UPOWER module as its BUS dependency was loaded...
UPOWER module started.
Trying to start BACKLIGHT module as its UPOWER dependency was loaded...
Trying to start DIMMER module as its UPOWER dependency was loaded...
Trying to start DPMS module as its UPOWER dependency was loaded...
Trying to start CLIGHTD module as its BUS dependency was loaded...
Clightd found, version: 3.0.
CLIGHTD module started.
Trying to start BACKLIGHT module as its CLIGHTD dependency was loaded...
Trying to start GAMMA module as its CLIGHTD dependency was loaded...
Trying to start DIMMER module as its CLIGHTD dependency was loaded...
Trying to start DPMS module as its CLIGHTD dependency was loaded...
Setted DPMS timeouts: Standby -> 300s, Suspend -> 420s, Off -> 600s.
DPMS module started.
Callback added for module DPMS on module UPOWER bus match.
Callback added for module DPMS on module INTERFACE bus match.
Trying to start INHIBIT module as its USERBUS dependency was loaded...
INHIBIT module started.
Trying to start DIMMER module as its INHIBIT dependency was loaded...
Loading loc from cache file: No such file or directory
Trying to start GAMMA module as its LOCATION dependency was loaded...
Set timeout of 0s 1ns on fd 8.
GAMMA module started.
Callback added for module GAMMA on module LOCATION bus match.
Callback added for module GAMMA on module INTERFACE bus match.
set_temp(): Unknown object '/org/clightd/clightd/Gamma'.
Next gamma alarm due to: Fri Dec  7 23:17:00 2018
Set timeout of 11891s 0ns on fd 8.
Trying to start BACKLIGHT module as its GAMMA dependency was loaded...
AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
Set timeout of 0s 1ns on fd 9.
BACKLIGHT module started.
Callback added for module BACKLIGHT on module UPOWER bus match.
Callback added for module BACKLIGHT on module INTERFACE bus match.
Callback added for module BACKLIGHT on module INTERFACE bus match.
Callback added for module BACKLIGHT on module INTERFACE bus match.
Average frames brightness: 0.065059.
Ambient brightness: 0.065 -> Backlight pct: 0.086
Set timeout of 5400s 0ns on fd 9.
Trying to start DIMMER module as its BACKLIGHT dependency was loaded...
idle_get_client(): Unknown object '/org/clightd/clightd/Idle'.
Clight Idle error.
Error while loading DIMMER module.
DIMMER module disabled.
INHIBIT module disabled.
INHIBIT module destroyed.
DIMMER module destroyed.

```

Running the default config. Using X11 rather than Wayland.

Is there anything obvious I should do or check? I've installed `ddcutil`. Thanks for your help!

Also, the Gamma doesn't seem to be working. Thoughts there?

FedeDP, 2018-12-08:
> set_temp(): Unknown object '/org/clightd/clightd/Gamma'.
> idle_get_client(): Unknown object '/org/clightd/clightd/Idle'.

It seems like Clightd was built without "GAMMA" and "IDLE" support. I think DDCUTIL support was disabled too.

How did you built clightd? Moreover, note that clightd by default enables all its functionalities, disabling them if the required libraries are not found. This is a weird behaviour that will be fixed once i merge [this PR](https://github.com/FedeDP/Clightd/pull/18).

I think that probably some requirements weren't met while building Clightd, and it silently disabled some functionalities.

glassdimly, 2018-12-09:
Hi @FedeDP,

Thanks so much for taking the time to answer my questions.

I have recompiled clightd with the dependencies, and now I'm getting:

```
clight --verbose
SIGNAL module started.
BUS module started.
USERBUS module being started as submodule of BUS...
USERBUS module started.
INTERFACE module being started as submodule of USERBUS...
INTERFACE module started.
Trying to start BACKLIGHT module as its INTERFACE dependency was loaded...
Trying to start GAMMA module as its INTERFACE dependency was loaded...
Trying to start DIMMER module as its INTERFACE dependency was loaded...
XORG module started.
Trying to start GAMMA module as its XORG dependency was loaded...
Trying to start DIMMER module as its XORG dependency was loaded...
Trying to start DPMS module as its XORG dependency was loaded...
Trying to start LOCATION module as its BUS dependency was loaded...
Set timeout of 3s 0ns on fd 7.
LOCATION module started.
Trying to start UPOWER module as its BUS dependency was loaded...
UPOWER module started.
Trying to start BACKLIGHT module as its UPOWER dependency was loaded...
Trying to start DIMMER module as its UPOWER dependency was loaded...
Trying to start DPMS module as its UPOWER dependency was loaded...
Trying to start CLIGHTD module as its BUS dependency was loaded...
Clightd found, version: 3.0.
CLIGHTD module started.
Trying to start BACKLIGHT module as its CLIGHTD dependency was loaded...
Trying to start GAMMA module as its CLIGHTD dependency was loaded...
Trying to start DIMMER module as its CLIGHTD dependency was loaded...
Trying to start DPMS module as its CLIGHTD dependency was loaded...
Setted DPMS timeouts: Standby -> 90s, Suspend -> 120s, Off -> 180s.
DPMS module started.
Callback added for module DPMS on module UPOWER bus match.
Callback added for module DPMS on module INTERFACE bus match.
Trying to start INHIBIT module as its USERBUS dependency was loaded...
INHIBIT module started.
Trying to start DIMMER module as its INHIBIT dependency was loaded...
New location received: 41.97, -87.66
Disarmed timerfd on fd 7.
Trying to start GAMMA module as its LOCATION dependency was loaded...
Set timeout of 0s 1ns on fd 8.
GAMMA module started.
Callback added for module GAMMA on module LOCATION bus match.
Callback added for module GAMMA on module INTERFACE bus match.
6500 gamma temp set.
Next gamma alarm due to: Sun Dec  9 15:47:00 2018
Set timeout of 10418s 0ns on fd 8.
Trying to start BACKLIGHT module as its GAMMA dependency was loaded...
AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
Set timeout of 0s 1ns on fd 9.
BACKLIGHT module started.
Callback added for module BACKLIGHT on module UPOWER bus match.
Callback added for module BACKLIGHT on module INTERFACE bus match.
Callback added for module BACKLIGHT on module INTERFACE bus match.
Callback added for module BACKLIGHT on module INTERFACE bus match.
Average frames brightness: 0.064810.
Ambient brightness: 0.065 -> Backlight pct: 0.096
Set timeout of 60s 0ns on fd 9.
Trying to start DIMMER module as its BACKLIGHT dependency was loaded...
DIMMER module started.
Callback added for module DIMMER on module UPOWER bus match.
Callback added for module DIMMER on module INHIBIT bus match.
Callback added for module DIMMER on module INTERFACE bus match.
Callback added for module DIMMER on module INTERFACE bus match.
```

The gamma and dimming are working great; but not on my external monitor on HDMI

```
$ which ddcutil 
/usr/bin/ddcutil

```

What steps should I take to enable brightness/gamma management on external monitor with ddutil?

Thanks in advance!

FedeDP, 2018-12-09:
> Thanks so much for taking the time to answer my questions.

You are welcome :)
Do you have an "i2c_clightd.conf" file in your `/usr/lib/modules-load.d/` folder?

glassdimly, 2018-12-10:
@FedeDP, I have no such file; should I have such a file? Where would a default exist? I don't see any in the repo or the docs.

How would I use this file to specify an external monitor?

FedeDP, 2018-12-10:
> I have no such file

This explains ddcutil issues :)
Well, clightd automatically installs it when DDC support is enabled at compile time: https://github.com/FedeDP/Clightd/blob/master/CMakeLists.txt#L118. 
Did you have ddcutil C library installed when you built clightd? 
Please note that Ubuntu repository-available `ddcutil` package does not carry its C library. You must build it from [git](https://github.com/rockowitz/ddcutil).

The C library is obviously used by Clightd to set external monitors backlight; that file is needed because it load required ddcutil module automatically at startup for you.

FedeDP, 2018-12-10:
I probably should make it clearer in Clightd wiki pages; will add it as soon as i find some spare time :)

glassdimly, 2018-12-13:
Alright, I [compiled ddcutil from source](http://www.ddcutil.com/building/), and I'm happy to say after following the instructions I found in ` /usr/lib/modules-load.d/i2c_clightd.conf`, my external monitor is now dimming!

Thanks, @FedeDP! Works great.

FedeDP, 2018-12-13:
You're welcome! Hopefully I get some spare time to merge the fixing PR in clightd soon enough :)

closed 2018-12-13 by FedeDP