FedeDP/Clight #55: clight make CPU hotest. :|

ANTISEC12, 2019-01-05:


ANTISEC12, 2019-01-05:
Repository      : AUR
Name            : clight
Version         : 3.1-2
Maintainer      : Nierro
URL             : https://github.com/FedeDP/Clight
AUR URL         : https://aur.archlinux.org/packages.php?ID=568716
License         : GPL
Votes           : 2
Popularity      : 1%
Installed       : Yes

OS: Arch Linux 
Kernel: x86_64 Linux 4.20.0-arch1-1-ARCH

~/.config/clight.conf
`####################
# Clight conf file #
####################

############
# TIMEOUTS #
############

## Timeouts between captures during day/night/event on AC
 ac_capture_timeouts = [ 60, 60, 60 ];

## Timeouts between captures during day/night/event on BATT
# batt_capture_timeouts = [ 1200, 5400, 600 ];

## Dimmer timeouts on AC/on BATT. 
## Set any of these to <= 0 to disable dimmer
## in the corresponding AC state.
# dimmer_timeouts = [ 45, 20 ];

## Timeouts for dpms on AC
## Set any of these to <= 0 to disable dpms on AC
# ac_dpms_timeouts = [ 900, 1200, 1800 ];

## Timeout for dpms on BATT
## Set any of these to <= 0 to disable dpms on BATT
# batt_dpms_timeouts = [ 300, 420, 600 ];

#########
# GAMMA #
#########

## Gamma temperature during day and night
# gamma_temp = [ 6500, 4000 ]

## Your desired latitude for gamma support (surise/sunset in this location)
# latitude = 40.9;

## Your desired longitude for gamma support (surise/sunset in this location)
# longitude = 7.16;

## Force set a sunrise time
# sunrise = "7:00";

## Force set a sunset time
# sunset = "19:00";

## Duration of an "event". Clight will enter "event" mode (more frequent screen recalibrations)
## from event_duration seconds before a sunrise/sunset, until event_duration seconds after.
# event_duration = 1800

###################
# DISABLE MODULES #
###################

## Uncomment to disable brightness support
# no_backlight = true;

## Uncomment to disable gamma support
 no_gamma = true;

## Uncomment to disable dpms tool
# no_dpms = true;

## Uncomment to disable org.freedesktop.PowerManagement.Inhibit 
## interface support
# no_inhibit = true;

## Uncomment to disable dimmer tool
 no_dimmer = true;

##################
# SMOOTH OPTIONS #
##################

## Uncomment to disable backlight smooth transitions
# no_smooth_backlight_transition = true;

## Uncomment to disable gamma smooth transitions
# no_smooth_gamma_transition = true;

## Uncomment to disable dimmer smooth transitions
# no_smooth_dimmer_transition = true;

## Backlight transition step in percentage
# backlight_trans_step = 0.05;

## Backlight transition step
# gamma_trans_step = 50;

## Dimmer transition step in percentage
# dimmer_trans_step = 0.05;

## Backlight transition timeout in ms
# backlight_trans_timeout = 30;

## Backlight transition timeout in ms
# gamma_trans_timeout = 300;

## Dimmer transition timeout in ms
# dimmer_trans_timeout = 30;

#############
# BACKLIGHT #
#############

## Y points used to compute ambient_brightness while ON AC -> screen backlight curve best-fit parameters 
## through polynomial regression. X values are simply array's indexes (from 0 to 10 included).
 ac_backlight_regression_points = [ 0.0, 0.0, 0.0, 0.20, 0.70, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0 ];

## Y points used to compute ambient_brightness while ON BATTERY -> screen backlight curve best-fit parameters 
## through polynomial regression. X values are simply array's indexes (from 0 to 10 included).
# batt_backlight_regression_points = [ 0.0, 0.15, 0.23, 0.36, 0.52, 0.59, 0.65, 0.71, 0.75, 0.78, 0.80 ];

## Number of frames or ALS device pollings to be captured
 captures = 5;

## Sensor device to be used (Webcam or ALS device)
 sensor_devname = "video0";

## Screen syspath to be use
 screen_sysname = "intel_backlight";

## Uncomment to disable keyboard backlight automatic calibration.
 #no_kdb_backlight = true;

## Set a threshold: if detected ambient brightness is below this threshold,
## capture will be discarded and no backlight change will be made.
# shutter_threshold = 0.10

##########
# DIMMER #
##########

## Change dimmer backlight level, in percentage
# dimmer_pct = 0.2;

###########
# GENERIC #
###########

## Verbose mode, useful in case of bugs:
## run clight in verbose mode,
## then open issue on github attaching log
 verbose = true;

## Disables automatic calibration for both screens backlight and Gamma.
## They must be manually triggered by bus api after setting this.
# no_auto_calibration = true;`

ANTISEC12, 2019-01-05:
> What do you mean by "hotest"?
> Are you sure this behavior does not depend on new 4.20.0 kernel?

>  Ie: is it using 100% CPU or is it just heating much more?
The hottest among the processes.

> Ie: is it using 100% CPU
 Yes, it's all about a core.

Sorry, Now it's running normally.
But ...  If the problem happened again... I report it.

ANTISEC12, 2019-01-05:
> What do you mean by "hotest"?
> Ie: is it using 100% CPU or is it just heating much more?
> Are you sure this behavior does not depend on new 4.20.0 kernel?

> What do you mean by "hotest"?
The hottest among the processes.

> Ie: is it using 100% CPU
Yes, it's all about a core.

Sorry, Now it's running normally.
But ...  If the problem happened again... I  will reported.

FedeDP, 2019-01-18:
I am closing it for now; feel free to open it if you need :)

closed 2019-01-18 by FedeDP

ANTISEC12, 2019-01-19:
ok, thanks alot 🌷🌷