FedeDP/Clight #56: Location not caching

yochananmarqos, 2019-01-07:
When booting my laptop when not connected to the internet, Clight does not remember the last location it acquired when online. I'm not sure if it's an issue with it not caching to begin with or not being able to load it from the cache.

I think it may only happen when there was no internet connection when I shutdown and no internet connection when booting up again.
```
(W)[08:16:04]{bus.c:308}	geoclue_get_client(): Failed to activate service 'org.freedesktop.GeoClue2': timed out (service_start_timeout=25000ms)
(W)[08:16:04]{location.c:107}	Geoclue2 appears to be unsupported.
(I)[08:16:04]{location.c:73}	Location 0.00 0.00 loaded from cache file!
(W)[08:16:04]{modules.c:89}	Error while loading LOCATION module.
```

Manjaro Linux Cinnamon
clight  3.1-2
clightd  3.1-1

FedeDP, 2019-01-07:
Can you try updating clight from git too (clight-git in aur)? I pushed a couple of small improvements in error handling in location.c that may improve your experience (or at least are very useful for debug purposes).

yochananmarqos, 2019-01-07:
When is `$HOME/.cache/clight` written to? Right now it's empty.

FedeDP, 2019-01-07:
Is is written when clight leaves (if any location was found). It should remain across reboots though (unless home cache directory gets cleaned by something else).

yochananmarqos, 2019-01-07:
When clight quits, does it write the location it acquired when it started when online, or does it query again before it quits? If there's no internet connection when it quits, then maybe that's why.

FedeDP, 2019-01-07:
It writes latest location it received by geoclue before quitting. I do not expect geoclue to pass us a 0.00:0.00 location when connection is missing though. Moreover, note that empty location is not cached (to avoid invalidating any other cached value by older clight runs).

yochananmarqos, 2019-01-18:
This morning when I turned my laptop on the gamma got stuck at 4000. I was connected to my phone hotspot last night, but turned it off right before I shutdown and I turned the hotspot on when I logged in, so it might have been offline at shutdown and startup. The cached location is wrong. I deleted `~/.cache/clight`, but after killing clight, restarting clightd & clight, it regenerates the cache file with the same incorrect location. Also the next gamma alarm is in the past. How do I force it to get a new location?

```
* Starting time:		Fri Jan 18 08:46:55 2019
I)[08:46:55]{clightd.c:48}	Clightd found, version: 3.1.
(I)[08:46:55]{dpms.c:64}	Setted DPMS timeouts: Standby -> 900s, Suspend -> 1200s, Off -> 1800s.
(I)[08:46:58]{location.c:74}	Location 39.90 -105.01 loaded from cache file!
(I)[08:46:58]{gamma.c:196}	4000 gamma temp set.
(I)[08:46:58]{gamma.c:79}	Next gamma alarm due to: Fri Jan 18 06:50:00 2019
```

FedeDP, 2019-01-18:
Hi!
The output you show me is loading your location from cache file. And that location is wrong, i guess.
The wrong location explains "next gamma alarm" being completely wrong for your real timezone.

How that location got cached is a real enigma. May be your phone hotspot is causing some weird issues? That should be an issue in geoclue though.

You can obviously stop location madness by forcing a location through config file; that is a possibility.
The fact is, if a location is cached it means that upon leaving clight is seeing that location. Clight receives locations by geoclue.

yochananmarqos, 2019-01-18:
Yeah, possibly geoclue was to blame. I'll save the latitude & longitude manually for now.

FedeDP, 2019-01-18:
I feel sorry for your bad luck with clight!

yochananmarqos, 2019-01-18:
Ha ha! Most of the time it works great, so no worries.

FedeDP, 2019-03-16:
@yochananmarqos Any news on this? Can i close it?

yochananmarqos, 2019-03-16:
I'm using GNOME now and use the built-in Night Light, so I don't have gamma enabled in Clight.

closed 2019-03-16 by yochananmarqos