FedeDP/Clight #59: [FeatureRequest?] Overwrite auto-exposure of webcam

bryango, 2019-01-22:
Recently I begin to notice that after I use the webcam, the clight-captured brightness changes significantly. I suspect that this behavior is caused by the automatic exposure. Steps to reproduce: 

1. Calibrate clight via dbus interface;
2. Open the webcam (using `vlc` or `ffmpeg`) and stream for a few seconds; you can see the video is automatically adjusting to the surroundings (ambient light);
3. Repeat step 1, and get a drastically different screen brightness.

Is it possible to overwrite this behavior? Thank you very much!

FedeDP, 2019-01-22:
Hi, thanks for this feature requeste!
As you can see [here](https://github.com/FedeDP/Clightd/blob/master/src/modules/sensors/camera.c#L121) Clightd is already attempting to disable some camera features (eg: auto white balance).
Reading [here](https://www.linuxtv.org/downloads/legacy/video4linux/API/V4L2_API/spec-single/v4l2.html), it seems like V4L2_CID_AUTOGAIN should be set for both automatic gain and exposure control. Clightd already set it where available.
May be we need to set V4L2_CID_CHROMA_AGC too?

bryango, 2019-01-22:
I will try that as soon as possible! :wink: However, it seems to me that this issue is not exactly caused by the auto-exposure itself; rather, it should be the result of _**some cached settings persisting**_ after using auto-exposure (by apps _other than_ clight). I guess that if clight _**reset a constant exposure**_ before any new captures, this problem could be solved. But I have no idea if that's possible. :laughing:

FedeDP, 2019-01-22:
>  I guess that if clight reset a constant exposure before any new captures, this problem could be solved

That would make sense :) But then again, Clight would change some settings (that eventually get cached). What happens if you then use a webcam application, eg: skype, that does not reset its own default values? 
We'd have clight that makes another application captures ugly.

bryango, 2019-01-22:
> I guess that if clight _**reset a constant exposure**_ before any new captures, this problem could be solved. But I have no idea if that's possible.

Yeah I agree, this is indeed a clumsy solution. :laughing:

mendelmunkis, 2019-02-11:
I've noticed that (at least on my laptop) auto-exposure tends to counteract the light that clight picks up from the screen itself

mendelmunkis, 2019-02-11:
On the second thought this should really be a new issue given as whats relevant to this one is just a minor workaround