FedeDP/Clight #61: Gamma location dependancy

mendelmunkis, 2019-02-10:
Is there any reason not to have an option to treat gamma like backlight (color temp based on ambient light)?

FedeDP, 2019-02-10:
Hi! Thanks for this request.

Honestly, i think nobody ever thought about that :)
I'd say that it can get very annoying to have your screen temperature continuously changing; I think (but i can be wrong obviously) that color temperature adjustments are only really useful to be "easy" for eyes during evening/night.

I'd like to hear some inputs about this feature request.

mendelmunkis, 2019-02-11:
I feel that the change is about as noticeable as the back-light brightness, and that having the temp go up at dawn if your in a room where the primary source of light is not a window is more annoying than semi-regular small adjustments. Of course in my opinion the best would probably be to change based  on ambient temp

FedeDP, 2019-02-11:
> I feel that the change is about as noticeable as the back-light brightness

Yes, this is true.

>  having the temp go up at dawn if your in a room where the primary source of light is not a window is more annoying than semi-regular small adjustments.

Well, as I said in my other comment, primary use case is to go easy on eyes before sleeping.

> Of course in my opinion the best would probably be to change based on ambient temp

I see.
I think we can provide both approaches; something like 2 gamma curves, one for DAY and one for NIGHT. This would provide same behaviour as now if any curve point is setted to same temp (ie: 6500 for DAY and 4000 for NIGHT), but would give use the ability to control gamma temperature more granularly.

mendelmunkis, 2019-02-11:
> > having the temp go up at dawn if your in a room where the primary source of light is not a window is more annoying than semi-regular small adjustments.
> 
> Well, as I said in my other comment, primary use case is to go easy on eyes before sleeping.
> 
I misunderstood you as meaning that primary use case is night/low-light conditions

FedeDP, 2019-02-11:
> I misunderstood you as meaning that primary use case is night/low-light conditions

It is, for BACKLIGHT; for GAMMA it only helps your eyes go into sleep-mode; but i like your idea. 
I can see two different curves for GAMMA for day and night being very useful, just like BACKLIGHT.

Something like -> DAY [ 5500, 7000 ], NIGHT [ 4000, 5500 ]. 

Again, I'd like to hear some more input on this though.

demoy, 2019-04-30:
I actually thought this was already a feature (I was hoping the config file would unlock it).

Prolong exposure to bluelight can damage the retina especially in older people (https://www.forbes.com/sites/fionamcmillan/2018/08/11/how-blue-light-damages-cells-in-your-eyes/#47bc47f5384b). The effects are accumulative (slow and incremental overtime without healing). People who spend all day in-front of computers could benefit from having the color temperature lowered to much ambient light.

Of course the caveat is the researchers shined bluelight directly into retina to test it's effects compared to other colors of light. Using electronics all day is not so extreme

FedeDP, 2019-04-30:
Hi @demoy ! Thanks for your input!
> People who spend all day in-front of computers could benefit from having the color temperature lowered to match ambient light.

I think this holds true; moreover this would be a Clight-unique feature compared to eg: redshift (i guess).

> I do agree it would be annoying to see the screen color change "for no reason". So it would have to be a feature people opt into: when you opt into a service and you see it working you don't find it annoying you find it great

Yes it would obviously be an option :)

FedeDP, 2019-04-30:
I'll leave an idea here: a new configuration mode for GAMMA where the two gamma temperatures (ie: DAY and NIGHT temps) would be used as boundaries; then screen temp would be selected after ambient brightness, eg: at 0.87, with { 6500, 4000 } temp values, we would set:
`0.87 * (6500 - 4000) + 4000 -> 6175` 

In this case, we won't need any location as screen temperature won't be "time-based" but instead would entirely relying upon ambient brightness.

What do you think? It seems simple enough to be developed for next release (unless any weird bug happens!)

demoy, 2019-04-30:
Took me awhile but I think I get it. It's also nice that you don't add mode specific variables alto that's probably just a nicety. (wondering mind) will you be able to set the backlight and gamma changes close enough together so the latter is less perceivable?

FedeDP, 2019-05-01:
Thanks to 1st may here in italy, i could quickly work on this idea. 
It is now in master; let me know if this works and feel free to close this in case :)
@mendelmunkis @demoy

demoy, 2019-05-01:
Cool it works. With the screen brightness changing at the same time I only noticed gamma because I looked out for it.

A thing I noticed (not sure if it's an issue):

New location received: 18.01, -76.80
Next gamma alarm due to: Wed May  1 17:57:00 2019

THESE show up after I restarted a couple times. The fact that they don't show up everytime made me wonder if they are unintentional

FedeDP, 2019-05-01:
Hi! Glad it works! :) I'm testing it and i'm start to find it a very nice feature too!

> New location received: xx, yy
> Next gamma alarm due to: Wed May 1 17:57:00 2019

This is ok; this means geoclue gave Clight a new location, thus Clight re-computes new Sunrise/Sunset for new location.
There is a check to verify that new location is at least 50km distant from last location before triggering an "update" for Sunrise/Sunset times; thus normally you'd only see "New location received..." message.

I will edit your previous post removing your current location!

mendelmunkis, 2019-05-01:
I wont have time to test for a few days but thank you for implementing this.

Sincerely,

Moshe Piekarski

--

There's no such thing as a stupid question,

But there are plenty of inquisitive idiots.

FedeDP, 2019-05-07:
I'm closing this one. Feel free to reopen it if any issue is found!

closed 2019-05-07 by FedeDP