FedeDP/Clight #62: screen light

mendelmunkis, 2019-02-11:
On laptops with poorly placed webcams the screen and keyboard back-lights can fool clight into believing that the ambient light level is higher than it is

mendelmunkis, 2019-02-11:
Sorry, i don,t know how to get debug info from clight, but if you tell me how I will try to attach demonstrative logs

FedeDP, 2019-02-11:
> On laptops with poorly placed webcams the screen and keyboard back-lights can fool clight into believing that the ambient light level is higher than it is.

I was aware of this issue; unfortunately i cannot find any simple way to filter away these light sources.
Have you got any idea?

mendelmunkis, 2019-02-11:
> Have you got any idea?
For a basic workaround maybe a config option to subtract some amount from the detected brightness.
Said amount being an advanced config option or possibly some percentage of screen brightness

FedeDP, 2019-02-11:
I'd prefer to find some corrective algorithm to automate it. 
Too many configuration options is confusing for end users (and i think Clight already provides lots of configurations).
If i cannot find such an algorithm i will probably think about this twice :)
If you're by any chance a developer and are willing to help, it would be great!

mendelmunkis, 2019-02-11:
> Too many configuration options is confusing for end users

The problem I see is that if this isn't an option it's begging for the inverse bug on computers with a better placed light sensor

>If you're by any chance a developer and are willing to help, it would be great!

I will help where I can, but my code tends not to be that great

>> Said amount being an advanced config option or possibly some percentage of screen brightness

>Percentage of screen brightness is not that bad though, and should work fine.

However the influence of the screen increases as the ambient light drops.
Best (but quite obviously impractical) would be to turn of the screen before sampling.

FedeDP, 2019-02-12:
> The problem I see is that if this isn't an option it's begging for the inverse bug on computers with a better placed light sensor

Yes, obviously.

> I will help where I can, but my code tends not to be that great

It would be great! I'll help you whenever you ask :)

> However the influence of the screen increases as the ambient light drops.
Best (but quite obviously impractical) would be to turn of the screen before sampling.

Yes. Moreover to correctly compute screen backlight compensations we need to know screen size too. This gets very complex. May be a small config option "backlight_pct_compensation" would help more while being simple (defaults to 0.0).

mendelmunkis, 2019-02-14:
> (defaults to 0.0).

Did you specifically want a float here? Is a percentage not granular enough?

FedeDP, 2019-02-14:
> Did you specifically want a float here? Is a percentage not granular enough? 

It would be a float used as percentage (ie: 0.1 -> 10%), as Clight always does for its config options.

mendelmunkis, 2019-02-14:
OK. Sorry, I should have read the config files more carefully.

FedeDP, 2019-02-14:
No problem! :)

mendelmunkis, 2019-02-14:
> Btw when screen backlight is low, keyboard backlight is high, thus we should compute it too when doing our correction.

Most cases I have seen either have keyboard light fixed or user adjusted not automatic. Given as it's not directly Clight controlled I think it makes sense to treat it as part of ambient

mendelmunkis, 2019-02-14:
Is there a particular order to the options in the config file?

FedeDP, 2019-02-14:
> Most cases I have seen either have keyboard light fixed or user adjusted not automatic. Given as it's not directly Clight controlled I think it makes sense to treat it as part of ambient

Well, indeed Clight does control keyboard backlight where it is supported (through UPower) :)

> Is there a particular order to the options in the config file?

Well, order is specified by headers (ie: TIMEOUTS, GAMMA, etc etc); there is no weighted order though.

mendelmunkis, 2019-02-14:
> Btw when screen backlight is low, keyboard backlight is high, thus we should compute it too when doing our correction

Is it better to have it as a separate option or have it calculated into backlight brightness?

FedeDP, 2019-02-15:
Btw, sorry if this sounds stupid...

> On laptops with poorly placed webcams the screen and keyboard back-lights can fool clight into believing that the ambient light level is higher than it is.

Can't you just change backlight curve to be more "conservative", ie: on higher ambient brightness, set lower backlight levels?

mendelmunkis, 2019-02-15:
> Can't you just change backlight curve to be more "conservative", ie: on higher ambient brightness, set lower backlight levels?

What happens in actual high ambient light?

FedeDP, 2019-02-15:
Well you can filter very high ambient light levels setting correct backlight level, something like:
`[ 0.0, 0.12, 0.24, 0.37, 0.51, 0.64, 0.70, 0.82, 0.93, 0.97, 1.0 ];`
Note how highest values are left unchanged from default values, while lowest values are slightly lowered to take into account screen + keyboard lights.

Btw even using a "backlight_pct_compensation" option would have same effect.

mendelmunkis, 2019-02-15:
While not perfect that does pretty much solve the issue.
It's a good thing that not everyone always sees the most complicated answer.

closed 2019-02-15 by mendelmunkis

FedeDP, 2019-02-15:
> While not perfect that does pretty much solve the issue.
> It's a good thing that not everyone always sees the most complicated answer.

I feel pretty stupid that it took nearly a whole week to come up with such a simple solution...lately i've been quite tired :)