FedeDP/Clight #64: Device or resource busy

yochananmarqos, 2019-02-13:
After updating `clight-git` to r524.c6c6af4-1 yesterday, it's not adjusting the brightness at startup. I'm using Manjaro GNOME now.

```
$ systemctl status clightd
● clightd.service - Bus service to manage various screen related properties (gamma, dpms, backlight)
   Loaded: loaded (/usr/lib/systemd/system/clightd.service; enabled; vendor preset: disabled)
   Active: active (running) since Wed 2019-02-13 07:58:13 MST; 1min 21s ago
 Main PID: 691 (clightd)
    Tasks: 1 (limit: 4915)
   Memory: 3.8M
   CGroup: /system.slice/clightd.service
           └─691 /usr/lib/clightd/clightd

Feb 13 07:58:13 paviliognome systemd[1]: Starting Bus service to manage various screen related properties (gamma, dpms, backlight)...
Feb 13 07:58:13 paviliognome systemd[1]: Started Bus service to manage various screen related properties (gamma, dpms, backlight).
Feb 13 07:58:52 paviliognome clightd[691]: setting V4L2_CID_AUTOGAIN: Invalid argument
Feb 13 07:58:53 paviliognome clightd[691]: (i2c_set_addr) Error in ioctl(I2C_SLAVE), errno=EBUSY(16): Device or resource busy
Feb 13 07:58:53 paviliognome clightd[691]: Registered Als sensor.
Feb 13 07:58:53 paviliognome clightd[691]: Registered Camera sensor.
Feb 13 07:58:53 paviliognome clightd[691]: [default]|DPMS|: New dpms timeouts:        Standby: 900s        Suspend: 1200s        Off:1800s.
Feb 13 07:58:53 paviliognome clightd[691]: [default]|SENSOR|: 5 frames captured by /dev/video0.
Feb 13 07:58:53 paviliognome clightd[691]: (i2c_set_addr) addr = 0x50. Returning EBUSY(-16): Device or resource busy
```

```
$ clight
Resource temporarily unavailable
A lock is present on /home/yochanan/.clight.log. Another clight instance running?
```
There's not much in the log:
```
(I)[07:58:52]{clightd.c:49}	Clightd found, version: 3.2.
(I)[07:58:52]{dpms.c:65}	Setted DPMS timeouts: Standby -> 900s, Suspend -> 1200s, Off -> 1800s.
(I)[07:58:54]{backlight.c:118}	Ambient brightness: 0.707 -> Backlight pct: 0.884
(I)[08:08:55]{backlight.c:118}	Ambient brightness: 0.809 -> Backlight pct: 0.941
(I)[08:12:56]{backlight.c:118}	Ambient brightness: 0.439 -> Backlight pct: 0.645
```

FedeDP, 2019-02-13:
Hi!
It seems clight is working fine. Clightd too is returning some sensible values for ambient brightness.

Are you using an external monitor?

yochananmarqos, 2019-02-13:
No, it's a laptop.

FedeDP, 2019-02-13:
Did you update clightd too? It seems the issue is in clightd:
> Feb 13 07:58:53 paviliognome clightd[691]: (i2c_set_addr) addr = 0x50. Returning EBUSY(-16): Device or resource busy

Moreover, did you alredy try to reboot?

yochananmarqos, 2019-02-13:
Yes, the info I posted was after rebooting. Maybe I jumped the gun reporting the issue as it does seem to be working.

I'm using the Mesa Radeon driver. Mesa was updated this week to 18.3.2 and the kernel to 4.20.7.

What device or resource is busy? Is it because it can't find an attached monitor?
```
$ journalctl -b | grep i2c
Feb 13 07:58:12 paviliognome kernel: i2c /dev entries driver
Feb 13 07:58:12 paviliognome systemd-modules-load[315]: Inserted module 'i2c_dev'
Feb 13 07:58:53 paviliognome clightd[691]: (i2c_set_addr) Error in ioctl(I2C_SLAVE), errno=EBUSY(16): Device or resource busy
Feb 13 07:58:53 paviliognome clightd[691]: (i2c_set_addr) addr = 0x50. Returning EBUSY(-16): Device or resource busy
```

```
$ sudo ddcutil detect --verbose
[sudo] password for yochanan: 
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

(i2c_set_addr) Error in ioctl(I2C_SLAVE), errno=EBUSY(16): Device or resource busy
(i2c_set_addr) addr = 0x50. Returning EBUSY(-16): Device or resource busy
No displays found
```
$ [sudo ddcutil environment --verbose](https://pastebin.com/wjTtGaUE)

FedeDP, 2019-02-13:
> What device or resource is busy? Is it because it can't find an attached monitor?

Ddcutils tries to find external monitors, but on some (buggy) monitors/video drivers it can cause some issues. 
It seems your issue is related to https://github.com/rockowitz/ddcutil/issues/26.

> Maybe I jumped the gun reporting the issue as it does seem to be working.
Is it working fine now?
Otherwise you can build clightd without DDCUTIL, as previously suggested. That should fix your issue.

yochananmarqos, 2019-02-13:
I built clightd without DDCUTIL and it made no diffference.
```
(I)[13:06:25]{clightd.c:49}	Clightd found, version: 3.2.
(I)[13:06:25]{dpms.c:65}	Setted DPMS timeouts: Standby -> 300s, Suspend -> 420s, Off -> 600s.
(W)[13:06:25]{bus.c:309}	capture_frames_brightness(): Device or resource busy
```

yochananmarqos, 2019-02-13:
I figured it out. I removed `acpi_backlight=vendor` from my grub kernel options and now everything seems to be working. I realized I didn't remember why I added it to begin with. It may have been an issue that was occurring with an earlier kernel that's no longer present.

closed 2019-02-13 by yochananmarqos

FedeDP, 2019-02-13:
It's great to hear it is now working fine! Hope you'll enjoy it!