FedeDP/Clight #65: Missing dependency on wiki

lig, 2019-02-18:
There is no mention of `dbus-devel` (or similar) package here https://github.com/FedeDP/Clight/wiki/Build

However, on Fedora 29 installing 3.1 release fails with the following message
```
CMake Error at CMakeLists.txt:76 (install):
  install FILES given no DESTINATION!
```

Installing `dbus-devel` fixes the issue.

FedeDP, 2019-02-18:
Hi, thanks for opening this issue.
Clight does require sd-bus library, and indeed it is stated in wiki: https://github.com/FedeDP/Clight/wiki/Build (libsystemd, first dep).

FedeDP, 2019-02-18:
Ah i see now; sorry i misread your issue. 
You are right, it is a build-dep indeed; thanks!

lig, 2019-02-18:
@FedeDP then it could be the way Fedora 29 packages are organized. I had `systemd/sd-bus.h` installed via `systemd-devel` package. However, `dbus-1` utility is not a part of this package. You need to install `dbus-devel` in order to avoid the error mentioned.

FedeDP, 2019-02-18:
It is fixed now; thank you very much for reporting this issue :)
https://github.com/FedeDP/Clight/wiki/Build

closed 2019-02-18 by lig