FedeDP/Clight #67: Trying to build under Fedora 29

brianread108, 2019-02-28:
cmake fails with 

```
-- Checking for modules 'popt;gsl;libconfig'
--   Package 'popt', required by 'virtual:world', not found
--   Package 'gsl', required by 'virtual:world', not found
--   Package 'libconfig', required by 'virtual:world', not found
CMake Error at /usr/share/cmake/Modules/FindPkgConfig.cmake:436 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPkgConfig.cmake:602 (_pkg_check_modules_internal)
  CMakeLists.txt:35 (pkg_check_modules)

```
Yet I do have all those modules:
```
#rpm -qa | grep -P '(popt|gsl|libconfig)'
libconfig-1.7.2-2.fc29.x86_64
popt-1.16-15.fc29.x86_64
gsl-2.4-7.fc29.x86_64

```

FedeDP, 2019-02-28:
Hi and thanks for the feedback!
Do fedora packages -devel files together with them? (ie: Ubuntu has splitted -devel packages).
And, what does 
```
pkg-config --libs libconfig gsl popt
```
output?

brianread108, 2019-02-28:
aha - I hadn;t quite understood the need for devel packages. Fedora package them separately and they are "<module>-devel" usually, ok, so I got them in now.
..And I get this now

```
-- Checking for modules 'popt;gsl;libconfig'
--   Found popt, version 1.16
--   Found gsl, version 2.4
--   Found libconfig, version 1.7.2
-- Checking for one of the modules 'libelogind;libsystemd>=221'
CMake Error at /usr/share/cmake/Modules/FindPkgConfig.cmake:659 (message):
  None of the required 'libelogind;libsystemd>=221' found
Call Stack (most recent call first):
  CMakeLists.txt:36 (pkg_search_module)

```

FedeDP, 2019-02-28:
You should also need systemd-devel package then :)

brianread108, 2019-02-28:
oh yes - ok, got it built ok.
Needed "sudo" on the make install.

Will try it out...thanks for the help.

FedeDP, 2019-02-28:
You're welcome! Hope you enjoy it!

brianread108, 2019-02-28:
Back already.
Here is my make install log:

```
build  sudo make install
[100%] Built target clight
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/bin/clight
-- Installing: /usr/share/dbus-1/services/org.clight.clight.service
-- Installing: /usr/local/etc/default/clight.conf
-- Installing: /etc/xdg/autostart/clight.desktop
-- Installing: /usr/share/applications/clightc.desktop
-- Installing: /usr/share/bash-completion/completions/clight
-- Installing: /usr/share/icons/hicolor/scalable/apps/clight.svg

```
It seems to have installed the global conf file in /usr/local/etc/default

Anyway I copied it to ~/.config and then ran clight:

```
 build  clight
check_clightd_version(): The name org.clightd.clightd was not provided by any .service files
No clightd found. Clightd is a mandatory dep.
No functional module is running. Nothing to do...leaving.

```
It asked me on screen to approve access to geoclue (I said yes)

FedeDP, 2019-02-28:
You need [Clightd](https://github.com/FedeDP/Clightd) too for Clight to work.
Please have a look at Clight [wiki pages](https://github.com/FedeDP/Clight/wiki/Build)!

brianread108, 2019-02-28:
Ok, I missed that, I also see I need libmodule as well. 
Probably be tomorrow now for getting those..

brianread108, 2019-02-28:
However:
Build of libmodule:

```
make
Scanning dependencies of target module
[ 12%] Building C object CMakeFiles/module.dir/Lib/map.c.o
[ 25%] Building C object CMakeFiles/module.dir/Lib/module.c.o
[ 37%] Building C object CMakeFiles/module.dir/Lib/modules.c.o
[ 50%] Building C object CMakeFiles/module.dir/Lib/priv.c.o
[ 62%] Building C object CMakeFiles/module.dir/Lib/pubsub.c.o
[ 75%] Building C object CMakeFiles/module.dir/Lib/stack.c.o
[ 87%] Building C object CMakeFiles/module.dir/Lib/poll_plugins/epoll_priv.c.o
[100%] Linking C shared library libmodule.so
[100%] Built target module
 brianr  …  Clight-master  libmodule-master  build  sudo make install
[sudo] password for brianr: 
[100%] Built target module
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/lib64/libmodule.so.4.0.0
-- Installing: /usr/local/lib64/libmodule.so.4
-- Installing: /usr/local/lib64/libmodule.so
-- Installing: /usr/local/include/module/map.h
-- Installing: /usr/local/include/module/module.h
-- Installing: /usr/local/include/module/module_cmn.h
-- Installing: /usr/local/include/module/module_easy.h
-- Installing: /usr/local/include/module/modules.h
-- Installing: /usr/local/include/module/modules_easy.h
-- Installing: /usr/local/include/module/stack.h
-- Installing: /usr/local/share/pkgconfig/libmodule.pc
-- Installing: /usr/local/share/licenses/libmodule/LICENSE

```
and then build of clightd

```
build  cmake -DENABLE_DDC=1 -DENABLE_GAMMA=1 -DENABLE_IDLE=1 -DENABLE_DPMS=1 ../
-- The C compiler identification is GNU 8.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.5.3") 
-- Checking for modules 'libudev;libmodule>=4.0.0'
--   Package 'libmodule', required by 'virtual:world', not found
CMake Error at /usr/share/cmake/Modules/FindPkgConfig.cmake:436 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPkgConfig.cmake:602 (_pkg_check_modules_internal)
  CMakeLists.txt:26 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/home/brianr/Documents/Clight-master/Clightd-master/build/CMakeFiles/CMakeOutput.log".

```
Clearly it is not finding the libmodule that I've built..

FedeDP, 2019-02-28:
It seems you're building/installing clight/clightd/libmodule in /usr/local.
Pkg-config may not be configured to look into it.
You have 2 options: either passing 
`-DCMAKE_INSTALL_PREFIX=/usr`
to cmake (clean up build folder before doing so), or setting PKG_CONFIG_PATH variable as needed, ie: 
`export PKG_CONFIG_PATH=/usr/local/share/pkgconfig`

brianread108, 2019-03-01:
Still not running.
```

build  clight
check_clightd_version(): The name org.clightd.clightd was not provided by any .service files
No clightd found. Clightd is a mandatory dep.
No functional module is running. Nothing to do...leaving.
 brianr  …  Clight-master  libmodule-master  build  

```

```
 build  sudo make install
[100%] Built target clight
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/bin/clight
-- Installing: /usr/share/dbus-1/services/org.clight.clight.service
-- Up-to-date: /usr/local/etc/default/clight.conf
-- Up-to-date: /etc/xdg/autostart/clight.desktop
-- Up-to-date: /usr/share/applications/clightc.desktop
-- Up-to-date: /usr/share/bash-completion/completions/clight
-- Up-to-date: /usr/share/icons/hicolor/scalable/apps/clight.svg

```
```

build  cmake -DCMAKE_INSTALL_PREFIX=/usr -DENABLE_DDC=1 -DENABLE_GAMMA=1 -DENABLE_IDLE=1 -DENABLE_DPMS=1 ../
-- The C compiler identification is GNU 8.2.1
-- The CXX compiler identification is GNU 8.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building Libmodule 4.0.0-Release for Linux.
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    ENABLE_DDC
    ENABLE_DPMS
    ENABLE_GAMMA
    ENABLE_IDLE

-- Build files have been written to: /home/brianr/Documents/Clight-master/libmodule-master/build

```
```

build  sudo make install
[100%] Built target module
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/lib64/libmodule.so.4.0.0
-- Up-to-date: /usr/lib64/libmodule.so.4
-- Up-to-date: /usr/lib64/libmodule.so
-- Up-to-date: /usr/include/module/map.h
-- Up-to-date: /usr/include/module/module.h
-- Up-to-date: /usr/include/module/module_cmn.h
-- Up-to-date: /usr/include/module/module_easy.h
-- Up-to-date: /usr/include/module/modules.h
-- Up-to-date: /usr/include/module/modules_easy.h
-- Up-to-date: /usr/include/module/stack.h
-- Installing: /usr/share/pkgconfig/libmodule.pc
-- Up-to-date: /usr/share/licenses/libmodule/LICENSE

```

brianread108, 2019-03-01:
I also tried 

cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DENABLE_DDC=1 -DENABLE_GAMMA=1 -DENABLE_IDLE=1 -DENABLE_DPMS=1 ../

FedeDP, 2019-03-01:
> CMake Warning:
   Manually-specified variables were not used by the project:
    ENABLE_DDC
    ENABLE_DPMS
    ENABLE_GAMMA
    ENABLE_IDLE

First issue..what version of clightd are you building?

Second: did you clean up build folder for each project before running cmake again?

Third: in each project's tree you can find an Arch/PKGBUILD file. Even if it is a build script for Archlinux, you can check its build() function and use same commands! eg, for clight: https://github.com/FedeDP/Clight/blob/master/Arch/PKGBUILD#L33.

brianread108, 2019-03-01:
Version of clight was pulled down from the master branch on github yesterday.

I did rm-fR * on the build folder before starting.

Ok, I'll check those files.

brianread108, 2019-03-01:
Still chasing down requirerments.
```
-- Checking for modules 'x11;xscrnsaver'
--   Package 'xscrnsaver', required by 'virtual:world', not found
CMake Error at /usr/share/cmake/Modules/FindPkgConfig.cmake:436 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPkgConfig.cmake:602 (_pkg_check_modules_internal)
  CMakeLists.txt:51 (pkg_check_modules)
  CMakeLists.txt:72 (optional_dep)

However I do have:

Package libXScrnSaver-1.2.3-2.fc29.x86_64 is already installed.

```

FedeDP, 2019-03-01:
Is libXScrnSaver-devel installed?

brianread108, 2019-03-01:
Sorry - not thinking well today - it wasn't. 

I'll let you know how I get on..

brianread108, 2019-03-04:
I am now having to compile ddcutil as there does not seem to be a pre-built version for fedora 29 (it stops at 28).
The build seems to go fine, but the subsequent build of clightd fails to find it.
Am attaching the logs.  Do I need to build a devel version? If so how?
[clightd.log](https://github.com/FedeDP/Clight/files/2927781/clightd.log)
[ddcutil.log](https://github.com/FedeDP/Clight/files/2927782/ddcutil.log)

FedeDP, 2019-03-04:
It seems like ddcutil was installed in /usr/local, am I right?
Try to build it in /usr (like you previously did for clight):
```
./autogen.sh
./configure --prefix=/usr
make
```

brianread108, 2019-03-05:
Ok,m tried that:

	ddcutil 0.9.5
	=============

	libtool version  	0:0:0

	prefix:                 /usr
	bindir:                 ${exec_prefix}/bin
	datarootdir:            ${prefix}/share
	datadir:                ${datarootdir}
	docdir:                 ${datarootdir}/doc/${PACKAGE_TARNAME}
	mandir:               	${datarootdir}/man

	adl_header_dir:         
	enable_lib:             yes
	enable_swig:            no
	enable_cython:          no
	enable_cffi:            no
	enable_gobject          no
	enable_usb:             yes
	enable_drm:             yes
	enable_x11:             yes
	enable_doxygen:         no
	enable_failsim:         no
	enable_use_api:         no
	include_testcases:      no

	compiler:               gcc
	cflags:                 -g -O2
	ldflags:                

However clightd build still fails to find it.

FedeDP, 2019-03-06:
That's weird. What does `pkg-config --libs ddcutil` say?

brianread108, 2019-03-08:
```
```
"EDIT: obviously you did install it too?" - "make install" you mean?

aha - after "make install" I get this:

```
pkg-config --libs ddcutil
-lddcutil -lglib-2.0 -lXrandr -lX11 

```
and the make for clightd seems to work.

"service clightd start" now works!

And got some results from "clight".

```
Clightd found, version: 3.2.
Setted DPMS timeouts: Standby -> 900s, Suspend -> 1200s, Off -> 1800s.
inhibit_check(): The name org.freedesktop.PowerManagement.Inhibit was not provided by any .service files
PowerManagement inhibition appears to be unsupported.
Error loading location from cache file.
init_kbd_backlight(): No such interface “org.freedesktop.UPower.KbdBacklight” on object at path /org/freedesktop/UPower/KbdBacklight
Keyboard backlight calibration unsupported.
Ambient brightness: 0.377 -> Backlight pct: 0.571
```

brianread108, 2019-03-08:
Here is the shell script I developed to finally succesfully build the packages under Fedora 29:
```
`#!/bin/sh
#Libmodule
cd /home/brianr/Documents
rm -fR clight 
mkdir clight
cd clight
echo "Getting from git `pwd`"
git clone https://github.com/FedeDP/Clight.git
git clone https://github.com/FedeDP/Clightd.git
git clone https://github.com/FedeDP/libmodule.git
git clone https://github.com/rockowitz/ddcutil.git
#
cd ddcutil
echo "Building ddcutil `pwd`"
./autogen.sh
./configure --prefix=/usr
make
make install
cd ..
#
cd libmodule
echo "Building Libmodule `pwd`"
rm -fR build
mkdir build
cd build
cmake ../
make
make install
cd ../..
#
cd Clightd
echo "Building clightd `pwd`"
rm -fR build
mkdir build
cd build
cmake \
        -G "Unix Makefiles" \
        -DCMAKE_INSTALL_PREFIX=/usr \
        -DCMAKE_INSTALL_LIBDIR=lib \
        -DCMAKE_BUILD_TYPE="Release" \
        -DENABLE_DDC=1 -DENABLE_GAMMA=1 -DENABLE_IDLE=1 -DENABLE_DPMS=1 \
        ..
make
make install
cd ../..
cd Clight
#
echo "Building clight `pwd`"
rm -fR build
mkdir build
cd build
cmake \
        -G "Unix Makefiles" \
        -DCMAKE_INSTALL_PREFIX=/usr \
        -DCMAKE_INSTALL_LIBDIR=lib \
        -DCMAKE_BUILD_TYPE="Release" \
        ..
make
make install`
```

FedeDP, 2019-03-08:
> Here is the shell script I developed to finally succesfully build the packages under Fedora 29:

That's great, thanks! I will probably add it to wiki, if you don't mind :)

Btw I am trying to directly add support for creating DEB/RPM packages inside cmake.
This way you can even install them as normal packages instead of "make install", thus tracking them more easily.

brianread108, 2019-03-08:
> That's great, thanks! I will probably add it to wiki, if you don't mind :)

Delighted to help.

FedeDP, 2019-03-08:
https://github.com/FedeDP/Clight/wiki/Build#dealing-with-dependencies
Again, thanks!

closed 2019-03-08 by FedeDP