FedeDP/Clight #70: Build fails due to `floating constant in preprocessor expression`

hyiltiz, 2019-03-14:
In Debian testing. Installed all dependencies, and passed `cmake ..`. But now when I do `make`, I am getting
```BASH
[  4%] Building C object CMakeFiles/clight.dir/src/modules/bus.c.o
/home/hyiltiz/src/Clight/src/modules/bus.c: In function ‘call’:
<command-line>: error: floating constant in preprocessor expression
/home/hyiltiz/src/Clight/src/modules/bus.c:103:5: note: in expansion of macro ‘LIBSYSTEMD_VERSION’
 #if LIBSYSTEMD_VERSION >= 234
     ^~~~~~~~~~~~~~~~~~
/home/hyiltiz/src/Clight/src/modules/bus.c: In function ‘add_match’:
<command-line>: error: floating constant in preprocessor expression
/home/hyiltiz/src/Clight/src/modules/bus.c:203:5: note: in expansion of macro ‘LIBSYSTEMD_VERSION’
 #if LIBSYSTEMD_VERSION >= 237
     ^~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/clight.dir/build.make:128: CMakeFiles/clight.dir/src/modules/bus.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/clight.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

FedeDP, 2019-03-14:
Hi! Thanks for opening this bug report!

What does `pkg-config --modversion libsystemd` print for you?

hyiltiz, 2019-03-18:
```
$ pkg-config --modversion libsystemd
Package libsystemd was not found in the pkg-config search path.
Perhaps you should add the directory containing `libsystemd.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libsystemd' found
```
Huh, I did not have systemd dev files? Let's install all dependencies first...

```BASH
~/s/C/build (master|✔) $ make clean
⏎                                                                                                                          

~/s/C/build (master|✔) $ cmake ..
-- Configuring done
-- Generating done
-- Build files have been written to: /home/hyiltiz/src/Clight/build
⏎                                                                                                                          

~/s/C/build (master|✔) $ make
[  4%] Building C object CMakeFiles/clight.dir/src/conf/config.c.o
[  9%] Building C object CMakeFiles/clight.dir/src/conf/opts.c.o
[ 14%] Building C object CMakeFiles/clight.dir/src/main.c.o
[ 19%] Building C object CMakeFiles/clight.dir/src/modules.c.o
[ 23%] Building C object CMakeFiles/clight.dir/src/modules/backlight.c.o
[ 28%] Building C object CMakeFiles/clight.dir/src/modules/bus.c.o
/home/hyiltiz/src/Clight/src/modules/bus.c: In function ‘call’:
<command-line>: error: floating constant in preprocessor expression
/home/hyiltiz/src/Clight/src/modules/bus.c:103:5: note: in expansion of macro ‘LIBSYSTEMD_VERSION’
 #if LIBSYSTEMD_VERSION >= 234
     ^~~~~~~~~~~~~~~~~~
/home/hyiltiz/src/Clight/src/modules/bus.c: In function ‘add_match’:
<command-line>: error: floating constant in preprocessor expression
/home/hyiltiz/src/Clight/src/modules/bus.c:203:5: note: in expansion of macro ‘LIBSYSTEMD_VERSION’
 #if LIBSYSTEMD_VERSION >= 237
     ^~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/clight.dir/build.make:128: CMakeFiles/clight.dir/src/modules/bus.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/clight.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
⏎                       
                                                                                                   
~/s/C/build (master|✔) $ pkg-config --modversion libsystemd
241
```

FedeDP, 2019-03-18:
Hi!
It seems like your compiler thinks that 241 is a float...
Can you try to completely wipe build folder and then running cmake again?
What's the output?

hyiltiz, 2019-03-18:
Thanks for the prompt responses. Here you go:

```BASH
~/s/Clight (master|✔) $ rm -rf build/
⏎                                                                                                                          

~/s/Clight (master|✔) $mkdir build
⏎                                                                                                                          

~/s/Clight (master|✔) $ cd build/
⏎                                                                                                                          

~/s/C/build (master|✔) $ cmake ..
-- The C compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for modules 'popt;gsl;libconfig'
--   Found popt, version 1.16
--   Found gsl, version 2.5
--   Found libconfig, version 1.5
-- Checking for one of the modules 'libelogind;libsystemd>=221'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/hyiltiz/src/Clight/build
⏎                                                                                                                          

~/s/C/build (master|✔) $ make
Scanning dependencies of target clight
[  4%] Building C object CMakeFiles/clight.dir/src/conf/config.c.o
[  9%] Building C object CMakeFiles/clight.dir/src/conf/opts.c.o
[ 14%] Building C object CMakeFiles/clight.dir/src/main.c.o
[ 19%] Building C object CMakeFiles/clight.dir/src/modules.c.o
[ 23%] Building C object CMakeFiles/clight.dir/src/modules/backlight.c.o
[ 28%] Building C object CMakeFiles/clight.dir/src/modules/bus.c.o
/home/hyiltiz/src/Clight/src/modules/bus.c: In function ‘call’:
<command-line>: error: floating constant in preprocessor expression
/home/hyiltiz/src/Clight/src/modules/bus.c:103:5: note: in expansion of macro ‘LIBSYSTEMD_VERSION’
 #if LIBSYSTEMD_VERSION >= 234
     ^~~~~~~~~~~~~~~~~~
/home/hyiltiz/src/Clight/src/modules/bus.c: In function ‘add_match’:
<command-line>: error: floating constant in preprocessor expression
/home/hyiltiz/src/Clight/src/modules/bus.c:203:5: note: in expansion of macro ‘LIBSYSTEMD_VERSION’
 #if LIBSYSTEMD_VERSION >= 237
     ^~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/clight.dir/build.make:128: CMakeFiles/clight.dir/src/modules/bus.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/clight.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
⏎                                                                                                                          

~/s/C/build (master|✔) $ gcc --version
gcc (Debian 8.3.0-2) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

⏎                                                                                                                          

~/s/C/build (master|✔) $ clang --version
clang version 7.0.1-8 (tags/RELEASE_701/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
⏎                                                                                                                          

~/s/C/build (master|✔) $ cc --version
[... 34 lines cut ...]
Scanning dependencies of target clight
[  4%] Building C object CMakeFiles/clight.dir/src/conf/config.c.o
[  9%] Building C object CMakeFiles/clight.dir/src/conf/opts.c.o
[ 14%] Building C object CMakeFiles/clight.dir/src/main.c.o
[ 19%] Building C object CMakeFiles/clight.dir/src/modules.c.o
[ 23%] Building C object CMakeFiles/clight.dir/src/modules/backlight.c.o
[ 28%] Building C object CMakeFiles/clight.dir/src/modules/bus.c.o
/home/hyiltiz/src/Clight/src/modules/bus.c:103:5: error: floating point literal in preprocessor expression
#if LIBSYSTEMD_VERSION >= 234
    ^
<command line>:2:28: note: expanded from here
#define LIBSYSTEMD_VERSION 239.3
                           ^
/home/hyiltiz/src/Clight/src/modules/bus.c:203:5: error: floating point literal in preprocessor expression                  
#if LIBSYSTEMD_VERSION >= 237
    ^
<command line>:2:28: note: expanded from here                                                                               
#define LIBSYSTEMD_VERSION 239.3
                           ^
2 errors generated.                                                                                                         
make[2]: *** [CMakeFiles/clight.dir/build.make:128: CMakeFiles/clight.dir/src/modules/bus.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/clight.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

```

FedeDP, 2019-03-18:
Before:

>  pkg-config --modversion libsystemd
241

Now:
> <command line>:2:28: note: expanded from here                                                                               
#define LIBSYSTEMD_VERSION 239.3

I guess it is not using libsystemd-dev but it is using libelogind-dev. Libelogind-dev is a standalone reimplementation of libsystemd-dev functionalities, that does not depend upon systemd.

Can you paste output of: `pkg-config --modversion libelogind`? Thanks!

The other (more obvious) issue is that Clight's cmake does not support "float" versioning very well.
I will have a look into this. But i guess that using libsystemd would let you build clight.

FedeDP, 2019-03-18:
It should be fixed after my latest commit :) Let me know, thanks!

hyiltiz, 2019-03-19:
After a pull, cleanup and cmake, `make` says (for both gcc and clang):
```BASH
make
Scanning dependencies of target clight
[  4%] Building C object CMakeFiles/clight.dir/src/conf/config.c.o
In file included from /usr/include/time.h:25,
                 from /media/hdd/home/hyiltiz/src/Clight/src/commons.h:1,
                 from /media/hdd/home/hyiltiz/src/Clight/src/utils/log.h:3,
                 from /media/hdd/home/hyiltiz/src/Clight/src/conf/config.h:1,
                 from /media/hdd/home/hyiltiz/src/Clight/src/conf/config.c:1:
/usr/include/features.h:424:12: fatal error: sys/cdefs.h: No such file or directory
 #  include <sys/cdefs.h>
            ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/clight.dir/build.make:63: CMakeFiles/clight.dir/src/conf/config.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/clight.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

```

hyiltiz, 2019-03-19:
`apt install --reinstall libc6 libc6-dev` fixed it. Now it builds successfully. Might want to include `libc6-dev` into the dependency list.

closed 2019-03-19 by hyiltiz

FedeDP, 2019-03-19:
Great to hear that! Thanks!

commit 8491e18e (2019-03-18): Avoid flaot libsystemd/libelogind version. Fixes #70.