FedeDP/Clight #71: Brightness after dim always 100%

Alkotronikk, 2019-03-19:
Currently the clight always restores the dim state to 100% regardless the backlight level the screen had previously. Similarly if I set my backlight below the dim level, the brightness level raises up to the dim state level.

Would it be possible to store the pre-dim brightness level and then to restore it after dim? And also would be possible to not dim a screen at all when the current brightness level is below or equal the current brightness level ? Cheers.

FedeDP, 2019-03-20:
> Currently the clight always restores the dim state to 100% regardless the backlight level the screen had previously

This should not be true; in fact it restores previous backlight, as you can see here: https://github.com/FedeDP/Clight/blob/master/src/modules/dimmer.c#L152. Can you attach your log file?

> Similarly if I set my backlight below the dim level, the brightness level raises up to the dim state level.

This should not be true too: https://github.com/FedeDP/Clight/blob/master/src/modules/dimmer.c#L170

Remember, though, that clight does not support external backlight modifications; ie if by "if I set my backlight below the dim level" you mean that  you're manually changing backlight, unfortunately clight doesn't know anything about that.

This is a kind of defect that was never really faced.

Alkotronikk, 2019-03-20:
Oh I see, though I'm not a C++ programmer so I'm unsure how the actual value is read. I'm using the [xorg-backlight](https://www.archlinux.org/packages/extra/x86_64/xorg-xbacklight/) package to set my backlight manually by mapping it to media keys on my laptop.  

Here are my _intel_backlight_ values:
`/sys/class/backlight/intel_backlight/max_brightness` is `937`
`/sys/class/backlight/intel_backlight/actual_brightness` is `843` (90%)

And after dim state, the `actual_brightness` is set to `937`.

Here's my [clight.log](https://github.com/FedeDP/Clight/files/2987319/clight.log). The brightness module is set as disabled since I didn't realy utilize the webcam functionality.

FedeDP, 2019-03-20:
> The brightness module is set as disabled since I didn't realy utilize the webcam functionality.

Ok! Then you fall right into the defect aforementioned: clight does not support external tools changing its backlight; when backlight module is disabled, dimmer set 100% screen backlight while starting and uses that as a "freezed" value.
Thus the issue you're encountering.

> This is a kind of defect that was never really faced.

Again, this holds true unfortunately.

FedeDP, 2019-03-20:
I removed your clight.log link in previous comment as it contained your current position.

Alkotronikk, 2019-03-20:
Alright, thanks for the log removal and I'm sorry for bothering you. I guess that this is not the issue with your app, so I'm gonna close this. 

I misunderstood what this app does, I guess I thought that when the app uses _intel_backlight_ for setting the brightness it would also read the current value before dimming and not relying on the internal storing only.

closed 2019-03-20 by Alkotronikk

FedeDP, 2019-03-20:
> I misunderstood what this app does, I guess I thought that when the app uses intel_backlight for setting the brightness it would also read the current value before dimming and not relying on the internal storing only.

It used to do that indeed! And i think it was better back then. 
The feature was dropped when external monitor support appeared, as it became quite a huge mess to get current backlight level for each monitor, dimming them and then, on Undimming, resuming old backlight. Moreover, with Clightd < 3.0 dimmer was managed from within Clight, thus it was impossible(read "very hard") to have different dimming timeouts (i mean, if you have 2 monitors, one at 86% backlight, the other at 56% backlight, and you want to dim to 20%, you have 2 different timers).

Indeed it should be now simplier; i may give this a shot after #43 :)

FedeDP, 2019-03-20:
And thanks for your time btw!

Alkotronikk, 2019-03-20:
No problem. Feel free to reopen to issue if you plan to implement this.