FedeDP/Clight #73: Under Fedora 29 - Still not runnjing in background from startup

brianread108, 2019-03-21:
The logs show that the intial startup fails as follows;

```
(W)[06:43:05]{bus.c:309}	check_clightd_version(): Connection timed out
(W)[06:43:05]{clightd.c:42}	No clightd found. Clightd is a mandatory dep.
(E)[06:43:05]{modules.c:242}	No functional module is running. Nothing to do...leaving.

```
This shows it (and clightd) running:
```

 brianr  ~  ps ax | grep clight
 2208 ?        S      0:00 clight
 2833 ?        Ss     0:00 /usr/lib/clightd/clightd
 2971 pts/0    S+     0:00 grep --color=auto clight
 brianr  ~  ps ax | grep clight
 2833 ?        Ss     0:00 /usr/lib/clightd/clightd
 3079 pts/0    S+     0:00 grep --color=auto clight
 brianr  ~  

```
Maybe a timing problem?

I get a dialog to allow access to geolocation.  - is that effecting the timing of things?

If I then run clight from the command line, things seem to work:

```
 brianr  ~  clight
Clightd found, version: 3.4.
Setted DPMS timeouts: Standby -> 900s, Suspend -> 1200s, Off -> 1800s.
inhibit_check(): The name org.freedesktop.PowerManagement.Inhibit was not provided by any .service files
PowerManagement inhibition appears to be unsupported.
Location 53.42 -3.00 loaded from cache file!
6500 gamma temp set.
Next gamma alarm due to: Thu Mar 21 17:52:00 2019
init_kbd_backlight(): No such interface “org.freedesktop.UPower.KbdBacklight” on object at path /org/freedesktop/UPower/KbdBacklight
Keyboard backlight calibration unsupported.
Ambient brightness: 0.218 -> Backlight pct: 0.350

```

FedeDP, 2019-03-21:
Hi and thanks for taking the time to open the issue!

> check_clightd_version(): Connection timed out

It seems that clight cannot find clightd during startup.

> I get a dialog to allow access to geolocation. - is that effecting the timing of things?

It can be; can you try to follow my [geoclue issues wiki page](https://github.com/FedeDP/Clight/wiki/Testing#geoclue-issues)?

I will remove your location from previous comment (clight.log) in the meantime!

brianread108, 2019-03-21:
Added that to the geoclue.conf, but still get this:
```

`Clight
* Software version:		3.1
* Starting time:		Thu Mar 21 11:25:05 2019

Starting options:

### Generic ###
* Verbose (debugging):		Disabled
* Automatic calibration:		Enabled
* Keyboard backlight:		Enabled
* Number of captures:		5
* Webcam device:		Unset
* Backlight path:		Unset
* Shutter threshold:		0.00
* User position:		Unset
* Daily screen temp:		6500
* Nightly screen temp:		4000
* User set sunrise:		Unset
* User set sunset:		Unset
* Event duration:		1800
* Dimmer backlight pct:		0.20

### Timeouts ###
* Daily timeouts:		AC 600	BATT 1200
* Nightly timeout:		AC 2700	BATT 5400
* Event timeouts:		AC 300	BATT 600
* Dimmer timeouts:		AC 45	BATT 20
* Dpms timeouts:		AC 900:1200:1800	BATT 300:420:600

### Modules ###
* BACKLIGHT:		Enabled
* GAMMA:		Enabled
* DIMMER:		Enabled
* DPMS:		Enabled

### Smooth ###
* Bright smooth trans:		Enabled
* Gamma smooth trans:		Enabled
* Dimmer smooth trans:		Enabled
* Bright smooth steps:		0.05
* Gamma smooth steps:		50
* Dimmer smooth steps:		0.05
* Bright smooth timeout:		30
* Gamma smooth timeout:		300
* Dimmer smooth timeout:		30

(W)[11:25:50]{bus.c:309}	check_clightd_version(): Connection timed out
(W)[11:25:50]{clightd.c:42}	No clightd found. Clightd is a mandatory dep.
(E)[11:25:50]{modules.c:242}	No functional module is running. Nothing to do...leaving.`
```

FedeDP, 2019-03-21:
Is Clightd systemd unit enabled?

brianread108, 2019-03-21:
> Is Clightd systemd unit enabled?

I presume so as if I run clight from a commandline it does the job.  Also 

```
 brianr  ~  ps ax | grep clightd
 2710 ?        Ss     0:00 /usr/lib/clightd/clightd
31906 pts/1    S+     0:00 grep --color=auto clightd
 brianr  ~  

```

FedeDP, 2019-03-21:
What about `systemctl status clightd`?

brianread108, 2019-03-21:
> What about `systemctl status clightd`? Can you paste here output?
```

 brianr  ~  systemctl status clightd
● clightd.service - Bus service to manage various screen related properties (ga>
   Loaded: loaded (/usr/lib/systemd/system/clightd.service; disabled; vendor pr>
   Active: active (running) since Thu 2019-03-21 11:25:25 GMT; 10h ago
 Main PID: 2710 (clightd)
    Tasks: 1 (limit: 4915)
   Memory: 1.4M
   CGroup: /system.slice/clightd.service
           └─2710 /usr/lib/clightd/clightd

Mar 21 11:25:25 fedora-28.bjsystems.co.uk systemd[1]: Starting Bus service to m>
Mar 21 11:25:25 fedora-28.bjsystems.co.uk systemd[1]: Started Bus service to ma>
l
```

FedeDP, 2019-03-22:
Ok, can you try enabling it? 
`sudo systemctl enable clightd.service`

Thanks! If this is the issue (and it most probably is) i'll add this note to the wiki page!

brianread108, 2019-03-22:
ok, that was it - logs show it running now... thanks.

closed 2019-03-22 by brianread108

FedeDP, 2019-03-22:
Great thanks! Added that to the wiki!