FedeDP/Clight #75: Package Clight/Clightd and dependencies in a container

demoy, 2019-04-29:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. EG: I'm always frustrated when [...]

I'm using Kubuntu and failed to find all the dependencies to compile the libmodule- clight- clightd suite, I'm too old and grumpy to switch to arch :), atleast not yet.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Please release this app through flatpak, docker, snappy or appimage?

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

 You could also suggest an alternative tool/script to use.. all the ones I found were broken

FedeDP, 2019-04-29:
Hi!

An user (@brianread108) submitted a script to install/update all deps at once, you can find it here: https://github.com/FedeDP/Clight/wiki/Build#dealing-with-dependencies.
Is this fine for you?

demoy, 2019-04-29:
BELOW is the error portion from running the script:

-- Checking for modules 'popt;gsl;libconfig'
--   No package 'gsl' found
CMake Error at /usr/share/cmake-3.13/Modules/FindPkgConfig.cmake:452 (message):
  A required package was not found

WHEN I check for gsl in the main respository I get: 

Package gsl is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libgslcblas0:i386 libgsl23:i386 gsl-bin:i386 libgslcblas0
  libgsl23 gsl-bin

I INSTALLED them (non-i386 ones) but the error persisted

FedeDP, 2019-04-29:
What Ubuntu version are you using?

Btw i never bothered much about appimage/flatpak etc etc. Are you able to help, in case?

demoy, 2019-04-29:
My ubuntu version is Ubuntu 19.04 ‘Disco Dingo’. I have not packaged for one of those container services either. I have only used docker because I have been traumatize by permission/dependency hell at work one too many times (everyone else uses windows)

FedeDP, 2019-04-29:
I could find [libgsl-dev](https://packages.ubuntu.com/disco/libgsl-dev) on Ubuntu 19.04, it should be the right one!

demoy, 2019-04-29:
Thanks for showing me the pattern (adding lib before and -dev after) I has able to crushout another 8+ dependencies this way. From what I can tell only clightd is still missing dependencies error below:

-- Checking for module 'ddcutil>=0.9.5'
--   No package 'ddcutil' found

THE version available in the respository is 0.9.2-1. Is it advisable for me to reduce the ddcutil version target?

FedeDP, 2019-04-29:
> THE version available in the respository is 0.9.2-1. Is it advisable for me to reduce the ddcutil version target?

I'd say no; ddcutil has had some API breaks and i don't think it would build!
Unfortunately Ubuntu does still not package ddcutil C library (there is no libddcutil-dev); your only luck is building from source!
See here for more details: https://github.com/FedeDP/Clightd/wiki/Ddcutil.

demoy, 2019-04-29:
Lol. Alice in Wonderland.. here into the rabbit hole, I hope I keep my head

FedeDP, 2019-04-29:
Installing arch would've been quicker indeed :D

demoy, 2019-04-29:
From what I read on (https://github.com/FedeDP/Clightd/wiki/Ddcutil) it's "to set external monitor brightness and thus supporting desktop PCs too". Can I comment it out as a dependency since I'm using a laptop?

FedeDP, 2019-04-29:
DDCutil is not a mandatory Clightd dep: https://github.com/FedeDP/Clightd/wiki/Build

If you build Clightd without "-DENABLE_DDC=1" option, it won't check it out :)

demoy, 2019-04-29:
Thank you! Working fine. Point me to where available settings are documented

closed 2019-04-29 by demoy

FedeDP, 2019-04-30:
Great to hear!
I'd like to have Clight/Clightd more easily available but i really got no time to spend on learning appimage/flatpak.
Moreover i'm not even sure Clight would work that way, as it has a "complex" architecture (eg: it requires a system service, Clightd; and it uses dbus.); i mean "complex" from a flatpak/appimage pov.

FedeDP, 2019-05-01:
@demoy  Moreover, i'd like to let you know that starting with next release, Clight and Clightd will have a new  command (to be used in place of "sudo make install"), that will build up a deb package ready to be installed with your package manager; have a look [here](https://github.com/FedeDP/Clight/wiki/Build#installation---debian).

This will surely help with dependencies and tracking clight/clightd installation.
The same applies to libmodule, that already supports deb package creation.