FedeDP/Clight #76: org.freedesktop.PowerManagement.Inhibit was not provided by any .service files

beanbagwarrior, 2019-05-26:
I am getting the following message when I'm starting clight

```
I inhibit_check(): The name org.freedesktop.PowerManagement.Inhibit was not provided by any .service files
PowerManagement inhibition appears to be unsupported.

```
I understand that there is a fallback in case Inhibit is not found, however I would like to know how to actually solve the issue.

I could not find a solution to that anywhere on the internet or your documentation. It would be good if you can add a section on how to solve the issue.

mindrunner, 2019-05-26:
Funny, we created the same ticket at the same time... 
#77

FedeDP, 2019-05-26:
Hi @beanbagwarrior !

Are you facing any issue (eg: dimmer kicks in while watching a movie)? 
Because that log does not mean that PowerManagegement won't work; it means that at the time clight starts, no PowerManagement API is appeared on dbus yet.
But clight still adds a match (to be notified) on PowerManagement API, thus if it later appears, clight is supporting it.

Hi @mindrunner, welcome back! :)
Are you using a DE? Or wm? In your case it seems PowerManagement dbus service is not appearing at all.
It is up to your DE to provide that dbus interface.

mindrunner, 2019-05-26:
I am using awesome-wm on arch linux.

FedeDP, 2019-05-26:
> If I install mate-power-manager and have it running, I see the same thing in my log.

And is dimmer kicking in while watching a movie?

mindrunner, 2019-05-26:
same problem with xfce4-power-manager. I also do not understand why I would need that. Isn't clight the kind of power management application (dimming, gamma, etc) which needs to be inhibited? So why is not clight exposing this interface to make vlc, chrome, etc register direclty there and then adjust brightness and gamma to ne "movie-optimized" values.

Don't get the point, why I need another power manager to achieve that.

beanbagwarrior, 2019-05-26:
Thanks @FedeDP, I just tested it out, it does not seem that the screen is dimming while watching a movie or a YouTube video. 

@mindrunner Haha, strange coincidence ;)

Not closing the thread for @mindrunner 's issue

mindrunner, 2019-05-26:
Just double checked:

open youtube, play video
start clight
screen dimms

Expected behaviour:
screen stays on 100% due to video playback

FedeDP, 2019-05-26:
@beanbagwarrior  
That log message is quite misleading though, i may fix that!

@mindrunner 
Clight is not a power manager: it is a screen backlight/gamma/dimming/dpms tool. It is not up to it to provide a system power management interface like org.freedesktop.PowerManagement.

> open youtube, play video
start clight
screen dimms

Did you restart your system after installing xfce4-power-manager? (just to be sure it is actually up and running...) and what does
`busctl --user introspect org.freedesktop.PowerManagement.Inhibit /org/freedesktop/PowerManagement/Inhibit org.freedesktop.PowerManagement.Inhibit`
output?

mindrunner, 2019-05-26:
Yes, the power manager is running.

```Failed to introspect object /org/freedesktop/PowerManagement/Inhibit of service org.freedesktop.PowerManagement.Inhibit: The name org.freedesktop.PowerManagement.Inhibit was not provided by any .service files
```

FedeDP, 2019-05-26:
It seems xfce4 too does not support it... it is really weird though!
[Here](https://forum.xfce.org/viewtopic.php?id=6513) is some very old post that states that xfce does not support it though.
I found [this](https://github.com/rodlie/powerkit) that does support this interface but unfortunately is not even on AUR. 
You can try with gnome-power-manager too.

mindrunner, 2019-05-26:
I installed the gnome-power-manager as well. But it seems, there is no binary to run. :( 

It's kind of frustrating, that power management in Linux is still pain in the ass...

FedeDP, 2019-05-26:
Unfortunately it seems org.freedesktop.PowerManagement.Inhibit interface is not as widely supported as i though, being it a freedesktop standard.

I may try to provide that interface myself within Clight as you suggested (where not already available).

mindrunner, 2019-05-26:
Cool, that could really help I think.

FedeDP, 2019-05-26:
Btw it seems at least xfce4-power-manager provides it: https://github.com/xfce-mirror/xfce4-power-manager/blob/master/src/org.freedesktop.PowerManagement.Inhibit.xml; i don't know why it does not appear on your system.

mindrunner, 2019-05-26:
Yeah, I saw that as well. But can't get it working on my system. :(

ochosi, 2019-07-28:
> Btw it seems at least xfce4-power-manager provides it: https://github.com/xfce-mirror/xfce4-power-manager/blob/master/src/org.freedesktop.PowerManagement.Inhibit.xml; i don't know why it does not appear on your system.

Yes, xfce4-power-manager does provide the inhibitor and it also works. You can call it with d-feet or gdbus call.

This works fine for me:
`gdbus introspect --session --dest org.freedesktop.PowerManagement --object-path /org/freedesktop/PowerManagement/Inhibit`

mindrunner, 2019-07-30:
Works for me, whereas clight still gives me the same warning:

```
[le@w530]: ~>$ gdbus introspect --session --dest org.freedesktop.PowerManagement --object-path /org/freedesktop/PowerManagement/Inhibit
node /org/freedesktop/PowerManagement/Inhibit {
  interface org.freedesktop.DBus.Properties {
    methods:
      Get(in  s interface_name,
          in  s property_name,
          out v value);
      GetAll(in  s interface_name,
             out a{sv} properties);
      Set(in  s interface_name,
          in  s property_name,
          in  v value);
    signals:
      PropertiesChanged(s interface_name,
                        a{sv} changed_properties,
                        as invalidated_properties);
    properties:
  };
  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml_data);
    signals:
    properties:
  };
  interface org.freedesktop.DBus.Peer {
    methods:
      Ping();
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };
  interface org.freedesktop.PowerManagement.Inhibit {
    methods:
      @org.freedesktop.DBus.GLib.Async("")
      Inhibit(in  s application,
              in  s reason,
              out u cookie);
      UnInhibit(in  u cookie);
      HasInhibit(out b has_inhibit);
      GetInhibitors(out as inhibitors);
    signals:
      HasInhibitChanged(b has_inhibit_changed);
    properties:
  };
};
[le@w530]: ~>$ clight 
[...]
Trying to start INHIBIT module as its USERBUS dependency was loaded...
inhibit_check(): The name org.freedesktop.PowerManagement.Inhibit was not provided by any .service files
```

FedeDP, 2019-08-23:
Damn, i forgot to reply!

> Works for me, whereas clight still gives me the same warning:

As i already said, that warning does not mean that clight is not binding on "HasInhibitChanged" signal. 
If inhibition does not work in Clight, it means xfce power manager is not sending "HasInhibitChanged" signal on dbus.

I am still trying to figure out a better solution; i don't really want to fully implement org.freedesktop.ScreenSaver dbus interface, as it does not really belong in Clight (ie: clight is not a screensaver/screen locker).
But org.freedesktop.ScreenSaver does not have any "HasInhibitChanged" signal, thus i cannot know when a ScreenSaver.Inhibit call is made, thus i cannot intercept them to pause DIMMER module.

On the other hand, org.freedesktop.PowerManagement interface is more about inhibiting power management functions, eg: suspend, hibernate etc etc.
This is the interface that eg: your torrent client uses to inhibit suspend while a torrent is downloading.
While it works within Clight as it exposes a nice "HasInhibitChanged", letting us intercept Inhibition states, it is kinda stupid to use it.

Finally, there is MPRIS dbus interface. We could match on "PlaybackStatus" property.
This is probably the cleanest road, but last time i poked around it, i could not make it work.

mindrunner, 2019-08-23:
Ok, interesting. I would assume, that clight should kind-of behave like a screensaver. 

- Dimming/Gamma mode on in normal operation mode. (screensaver will lock when idling)
- Dimming/Gamma forced to 100%/Daymode when watching movies (screensaver will never lock)

However, this does not seem to work reliabliy anyways. Watching movies with VLC will prevent my screensaver from locking. But browserbased watching (youtube, netflix, plex, etc) does not prevent my screensaver from locking. So not an optimal solution... :/

FedeDP, 2019-08-23:
At least chromium and firefox do send "org.freedesktop.ScreenSaver.Inhibit", thus they should inhibit DPMS/dimmer while eg: watching a movie on youtube/netflix.

Yes Clight kinda is a ScreenSaver; i could try to provide a inhibit/unhinibit only interface in clight, but what would happen when clight is run inside a full DE (eg: gnome/kde) that already provides that API through they powermanagement daemons?
Moreover, there is so little documentation about that bus interface: https://people.freedesktop.org/~hadess/idle-inhibition-spec/re01.html.

mindrunner, 2019-08-23:
> At least chromium and firefox do send "org.freedesktop.ScreenSaver.Inhibit", thus they should inhibit DPMS/dimmer while eg: watching a movie on youtube/netflix.
I am using lightdm, awesome-wm, xautolock and i3lock. chromium and firefox are not able to prevent my lockscreen from triggering. I have no idea why is that....

> but what would happen when clight is run inside a full DE (eg: gnome/kde) that already provides that API through they powermanagement daemons?
What would you expect to happen? I assume, that KDE, GNOME, etc are using the same interfaces, since they are the "standard". Not sure, where this could interfere.

FedeDP, 2019-08-23:
> What would you expect to happen? I assume, that KDE, GNOME, etc are using the same interfaces, since they are the "standard". Not sure, where this could interfere.

I mean, if clight starts to offer org.freedesktop.ScreenSaver server API (just like these DEs do), the first to start will take ownership of that dbus interface, and i am pretty sure it would not be Clight :)

mindrunner, 2019-08-23:
Why does clight need to be the server? It should just listen / poll for either (allowed to lock, or not alloed to lock)

Maybe I have a total misunderstanding of things, sorry. I am confused.

FedeDP, 2019-08-23:
Mmh no problem, let's start from scratch :) 
* a ScreenSaver app (in DEs it is usually the PowerManager) exposes org.freedesktop.ScreenSaver API on bus; ScreenSaver API is needed to inhibit screen dimming and dpms while watching a video. We call this the server application.
* An app, be it totem media player or firefox on youtube, as soon as a video start playing, calls org.freedesktop.ScreenSaver.Inhibit. We call firefox a client application.
* The server, receiving the Inhibit request, do its job (eg: gnome stops its screen dimming tool)

Now, we want clight to be another server. But there can be only one org.freedesktop.ScreenSaver 
 server in a user session.

FedeDP, 2019-08-23:
Moreover, only server knows that a inhibition has been requested, as in org.freedesktop.ScreenSaver API there is no signal sent whenever an application requests an inhibition.

FedeDP, 2019-08-25:
@mindrunner As you suggested, indeed org.freedesktop.ScreenSaver is the most compliant solution, and probably the best among all.

In commits 757ab602491f620f8ea7db436b702e3b73ef97a2 and 6258ee4af28ec3bf083a456743a9cc47c5082a88 (libmodule branch) i started a port to provide this freedesktop specification inside Clight.

It should go like this: if org.freedesktop.ScreenSaver API is available (ie: Clight can bind on it), Clight will take its ownership offering "Inhibit" and "UnInhibit" methods.
Else, well you're out of luck, and you'll have to manually inhibit screen dimmer or maybe through a script started together with your video player, eg:
```
busctl --expect-reply=false --user call org.clight.clight /org/clight/clight org.clight.clight Inhibit "b" true && mpv
```

Old org.freedesktop.PowerManagement API support has been completely dropped and conf option to avoid inhibit management has been dropped too (i always found it rather useless, as it does not harm anything to leave it on, it just offers a much more pleasant experience).

So, summing it up:
* Best case: org.freedesktop.ScreenSaver available -> you'll have everything working fine; DIMMER and DPMS will be inhibited in a much better way than now: eg: chromium will only keep is inhibition ON while you're in a tab that is playing video. If you change tab, ScreenSaver inhibition will be dropped.
Moreover, ScreenSaver inhibition is only enabled for videos.
Old org.freedesktop.PowerManagement API was inhibited by eg: torrent downloading, file manager copying, audio playing...
* Worst case: org.freedesktop.ScreenSaver API is already owned by another software, eg: your DE powermanager. You are out of luck.
You will have to manually enable ScreenSaver inhibition from Clight desktop file (or Clight dbus interface). Note that org.clight.clight.Inhibit interface remained the same, and it is now internally managed like an org.freedesktop.ScreenSaver inhibition held by "Clight" (ie: it really is the same).

There is no average case :)
Unfortunately i think we cannot provide a better experience than this...

mindrunner, 2019-08-25:
> Now, we want clight to be another server. But there can be only one org.freedesktop.ScreenSaver
server in a user session.

Wonder whats the design decision behind that. Sounds quite restrictive...

Will go through the rest of your comments soon and do some testing. Thanks for the nice clarification

FedeDP, 2019-08-25:
> Wonder whats the design decision behind that. Sounds quite restrictive...

Well, let's think about it: when eg: chromium calls org.freedesktop.ScreenSaver through dbus, dbus needs to know to whom this request should be forwarded; it means that every interface name is unique.

> Will go through the rest of your comments soon and do some testing. Thanks for the nice clarification

Unfortunately it is not super easy to test it out: clight libmodule branch requires an unreleased version of libmodule (libmodule built from git master) and an unreleased version of Clightd (clightd built from libmodule5 branch).

mindrunner, 2019-08-30:
> Well, let's think about it: when eg: chromium calls org.freedesktop.ScreenSaver through dbus, dbus needs to know to whom this request should be forwarded; it means that every interface name is unique.
Subscribers usually do not need to be unique. Multiple instances can subscribe to a bus to receive notifications. This is what buses were invented. Anyways, there were probably reasons and this is not the right place to discuss I suppose.

> Unfortunately it is not super easy to test it out: clight libmodule branch requires an unreleased version of libmodule (libmodule built from git master) and an unreleased version of Clightd (clightd built from libmodule5 branch).
Alright. I am caught up with other stuff anyways at the moment. Any ETA when this can be on stable? Or a easily testable version?

FedeDP, 2019-08-30:
> Subscribers usually do not need to be unique. Multiple instances can subscribe to a bus to receive notifications. This is what buses were invented. Anyways, there were probably reasons and this is not the right place to discuss I suppose.

Yes! But Clight is not subscribing to any Signal, as org.freedesktop.ScreenSaver API has no public signals about whether an inhibitions has been requested; that is the opposite of org.freedesktop.PowerManagement API that had a nice "Inhibited" signal.

Thus, Clight must provide and implement the ScreenSaver specification on its own, because it cannot subscribe to any signal. Hopefully i explained myself this time :)

> Any ETA when this can be on stable? Or a easily testable version?

I'd say as soon as i find some time :) 
That's the roadmap:
* Release libmodule 5.0.0 as stable, without updating Archlinux aur packages as it contains API breaks and currently released Clight and Clightd would break
* Merge both Clight libmodule's branch and Clightd libmodule5's branch into respective masters

Here it will be very easy to test it!
Then a couple of weeks of public testing and hopefully a release.
I hope to release Clight and Clightd 4.0 early october.

mindrunner, 2019-08-30:
> Thus, Clight must provide and implement the ScreenSaver specification on its own, because it cannot subscribe to any signal. Hopefully i explained myself this time 

Well, I suppose this is one of the reasons why Linux desktops are still not ready for the mainstream... 
Anyways, also this is another discussion which should not be here....

Thanks for the input! Let me know when the testing build is available! Exciting to try it out.

FedeDP, 2019-09-07:
The work is now in master. It does require libmodule 5.0.0 though (released but unavailable yet on AUR to avoid breaking current clight/clightd installations).

mindrunner, 2019-09-11:
Hmm, so we need a libmodule-git? What is the preferred way of getting this going on arch?

FedeDP, 2019-09-12:
We need libmodule 5.0.0 and clightd-git. Well, about clight-git and clightd-git it's quite easy as there are respective packages.
About libmodule, you can find an updated PKGBUILD in its repository: https://github.com/FedeDP/libmodule/blob/master/Extra/Arch/PKGBUILD. You can use that PKGBUILD to build libmodule 5.0.0 :)

mindrunner, 2019-09-17:
I reinstalled everything today (libmodule-5, clightd-git clight-git) and I am getting this when I try to start clight:
```
[le@w530]: ~/src/libmodule/Extra/Arch>$ clight
Bad file descriptor
A lock is present on /home/le/.local/share/clight/clight.log. Another clight instance running?
```
After a `mkdir -p /home/le/.local/share/clight`, the application is starting.

FedeDP, 2019-09-17:
Thanks, nice catch! I will fix it asap!

mindrunner, 2019-09-20:
Cheers. Unfortunately, my webcam is not working at the moment and I am not qute sure if its a software or hardware issue. Without webcam, I cannot proper test clight. Will pop in later I suppose.

FedeDP, 2019-09-20:
I'll answer there!

FedeDP, 2019-10-03:
I just pushed a commit that fixes "/ScreenSaver" (instead of /org/freedesktop/ScreenSaver) issue.
Moreover, when org.freedesktop.ScreenSaver bus name is already owned by some other service, clight will now gracefully fallback at monitoring it, just like `busctl --monitor org.freedesktop.ScreenSaver` does.
This way it can still receive updates about ScreenSaver inhibition and works flawlessly!

mindrunner, 2019-10-09:
I am following up on your last posts in #81 

> Yes, you can subscribe to TEMP_UPD to receive current user-setted temperature, eg: 6500 during the DAY; then you can store it to a global variable; moreover, you can set daytime = -1 in your request, this way it will use current daytime.

Yeah, I saw that. But this feels quite error prone. As far as I understand, I will only get updates as soon as the state changes. So initially, I do not know the user's settings. Why can't I use `dbus` to read the settings from within the module? Is it because I cannot communicate with myself via dbus?

> Obviously you can merge this custom module with the inhibit_bl one.

Was thinking about that. But first needs to work :)

> And please, if you don't mind, share your feedback with custom modules, ie: if you find them hard or whatever :)

So far, it makes sense to me and seems like an easy to understand interface.

> I will think of a better approach to inhibit modules, eg: a request with a list of modules to be inhibited.

This sounds exacly like what I need :)

> On a second thought, i actually think that inhibiting GAMMA is not really good for your eyes; i don't understand the reason behind that!
I can see inhibiting DIMMING and DPMS (obviously...otherwise your screen would switch off/dim during a movie) useful; inhibiting BACKLIGHT already seems quite exaggerated to me, but i can understand that one would wish to set maximum backlight and leave it untouched until the end of a movie.
But i cannot see any real use case for GAMMA inhibition. I think you should "fallback" to a custom module for that functionality; the one provided above should work fine.

Well, you are right. We use gamma for a reason. We want to reduce the blue light emission to not simulate daylight for the brain at night. However, watching a movie with gamma 4500 feels a little bit vintage... I prefer removing the blue light cancellation to see 'real'-colors. This might be just a personal thing. But this is why, we can implement as a personal module. :)

FedeDP, 2019-10-10:
> As far as I understand, I will only get updates as soon as the state changes. So initially, I do not know the user's settings.

You will get new temp as soon as GAMMA starts and set screen temperature; i think it won't take more than a couple of seconds :)

> Why can't I use dbus to read the settings from within the module? Is it because I cannot communicate with myself via dbus?

I decided to hide both conf and state for custom modules to avoid having them changed in wrong ways by custom modules.  
But you should be able to parse a `busctl get-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf ...` commands with something like [popen](http://man7.org/linux/man-pages/man3/popen.3.html).

> But this is why, we can implement as a personal module. :) 

:muscle:

mindrunner, 2019-10-10:
> You will get new temp as soon as GAMMA starts and set screen temperature; i think it won't take more than a couple of seconds :)

But do I get both, day and night values? Or only the current one? I need both settings to implement my logic.

> I decided to hide both conf and state for custom modules to avoid having them changed in wrong ways by custom modules.
But you should be able to parse busctl get-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf ... commands with something like popen and sscanf.

I cannot get it working here. if I call `busctl` with `popen` in `init()`, I am getting this:
`Failed to get property DayTemp on interface org.clight.clight.Conf: Process org.clight.clight exited with status 1
`

My assumption is, that it is because clight is not yet fully up and running in init() state. So, I tried to make the calls in my `receive()` callback. However, it does not seem to work either:
`Failed to get property NightTemp on interface org.clight.clight.Conf: Connection timed out`

mindrunner, 2019-10-10:
Code demonstrating both cases:
```
#include <clight/public.h>
#include <stdlib.h>

CLIGHT_MODULE("INHIBIT_GAMMA");

DECLARE_MSG(temp_req, TEMP_REQ);

int night_temp = -1;
int day_temp = -1;

static void init(void) {
  if(night_temp == -1) {
    FILE *f;
    f = popen ("busctl --user get-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf NightTemp", "r");
    fscanf(f, "i %d", night_temp);
    printf("Night Temp: %d\n", night_temp);
  }
  if(day_temp == -1) {
    FILE *f;
    f = popen ("busctl --user get-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf DayTemp", "r");
    fscanf(f, "i %d", day_temp);
    printf("Day Temp: %d\n", day_temp);
  }
  M_SUB(INHIBIT_UPD);
}

static void receive(const msg_t *msg, const void *userdata) {
  switch (MSG_TYPE()) {
    case INHIBIT_UPD: {
      inhibit_upd *up = (inhibit_upd *)MSG_DATA();
      if(night_temp == -1) {
        FILE *f;
        f = popen ("busctl --user get-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf NightTemp", "r");
        fscanf(f, "i %d", night_temp);
        printf("Night Temp: %d\n", night_temp);
      }
      if(day_temp == -1) {
        FILE *f;
        f = popen ("busctl --user get-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf DayTemp", "r");
        fscanf(f, "i %d", day_temp);
        printf("Day Temp: %d\n", day_temp);
      }

      if (up->new) {
        if(day_temp > 0) {
          INFO("We are now inhibited! Set daylight screen temperature.\n");
          temp_req.temp.daytime = NIGHT;
          temp_req.temp.new = day_temp;
          M_PUB(&temp_req);
        } else {
          WARN("Day Temp unknown, not setting anything\n");
        }
      } else {
        if(night_temp > 0) {
          INFO("We're not inhibited anymore. Set auto gama.\n");
          temp_req.temp.daytime = NIGHT;
          temp_req.temp.new = night_temp;
          M_PUB(&temp_req);
        } else {
          WARN("Night Temp unknown, not setting anything\n");
        }
      }
      break;
    }
    default:
      break;
    }
}
```

Clight Log:
```
[le@w530]: ~>$ clight
SCREEN forcefully disabled as Clightd was built without screen support.
Clightd found, version: 4.0.
'/home/le/.local/share/clight/modules.d/backlight' loaded.
'/home/le/.local/share/clight/modules.d/gamma' loaded.
Set timeout of 3s 0ns on fd 11.
Failed to get property NightTemp on interface org.clight.clight.Conf: Process org.clight.clight exited with status 1
Night Temp: -1
Failed to get property DayTemp on interface org.clight.clight.Conf: Process org.clight.clight exited with status 1
Day Temp: -1
Initial AC state: connected.
New location received: 14.72, -91.16.
Set timeout of 0s 1ns on fd 37.
Emitting Location property
Normal transition to 6500 gamma temp started.
Next alarm due to: Thu Oct 10 17:18:00 2019
Set timeout of 24776s 0ns on fd 37.
AC curve: y = 0.055245 + 0.180897x + -0.008590x^2
BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
Keyboard backlight calibration supported.
Set timeout of 0s 1ns on fd 40.
Emitting Sunrise property
Average frames brightness: 0.507839.
Ambient brightness: 0.508 -> Backlight pct: 0.752.
Set timeout of 600s 0ns on fd 40.
Emitting Sunset property
Emitting DayTime property
Emitting Temp property
Emitting AmbientBr property
Emitting BlPct property
Emitting KbdPct property
New ScreenSaver inhibition held by vlc: Playing some media.. Cookie: 1804289383
ScreenSaver inhibition enabled.
Being paused.
Failed to get property NightTemp on interface org.clight.clight.Conf: Connection timed out
Night Temp: -1
```

mindrunner, 2019-10-11:
@FedeDP i am still confused about the chrome/chromium inhibit problem. Can you confirm that it still works on your side. Also, can you give me some information about your setup? Distribution, package version, etc? 

Thank you :)

FedeDP, 2019-10-11:
> But do I get both, day and night values? Or only the current one? I need both settings to implement my logic.

I see, sorry! I forgot about that...

> My assumption is, that it is because clight is not yet fully up and running in init() state

You are probably right; note that with libmodule you will receive PubSub messages wheneveer a module gets started, thus you could wait for "INTERFACE" module started and then get those values: https://libmodule.readthedocs.io/en/latest/src/pubsub.html#system-messages .
But there is too much libmodule involved in here :)
Here is an example, if you wish to play with it: https://github.com/FedeDP/libmodule/blob/master/Samples/Easy/doggo.c#L82 .

About popen, note that you need to pass variable address to store its value:
```
FILE *f = popen ("busctl --user get-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf NightTemp", "r");
fscanf(f, "i %d", &night_temp); // note "&night_temp" here
printf("Night Temp: %d\n", night_temp);
```
Note that i am not sure it will work though... I need to play with it when i got time!

> Can you confirm that it still works on your side. Also, can you give me some information about your setup? Distribution, package version, etc?

Yes it works; i am on archlinux, latest chromium available, on KDE.
On which DE are you on? 
This is the code that chromium uses to select which interface it should use to inhibit: https://chromium.googlesource.com/experimental/chromium/src/+/refs/wip/bajones/webvr/device/power_save_blocker/power_save_blocker_x11.cc#453.
It seems chromium does support inhibition only when on Gnome, Kde, or xfce. It seems weird to me though; see comments here: https://chromium.googlesource.com/experimental/chromium/src/+/refs/wip/bajones/webvr/device/power_save_blocker/power_save_blocker_x11.cc#39

As you can see, on gnome a different bus name is used: https://chromium.googlesource.com/experimental/chromium/src/+/refs/wip/bajones/webvr/device/power_save_blocker/power_save_blocker_x11.cc#55.

FedeDP, 2019-10-11:
I now see why popen is not working! Stupid me!
It cannot work as same process that should answer (clight, that is interface owner), is also waiting for the answer... that is why you get "Connection timed out".

FedeDP, 2019-10-11:
> This is the code that chromium uses to select which interface it should use to inhibit

We should probably ask chromium developers to default to freedesktop api on unsupported DEs/WMs by opening a new issue on https://bugs.chromium.org/p/chromium/issues/list :)

mindrunner, 2019-10-11:
> > This is the code that chromium uses to select which interface it should use to inhibit
> 
> We should probably ask chromium developers to default to freedesktop api on unsupported DEs/WMs by opening a new issue on https://bugs.chromium.org/p/chromium/issues/list :)

Thanks for pointing me there. I already have this on my bucket list. But before creating a ticket, I will patch chromium and test whether it works for me or not. Also, if we have a patch, it might get into upstream sooner. :)

mindrunner, 2019-10-12:
Created bug-report here:
https://bugs.chromium.org/p/chromium/issues/detail?id=1013812

Workaround to activate inhibition:
`XDG_CURRENT_DESKTOP=XFCE chromium`

mindrunner, 2019-10-13:
I tried to wait for the `INTERFACE` module being loaded. Still timing out when using `busctl`. I will propbably stiuck with my hardcoded solution for the following reasons.

1) Interface does not support accessing the config. Everything else are dirty workarounds.
2) My use case seems not applicable for the mainstream, so no point in putting effort into this.

Thanks for the support and maybe at some point there will be a more flexible interface.

FedeDP, 2019-10-16:
> Workaround to activate inhibition:

I will add this to wiki page, thanks! It was so easy yet i did not think about that...

> Thanks for the support and maybe at some point there will be a more flexible interface.

The best thing here would be to allow custom modules to access clight's conf and state readonly.
I am not sure about it though as i don't like custom modules possibly messing with them.

I think, being a custom module, it is fine to hardcode those values. Moreover, you can even parse clight conf file manually to lookup those values, or use an ugly `popen("grep XXX /etc/defaults/clight.conf");` and then parse its output.

FedeDP, 2019-10-16:
I added a wiki comment about chrome inhibition: https://github.com/FedeDP/Clight/wiki/ScreenSaver-Specification#chromiumchrome-notes.

Thanks @mindrunner for all your testing and work!
I think we can consider this issue closed ;)

I hope to release new clight soonish; i have some more work on Clightd though.

closed 2019-10-16 by FedeDP

mindrunner, 2019-10-16:
> The best thing here would be to allow custom modules to access clight's conf and state readonly.
I am not sure about it though as i don't like custom modules possibly messing with them.

I think, this is a good idea.

> I think, being a custom module, it is fine to hardcode those values. Moreover, you can even parse clight conf file manually to lookup those values, or use an ugly popen("grep XXX /etc/defaults/clight.conf"); and then parse its output.

Yeah, was thinking about that. But then got lazy... :)

> I hope to release new clight soonish; i have some more work on Clightd though.

Cool :) Is there a roadmap? I finally can use clight in production because of the new module and dynamic inhbit stuff. Thats great. No more disfunctional redshift messing with my screen :)

> I added a wiki comment about chrome inhibition: https://github.com/FedeDP/Clight/wiki/ScreenSaver-Specification#chromiumchrome-notes.

I would like to add something. I figured out, that using this workaround has more catches. E.g. you get logged out of all sessions if you change this setting.... Not sure, if there is more stuff..

>Thanks @mindrunner for all your testing and work!
I think we can consider this issue closed ;)

You are welcome! I think so too. Thanks for all the support.

FedeDP, 2019-10-16:
> Cool :) Is there a roadmap? I finally can use clight in production because of the new module and dynamic inhbit stuff. Thats great. No more disfunctional redshift messing with my screen :)

I am so proud to hear that!

I think next version is gonna be really huge and i had lot of fun developing all this stuff! And i think it is already well-tested.  
I still have to merge a PR in Clightd since 2 months or so; it will greatly improve ambient brightness computation with camera captures; the downside is that it will "break" lots of clight users backlight curves.  
This is not a big deal as there is a major version bump too in both Clight and Clightd, thus users should expect some changes.

After that, i should only fix some wiki pages and write a changelog for both :)

> I would like to add something. I figured out, that using this workaround has more catches. E.g. you get logged out of all sessions if you change this setting.... Not sure, if there is more stuff..

Ouch :( let's hope chromium developers listen to your enhancement request! I will follow it, may be i'll add a comment with a small patch.

> You are welcome! I think so too. Thanks for all the support.

Thanks again!

mindrunner, 2019-10-16:
> I still have to merge a PR in Clightd since 2 months or so; it will greatly improve ambient brightness computation through camera captures; the downside is that it will "break" lots of clight users backlight curves.

Happy to hear that. For me personally, the default setting is too dark. Especially in combination with gamma shift, it feels exhausting for the eyes and some text is hard to read. i adapted my curve, but I am excited to see more improvements shipped by default :)

> Ouch :( let's hope chromium developers listen to your enhancement request! I will follow it, may be i'll add a comment with a small patch.

I was planning to submit a patch, but since I have trouble compiling the source, I could not test it. (Created another ticket for that, but did not get any response.)

FedeDP, 2019-11-24:
@mindrunner  I saw your chromium bug report got fixed! THIS IS HUGE! :)

mindrunner, 2019-11-24:
Yeah man. Its awesome! I was super stoked as well. Also they did it proper and not in a one line quick fis as i would have been doing hehe. Hope this is in stable builds soon :)