FedeDP/Clight #77: inhibit_check(): The name org.freedesktop.PowerManagement.Inhibit was not provided by any .service files PowerManagement inhibition appears to be unsupported.

mindrunner, 2019-05-26:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> 4 aur/clightd 3.4-1 (+2 0.06%) (Installed)
> 3 aur/clight 3.1-3 (+3 0.13%) (Installed)

**Used distribution:**
> arch

**Describe the bug**
Inhibit seems unsupported on my system, thus clight dims and gammas the screen on video watching for example

**Expected behavior**
starting to watch a video should bring my screen to max-backlight and force daylight mode 

**To Reproduce**
Not sure.

[.clight.log](https://github.com/FedeDP/Clight/files/3220944/default.clight.log)

FedeDP, 2019-05-26:
Hi @mindrunner !
Two issues (#77 and #76) created in a couple of minutes with the same issue!! What the odds?
Unfortunately your issue was created just a bit later :)
Can we move on other issue? 
I'll delete this (can you please copy/paste your message in other conversation?)

Thanks!

mindrunner, 2019-05-26:
I think, this ticket has more information attached, such as log, etc. if we close the other one as duplicate, we dont have to move stuff around.