FedeDP/Clight #78: Wrong daylight calculation

mindrunner, 2019-05-26:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> latest

**Used distribution:**
> arch

**Describe the bug**
Leaving sunrise and sunset commented in the config file, will result in screen going to night mode at daytime. The location I receive from geoclue seems valid. If i manually set sunrise to 7am and sunset to 7pm, it works without a problem.

**Expected behavior**
Automatically detect sunrise and sunset times by geolocation 

**To Reproduce**
Not sure

[.clight.log](https://github.com/FedeDP/Clight/files/3220956/default.clight.log)

FedeDP, 2019-05-26:
Hi! Thanks for opening this issue!
Can you share a new verbose clight.log without specifying Sunrise/Sunset times? That way i can see the bug.
Or you can share with me your location (not necessarily accurate), eg: 38.7:-42.4, just to understand your quadrant and thus where the algorithm to compute sunrise/sunset is wrong.

mindrunner, 2019-05-26:
Attached the log. It also contains the location. I am in TZ America/Guatemala. Sun 26 May 2019 12:33:21 PM CST now.
---> Daytime!

[.clight.log](https://github.com/FedeDP/Clight/files/3220984/default.clight.log)

FedeDP, 2019-05-26:
Thank you! I will use your position trying to find the bug, and hopefully later this week (most probably next weekend) i'll have this fixed. 
Thanks again, for now you can fallback to set Sunrise/Sunset times :)

FedeDP, 2019-06-01:
This issue should now be fixed in current master. Note that current master requires clightd from -git.
Let me know if it works for you!

mindrunner, 2019-06-05:
Looks better now, thank you! :)

closed 2019-06-05 by mindrunner