FedeDP/Clight #79: Exiting clight leaves gamma as it is.

mindrunner, 2019-05-26:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> latest

**Used distribution:**
> arch

**Describe the bug**
Exiting clight leaves for example night-mode on. Thus, there is no way to temporary disable night mode at nighttime.

**Expected behavior**
Bring system to a consistent state again. Could be a config option as well. 

**To Reproduce**
Wait for sunset. Kill clight. 
[.clight.log](https://github.com/FedeDP/Clight/files/3220988/default.clight.log)

Please attach a Clight verbose log too!

FedeDP, 2019-05-26:
Hi! You can set a different temperature through Clight's bus api: https://github.com/FedeDP/Clight/wiki/Api#bus-properties

Something like:
`busctl --user set-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf NightTemp "i" 6500`
should work.

mindrunner, 2019-05-26:
That does not solve my problem. I want to disable all clight auto-adjustment, thus quitting clight. But not preserving the current state. This is closely related to my other ticket #76 . 

I need to exit clight to be able to watch a movie, because it does not act accordingly to the Inhibit feature of the OS.

FedeDP, 2019-05-26:
You can just forcefully inhibit dimmer (right click on clight desktop file and click on "Pause screen dimmer").
The same can be achieved with 
`busctl --user call org.clight.clight /org/clight/clight org.clight.clight Inhibit "b" true`
This is exactly the same thing happening when org.freedesktop.PowerManagement.Inhibit is supported.

mindrunner, 2019-05-26:
That might be an option. Is there such a command to forcefully enable day or night mode as well?

A system tray application would be great!

FedeDP, 2019-05-26:
> That might be an option. Is there such a command to forcefully enable day or night mode as well?

Unfortunately not...i never considered that! What would be its purpose?

> A system tray application would be great!

I know :( if only one had infinite time! I would love to see it!

mindrunner, 2019-06-05:
So, lets stick with the cli command for now. Maybe I will sit down and write a small systray app for that...

closed 2019-06-05 by mindrunner

FedeDP, 2019-06-05:
It would be great!!

mindrunner, 2019-06-05:
Is there a documentation about the full bus api clight supports?

FedeDP, 2019-06-06:
> Is there a documentation about the full bus api clight supports?

Yes, it is! Any API with no documentation is pretty useless :) 
Take a look at https://github.com/FedeDP/Clight/wiki/Api !