FedeDP/Clight #80: Support for manual brightness control

beanbagwarrior, 2019-05-26:
**Is your feature request related to a problem? Please describe.**
The Arch Wiki describes under [backlight](https://wiki.archlinux.org/index.php/Backlight)

> Screen brightness might be tricky to control. On some machines physical hardware switches are missing and software solutions may not work well. However, it is generally possible to find a functional method for a given hardware. This article aims to summarize all possible ways to adjust the backlight. 

Setting up and using a window manager i3 I can confirm, that solving such issue can be quite tricky and tedious.

Given your software can already control the brightness of the screen. Why not extend it to support manual brightness control too?

**Describe the solution you'd like**
Ideally, the software should calibrate before starting to adjust the brightness based on the brightness, but even then sometimes the screen is too bright or too dim. 

To solve that you may have a program which one can control brightness manually through the command line. For example,  ```clight --manual increase/decrease```
This would suspend the clight automatic adjustment, until we do something like ```clight --auto``` 
This would allow us to easily set keybindings of our choice in our window manager or even in desktop environments.

**Describe alternatives you've considered**
I've looked into the Arch Wiki for solution to this problem, and most of it looks hacky. Such a combined system would be an elegant solution to this problem, and would make this piece of software even more useful.

FedeDP, 2019-05-26:
Hi!
As you can see right-clicking on desktop icon, you can already suspend clight automatic backlight calibration.
Then, you can play with [Clightd](https://github.com/FedeDP/Clightd) to increase/decrease backlight: https://github.com/FedeDP/Clightd/wiki/Api#orgclightdclightdbacklight.
An example:
`busctl call org.clightd.clightd /org/clightd/clightd/Backlight org.clightd.clightd.Backlight RaiseAll "d(bdu)s" 0.1 0 0 0 ""`
This will raise any monitor connected backlight of about 10%. 

Note only that Clight dimmer module does not play well with external backlight changes, as it stores current backlight level (setted by Clight) to resume same level when leaving dimmed state. If someone else touches backlight, it would resume a "wrong" level.
It is not a big issue i guess.

beanbagwarrior, 2019-05-26:
I see, my bad I did not go through the clightd wiki before, what I want to do can be done with clightd itself, clight uses clightd to change backlight intensity if I am not wrong.

Thanks anyway! :+1:

closed 2019-05-26 by beanbagwarrior

FedeDP, 2019-05-26:
> clight uses clightd to change backlight intensity if I am not wrong.

You're right...and you're welcome too! :)