FedeDP/Clight #81: Add Brightness module support to org.freedesktop.PowerManagement.Inhibit integration

NICHOLAS85, 2019-05-27:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. EG: I'm always frustrated when [...]

When watching longer videos which are darker than normal I tend to turn up my brightness in order to see better. Clight kicks in halfway through the video, taking it's ambient brightness measurements and drops my brightness down to what it detects it should be at. I then have to turn my brightness back up and wait until Clight inevitably takes it's measurements again.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Add a config option which can suspend backlight adjustment using the org.freedesktop.PowerManagement.Inhibit interface the same way the dimmer module can be suspended

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Simply killing Clight before watching longer videos/movies. This is not as convenient, unfortunately.

FedeDP, 2019-05-27:
Hi! This is a nice and very simple idea. Note that it was already on my todo: https://github.com/FedeDP/Clight/blob/master/TODO.md#mpris (something similar at least).

Btw you can pause/resume backlight autocalibration by right-clicking on desktop file and choosing "Pause backlight calibration", or manually issuing:
`busctl --user set-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf NoAutoCalib "b" true`

FedeDP, 2019-09-07:
The work is now in master. It does require libmodule 5.0.0 though (released but unavailable yet on AUR to avoid breaking current clight/clightd installations).

FedeDP, 2019-09-07:
> When watching longer videos which are darker than normal I tend to turn up my brightness in order to see better. Clight kicks in halfway through the video, taking it's ambient brightness measurements and drops my brightness down to what it detects it should be at

Note that clight does now support [user custom modules](https://github.com/FedeDP/Clight/wiki/Custom-Modules) and it does install a skeleton custom module to solve this issue: inhibit_bl.skel.

NICHOLAS85, 2019-09-08:
Hey @FedeDP, I've found the `inhibit_bl.skel` file and placed a copy into ~/.local/clight/modules.d/. After restarting clight I find the following messages in clight.log
```
(I)[17:16:45]{main.c:134}       Clightd found, version: 4.0.
(W)[17:16:45]{main.c:166}       '/home/nicholas/.local/share/clight/modules.d/inhibit_bl.skel' failed to load.
(I)[17:16:45]{upower.c:68}      Initial AC state: connected.
(W)[17:16:45]{interface.c:225}  Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
(I)[17:16:45]{backlight.c:238}  Keyboard backlight calibration enabled.
(I)[17:16:46]{backlight.c:259}  Ambient brightness: 0.435 -> Backlight pct: 0.605.
```
It seems theres a problem loading the file, am I missing a step?
I've built clight, clightd, libmodule, and ddcutil from source today so everything should be up to date.

FedeDP, 2019-09-09:
Hi! You must build it and then place built binary in the ~/.local/share/clight/modules.d/ folder!
Have a look at [wiki](https://github.com/FedeDP/Clight/wiki/Custom-Modules)!

NICHOLAS85, 2019-09-09:
I see, I wasn't aware I had to rename your .skel file before building it and simply ran into an error. I'll give it a try later today!

FedeDP, 2019-09-09:
I hope it works great for you!

mindrunner, 2019-09-20:
I have installed clight-git, clightd-git and libmodule-5.0.0-1.

Enabled all modules in the config. Starting VLC, screen stayes dimmed and in night mode.

Is it not supposed to make everything bright automatically now?

FedeDP, 2019-09-20:
Inhibition was working in one of your latest [comments](https://github.com/FedeDP/Clight/issues/76#issuecomment-530858639).
Can you paste your .local/share/clight/clight.log?

mindrunner, 2019-09-20:
Yeah, I can remember that. I wanted to continue testing yesterday, but did not work anymore.

```
(W)[19:43:36]{opts.c:193}	SCREEN forcefully disabled as Clightd was built without screen support.

Clight
* Software version:		4.0
* Global config dir:		/etc/default
* Global data dir:		/usr/share/clight
* Starting time:		Thu Sep 19 19:43:36 2019

Starting options:

### Generic ###
* Verbose (debugging):		Enabled
* Autocalibration:		Enabled
* Inhibit autocalibration:		Disabled
* Keyboard backlight:		Enabled
* Number of captures:		5
* Webcam device:		Unset
* Backlight path:		Unset
* Shutter threshold:		0.00
* Ambient gamma:		Disabled
* Screen contrib:		0.10
* Screen samples:		10
* User position:		Unset
* Daily screen temp:		6500
* Nightly screen temp:		4000
* User set sunrise:		Unset
* User set sunset:		Unset
* Event duration:		1800
* Dimmer backlight pct:		0.10

### Timeouts ###
* Daily timeouts:		AC 600	BATT 1200
* Nightly timeout:		AC 2700	BATT 5400
* Event timeouts:		AC 300	BATT 600
* Dimmer timeouts:		AC 250	BATT 250
* Dpms timeouts:		AC 900	BATT 300
* Screen timeouts:		AC 30	BATT -1

### Modules ###
* Backlight:		Enabled
* Gamma:		Enabled
* Dimmer:		Enabled
* Dpms:		Enabled
* Screen:		Disabled

### Smooth ###
* Bright smooth trans:		Enabled
* Gamma smooth trans:		Enabled
* Dimmer smooth trans:		ENTER: Enabled, EXIT: Enabled
* Bright smooth steps:		0.05
* Gamma smooth steps:		50
* Dimmer smooth steps:		ENTER: 0.05, EXIT: 0.05
* Bright smooth timeout:		30
* Gamma smooth timeout:		300
* Dimmer smooth timeout:		ENTER: 30, EXIT: 30
* Gamma long transition:		Disabled

(I)[19:43:36]{main.c:134}	Clightd found, version: 4.0.
(D)[19:43:36]{timer.c:38}	Set timeout of 3s 0ns on fd 11.
(I)[19:43:36]{upower.c:68}	Initial AC state: connected.
(I)[19:43:39]{location.c:103}	14.72 -91.16 loaded from cache file!
(I)[19:43:39]{location.c:84}	New location received: 14.72, -91.16.
(D)[19:43:39]{timer.c:38}	Set timeout of 0s 1ns on fd 31.
(I)[19:43:39]{gamma.c:313}	Normal transition to 4000 gamma temp started.
(I)[19:43:39]{gamma.c:155}	Next alarm due to: Fri Sep 20 05:23:00 2019
(D)[19:43:39]{timer.c:38}	Set timeout of 34761s 0ns on fd 31.
(D)[19:43:39]{my_math.c:61}	AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
(D)[19:43:39]{my_math.c:61}	BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
(I)[19:43:39]{backlight.c:230}	Keyboard backlight calibration enabled.
(D)[19:43:39]{timer.c:38}	Set timeout of 0s 1ns on fd 34.
(D)[19:43:39]{backlight.c:317}	Average frames brightness: 0.425549.
(I)[19:43:39]{backlight.c:251}	Ambient brightness: 0.426 -> Backlight pct: 0.629.
(D)[19:43:39]{timer.c:38}	Set timeout of 2700s 0ns on fd 34.
(I)[19:49:51]{gamma.c:313}	Normal transition to 6000 gamma temp started.
(I)[20:00:17]{upower.c:38}	AC cable connected.
(D)[20:00:17]{timer.c:38}	Set timeout of 0s 1ns on fd 34.
(D)[20:00:18]{backlight.c:317}	Average frames brightness: 0.062745.
(I)[20:00:18]{backlight.c:251}	Ambient brightness: 0.063 -> Backlight pct: 0.083.
(D)[20:00:18]{timer.c:38}	Set timeout of 5400s 0ns on fd 34.
(I)[20:02:07]{upower.c:38}	AC cable disconnected.
(D)[20:02:07]{timer.c:38}	Set timeout of 0s 1ns on fd 34.
(D)[20:02:08]{backlight.c:317}	Average frames brightness: 0.062745.
(I)[20:02:08]{backlight.c:251}	Ambient brightness: 0.063 -> Backlight pct: 0.092.
(D)[20:02:08]{timer.c:38}	Set timeout of 2700s 0ns on fd 34.
(D)[20:04:20]{display.c:35}	Entering dimmed state...
(D)[20:04:20]{display.c:74}	A lower than dimmer_pct backlight level is already set. Avoid changing it.
(D)[20:15:10]{display.c:40}	Entering dpms state...
(D)[20:21:54]{display.c:45}	Leaving dpms state...
(D)[20:21:54]{display.c:50}	Leaving dimmed state...
(I)[20:22:13]{signal.c:50}	Received 2. Leaving.
```

FedeDP, 2019-09-20:
Can you paste another clight log where you start clight and then open up a steam game or just inhibit clight manually from its desktop file?
It seems like org.freedesktop.ScreenSaver interface is indeed starting up; i can't see inhibit_bl module being loaded though; here is a log of it being loaded:
```
(W)[20:11:30]{config.c:32}      Config file /home/federico/.config/clight.conf not found.
(W)[20:11:30]{opts.c:193}       SCREEN forcefully disabled as Clightd was built without screen support.
Clight
* Software version:             4.0
* Global config dir:            /etc/default
* Global data dir:              /usr/share/clight
* Starting time:                Fri Sep 20 20:11:30 2019
Starting options:
### Generic ###
* Verbose (debugging):          Disabled
* Autocalibration:              Enabled
* Inhibit autocalibration:              Disabled
* Keyboard backlight:           Enabled
* Number of captures:           5
* Webcam device:                Unset
* Backlight path:               Unset
* Shutter threshold:            0.00
* Ambient gamma:                Disabled
* Screen contrib:               0.10
* Screen samples:               10
* User position:                Unset
* Daily screen temp:            6500
* Nightly screen temp:          4000
* User set sunrise:             Unset
* User set sunset:              Unset
* Event duration:               1800
* Dimmer backlight pct:         0.20
### Timeouts ###
* Daily timeouts:               AC 600  BATT 1200
* Nightly timeout:              AC 2700 BATT 5400
* Event timeouts:               AC 300  BATT 600
* Dimmer timeouts:              AC 45   BATT 20
* Dpms timeouts:                AC 900  BATT 300
* Screen timeouts:              AC 30   BATT -1
### Modules ###
* Backlight:            Enabled
* Gamma:                Enabled
* Dimmer:               Enabled
* Dpms:         Enabled
* Screen:               Disabled
### Smooth ###
* Bright smooth trans:          Enabled
* Gamma smooth trans:           Enabled
* Dimmer smooth trans:          ENTER: Enabled, EXIT: Enabled
* Bright smooth steps:          0.05
* Gamma smooth steps:           50
* Dimmer smooth steps:          ENTER: 0.05, EXIT: 0.05
* Bright smooth timeout:                30
* Gamma smooth timeout:         300
* Dimmer smooth timeout:                ENTER: 30, EXIT: 30
* Gamma long transition:                Enabled
(I)[20:11:30]{main.c:134}       Clightd found, version: 4.0.
(I)[20:11:30]{main.c:164}       '/home/federico/.local/share/clight/modules.d/inhibit_bl' loaded.
(I)[20:11:30]{main.c:164}       '/home/federico/.local/share/clight/modules.d/daytime' loaded.
(I)[20:11:50]{upower.c:68}      Initial AC state: connected.
(W)[20:11:50]{interface.c:225}  Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
(I)[20:11:50]{location.c:84}    New location received: xxxx, xxx.
(I)[20:11:50]{gamma.c:313}      Normal transition to 4000 gamma temp started.
(I)[20:11:50]{gamma.c:155}      Next alarm due to: Sat Sep 21 06:38:00 2019
(I)[20:11:50]{backlight.c:228}  Keyboard backlight calibration unsupported.
(I)[20:11:50]{daytime.c:37}     We're now during the night!
(I)[20:11:51]{backlight.c:251}  Ambient brightness: 0.390 -> Backlight pct: 0.587.
```

mindrunner, 2019-09-20:
Ok, I was not aware that I need to manually compile this as a module. Is there a plan to put the modules in AUR?

Anyways, if I run steam, I get the following log:
```
New ScreenSaver inhibition held by My SDL application: Playing a game. Cookie: 1804289383
ScreenSaver inhibition enabled.
Being paused.
Being paused.
Dropped ScreenSaver inhibition held by cookie: 1804289383.
ScreenSaver inhibition disabled.
Being resumed.
Being resumed.
New ScreenSaver inhibition held by My SDL application: Playing a game. Cookie: 846930886
ScreenSaver inhibition enabled.
Being paused.
Being paused.
Dropped ScreenSaver inhibition held by cookie: 846930886.
ScreenSaver inhibition disabled.
Being resumed.
Being resumed.
```

However, starting VLC (fullscreen, playing video), mplayer and derivates as well as browser (netflix, youtube) videos, do not seem to inhibit clight.

FedeDP, 2019-09-20:
> Ok, I was not aware that I need to manually compile this as a module.

Yes, see [wiki](https://github.com/FedeDP/Clight/wiki/Custom-Modules#build) :)

> Is there a plan to put the modules in AUR?

I don't think so; I will list them in wiki pages (even user-suggested ones obviously).  

> Anyways, if I run steam, I get the following log:

Mmmh that is weird; if it works for steam it means org.freedesktop.ScreenSaver API is working fine; thus it should work for all apps.

mindrunner, 2019-09-20:
Can you confirm that video players and browser inhibit clight on your box?

FedeDP, 2019-09-23:
Unfortunately i am currently quite busy with real life... i will try and respond back asap.
I am sure i tried with chromium (youtube -> inhibit; gplay music -> no inhibit).  
Moreover, i am pretty sure i tested it with mpv too.

Again, i'll try asap and let you know!

mindrunner, 2019-09-23:
Okay, cool thank you

FedeDP, 2019-09-30:
I did a quick test with vlc and mpv:
* mpv seems to not use org.freedekstop.ScreenSaver inhibition
* vlc calls 
```
‣ Type=method_call  Endian=l  Flags=0  Version=1  Priority=0 Cookie=4
Sender=:1.137  Destination=org.freedesktop.ScreenSaver  Path=/ScreenSaver  Interface=org.freedesktop.ScreenSaver  Member=Inhibit
UniqueName=:1.137
MESSAGE "ss" {
STRING "vlc";
STRING "Riproduzione di alcuni media.";
};
```
Note Path above: **Path=/ScreenSaver** 

Clight expects _/org/freedesktop/ScreenSaver_ as Path. I don't know whether it is a bug in clight or in vlc...
As far as i can tell, correct Path should be the one used by Clight though... eg: chromium and libreoffice use it (libreoffice to inhibit screensaver during presentations).
But i see kde uses both paths in their implementation so i guess i will too.
Thanks for reporting this issue!

If you wish to test this out too with any other app, you should check
`busctl --user monitor org.freedesktop.ScreenSaver`
output when starting a video in your desired app!

FedeDP, 2019-09-30:
Btw it seems firefox too uses /ScreenSaver path:
```
‣ Type=method_call  Endian=l  Flags=0  Version=1  Priority=0 Cookie=51
  Sender=:1.36  Destination=org.freedesktop.ScreenSaver  Path=/ScreenSaver  Interface=org.freedesktop.ScreenSaver  Member=Inhibit
  UniqueName=:1.36
  MESSAGE "ss" {
          STRING "firefox";
          STRING "video-playing";
  };
```

mindrunner, 2019-10-01:
Her my findings:

I fired up `busctl --user monitor org.freedesktop.ScreenSaver` and tried different things.

1) firefox (youtube, netflix) -> see log further down
2) google-chrome (youtube, netflix) -> no reaction
3) mplayer -> no reaction
4) spotify -> no reaction
5) vlc -> no reaction

The only output I ever got:

```
[le@w530]: ~>$ busctl --user monitor org.freedesktop.ScreenSaver
Monitoring bus message stream.
‣ Type=method_call  Endian=l  Flags=0  Version=1  Priority=0 Cookie=3
  Sender=:1.186  Destination=org.freedesktop.ScreenSaver  Path=/ScreenSaver  Interface=org.freedesktop.ScreenSaver  Member=Inhibit
  UniqueName=:1.186
  MESSAGE "ss" {
          STRING "firefox";
          STRING "video-playing";
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1  Priority=0 Cookie=5
  Sender=:1.186  Destination=org.freedesktop.ScreenSaver  Path=/ScreenSaver  Interface=org.freedesktop.ScreenSaver  Member=Inhibit
  UniqueName=:1.186
  MESSAGE "ss" {
          STRING "firefox";
          STRING "audio-playing";
  };
```

FedeDP, 2019-10-02:
Mmmh
1) -> nice
2) -> i tested it myself on chromium and it is working. Which chrome version are you on?
3) Mplayer does not seem to support org.freedesktop.ScreenSaver.Inhibit method.
4) Spotify is an audio player and should not inhibit ScreenSaver, it is fine
5) -> i tested it on vlc 3.0.8 and it is working

mindrunner, 2019-10-02:
1) :)
2) Arch Linux
aur/brave-bin 1:0.68.142-1 (+184 8.98%) (Installed)
extra/chromium 77.0.3865.90-1 (50.9 MiB 177.3 MiB) (Installed)
aur/google-chrome 77.0.3865.90-1 (+1839 12.74%) (Installed)
---> All three not working
3) Yeah, I expeced it to not work
4) You are right, but since firefox sends "audio-playing", I was curioius. So if firefox inhibits with "audio-playing", do you ignore it? Do screensavers ignore it?
5) extra/vlc 3.0.8-1 (10.8 MiB 59.8 MiB) (Installed)

What I am doing is:
1) Open two terminals
2) T1: `busctl --user monitor org.freedesktop.ScreenSaver`
3) T2: `vlc <videofile>`
4)
[image]

NICHOLAS85, 2019-10-03:
The latest commits dc8c6fb9fc066e5912a8f9fa3caae9dbc062f469 and 338125b6d904483ae29d3c9755cd78fa67e2da5c fixed the problems the custom module `inhibit_bl` had with inhibiting during videos in firefox. I have yet to test it in other places but it looks like a good sign!

FedeDP, 2019-10-03:
This is great!

mindrunner, 2019-10-03:
Is it also inhibiting night mode when on video?

FedeDP, 2019-10-04:
What do you mean? Setting DAY screen temperature?
Right now that custom module does not do that, but it is pretty straightforward to achieve it.

mindrunner, 2019-10-04:
Yes, exactly. I do not think, it makes any sense to watch movies with night color temp. Doesn't it make more sense to have the application working in a logical way? My impression is just, that this should be part of the functionality. I doubt, there is many users out there who wants to develop a module before. 
Or why do we inhibit dimming by default, but not the color temp?

mindrunner, 2019-10-04:
Hi, I am trying to implement a module which sets 'day'-temp when inhibited and auto-mode when not. However, I am not fully understanding what I am supposed to do?

As far as I can see, I can set/change the screen temp with TEMP_REQ. But that is not what I want.  I am looking for something like NO_AUTOCALIB_REQ for the gamma. Or how would you implement it?

mindrunner, 2019-10-04:
Testing the dimming inhibit module from wiki right now. Compiling it and copied it to my modules folder. 

Like already discussed, vlc,mplayer,chrome, etc do not trigger anything here. Firefox produces following log 
```
Emitting KbdPct property
New ScreenSaver inhibition held by firefox: video-playing. Cookie: 1804289383
ScreenSaver inhibition enabled.
Being paused.
Emitting Inhibited property
Being paused.
Dropped ScreenSaver inhibition held by cookie: 1804289383.
ScreenSaver inhibition disabled.
Being resumed.
Emitting Inhibited property
Being resumed.
```

However, it remains in a dimmed state. That does not make sense to me. I think we want 100% brightness while watching a video, right?

NICHOLAS85, 2019-10-05:
@mindrunner Did you make sure to rebuild clight from git before testing the inhibit_bl module? It does set the brightness to 100% for me when I've tested it. I've also looked at the day time gamma setting and am also having trouble figuring out how to revert it back to what it was before inhibition. I've figured a way to simply inhibit redshift if a user is running that.

https://pastebin.com/QixzRMyd

mindrunner, 2019-10-05:
Hi @NICHOLAS85 
I had the compiled module in the wrong folder, so it was not being load. This seems good now and my backlight is set to 100% when starting vlc or firefox video.

I tried this for the gamma thing:
```
#include <clight/public.h>
#include <stdlib.h>

CLIGHT_MODULE("INHIBIT_GAMMA");

DECLARE_MSG(temp_req, TEMP_REQ);

static void init(void) {
    /* Subscribe to inhibit state */
    M_SUB(INHIBIT_UPD);
}

static void receive(const msg_t *msg, const void *userdata) {
    switch (MSG_TYPE()) {
    case INHIBIT_UPD: {
        inhibit_upd *up = (inhibit_upd *)MSG_DATA();
        if (up->new) {
	    system("busctl --user set-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf NightTemp \"i\" 6500");
            INFO("We are now inhibited! Set daylight screen temperature.\n");
        } else {
	    system("busctl --user set-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf NightTemp \"i\" 4500");
            INFO("We're not inhibited anymore. Set auto gama.\n");
        }
        break;
    }
    default:
        break;
    }
}

```
But getting this error:
```
Failed to set property NightTemp on interface org.clight.clight.Conf: Connection timed out
```

It's not the right way to do it anyways. @FedeDP can you point us to the right direction :)

FedeDP, 2019-10-08:
> Got it working, but for now, the numbers are hardcoded. Can I get the user's values for day/night anywhere?

Yes, you can subscribe to TEMP_UPD to receive current user-setted temperature, eg: 6500 during the DAY; then you can store it to a global variable; moreover, you can set daytime = -1 in your request, this way it will use current daytime.
Eg (NOT TESTED!):
```
CLIGHT_MODULE("INHIBIT_GAMMA");

DECLARE_MSG(temp_req, TEMP_REQ);

static int real_temp = 6500; // default value

static void init(void) {
    temp_req.temp.daytime = -1; // use current daytime

    /* Subscribe to inhibit state and temp updates */
    M_SUB(INHIBIT_UPD);
    M_SUB(TEMP_UPD);
}

static void receive(const msg_t *msg, const void *userdata) {
    switch (MSG_TYPE()) {
    case INHIBIT_UPD: {
        inhibit_upd *up = (inhibit_upd *)MSG_DATA();
        if (up->new) {
            INFO("We are now inhibited! Set daylight screen temperature.\n");
	    temp_req.temp.new = 6500;
            M_PUB(&temp_req);
        } else {
            INFO("We're not inhibited anymore. Set auto gama.\n");

	    temp_req.temp.new = real_temp;
            M_PUB(&temp_req);
        }
        break;
    }
    case TEMP_UPD: {
        temp_upd *up = (temp_upd *)MSG_DATA();
        real_temp = up->new;
        break;
    }
    default:
        break;
    }
}
```

Obviously you can merge this custom module with the inhibit_bl one.

And please, if you don't mind, share your feedback with custom modules, ie: if you find them hard or whatever :)
I hope they're simple enough!

Thanks and sorry for the late feedback!

FedeDP, 2019-10-08:
>  I will think of a better approach to inhibit modules, eg: a request with a list of modules to be inhibited.

On a second thought, i actually think that inhibiting GAMMA is not really good for your eyes; i don't understand the reason behind that!
I can see inhibiting DIMMING and DPMS (obviously...otherwise your screen would switch off/dim during a movie) useful; inhibiting BACKLIGHT already seems quite exaggerated to me, but i can understand that one would wish to set maximum backlight and leave it untouched until the end of a movie.
But i cannot see any real use case for GAMMA inhibition. I think you should "fallback" to a custom module for that functionality; the one provided above should work fine.

FedeDP, 2019-10-08:
Closing this as requested feature (BACKLIGHT module support to Inhibit) has been satisfied.
If you have any more issues with org.freedesktop.Inhibit interface, let me know in #76 .

closed 2019-10-08 by FedeDP