FedeDP/Clight #82: Display not found when using external monitor

SevereOverfl0w, 2019-06-05:
**Clight version the issue has been seen with:**
> 3.1

**Used distribution:**
> Void Linux

**Describe the bug**
When using an external monitor (only, laptop monitor is `xrandr --off`) when I run `clight`, `clightd` starts outputting `Display not found` and nothing happens.

To expediate things, I looked at #35, and ran this diagnostic command:

```
❯ sudo ddcutil detect          
Invalid display
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      Mfg id:           SHP
      Model:            Unspecified
      Serial number:    Unspecified
      Manufacture year: 2015
      EDID version:     1.4
   DDC communication failed

Display 1
   I2C bus:             /dev/i2c-4
   EDID synopsis:
      Mfg id:           AUS
      Model:            ASUS VP28U
      Serial number:    Unspecified
      Manufacture year: 2018
      EDID version:     1.3
   VCP version:         2.2
```

**Expected behavior**
Screen dims like when not connected.

**To Reproduce**
I'm using a dell xps 8360, and an ASUS VP28U external monitor over a Type-C HDMI adapter.

Please attach a Clight verbose log too!

```
❯ clight --verbose
SIGNAL module started.
BUS module started.
USERBUS module being started as submodule of BUS...
USERBUS module started.
INTERFACE module being started as submodule of USERBUS...
INTERFACE module started.
Trying to start BACKLIGHT module as its INTERFACE dependency was loaded...
Trying to start GAMMA module as its INTERFACE dependency was loaded...
Trying to start DIMMER module as its INTERFACE dependency was loaded...
XORG module started.
Trying to start GAMMA module as its XORG dependency was loaded...
Trying to start DIMMER module as its XORG dependency was loaded...
Trying to start DPMS module as its XORG dependency was loaded...
Trying to start LOCATION module as its BUS dependency was loaded...
geoclue_get_client(): The name org.freedesktop.GeoClue2 was not provided by any .service files
Geoclue2 appears to be unsupported.
Loading loc from cache file: No such file or directory
Error while loading LOCATION module.
LOCATION module disabled.
Disabling module GAMMA as its hard dep LOCATION was disabled...
GAMMA module disabled.
Trying to start BACKLIGHT module as its GAMMA soft dep was disabled...
GAMMA module destroyed.
LOCATION module destroyed.
Trying to start UPOWER module as its BUS dependency was loaded...
UPOWER module started.
Trying to start BACKLIGHT module as its UPOWER dependency was loaded...
Trying to start DIMMER module as its UPOWER dependency was loaded...
Trying to start DPMS module as its UPOWER dependency was loaded...
Trying to start CLIGHTD module as its BUS dependency was loaded...
Clightd found, version: 3.4.
CLIGHTD module started.
Trying to start BACKLIGHT module as its CLIGHTD dependency was loaded...
AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
Set timeout of 0s 1ns on fd 7.
BACKLIGHT module started.
Callback added for module BACKLIGHT on module UPOWER bus match.
Callback added for module BACKLIGHT on module INTERFACE bus match.
Callback added for module BACKLIGHT on module INTERFACE bus match.
Callback added for module BACKLIGHT on module INTERFACE bus match.
Trying to start DIMMER module as its CLIGHTD dependency was loaded...
Trying to start DPMS module as its CLIGHTD dependency was loaded...
Setted DPMS timeouts: Standby -> 900s, Suspend -> 1200s, Off -> 1800s.
DPMS module started.
Callback added for module DPMS on module UPOWER bus match.
Callback added for module DPMS on module INTERFACE bus match.
Trying to start INHIBIT module as its USERBUS dependency was loaded...
inhibit_check(): The name org.freedesktop.PowerManagement.Inhibit was not provided by any .service files
PowerManagement inhibition appears to be unsupported.
INHIBIT module started.
Trying to start DIMMER module as its INHIBIT dependency was loaded...
Average frames brightness: 0.045421.
Ambient brightness: 0.045 -> Backlight pct: 0.060
Set timeout of 600s 0ns on fd 7.
Trying to start DIMMER module as its BACKLIGHT dependency was loaded...
idle_get_client(): Unknown object '/org/clightd/clightd/Idle'.
Clight Idle error.
Error while loading DIMMER module.
DIMMER module disabled.
INHIBIT module disabled.
INHIBIT module destroyed.
DIMMER module destroyed.
^Creceived signal 2. Leaving.
DPMS module destroyed.
BACKLIGHT module destroyed.
CLIGHTD module destroyed.
UPOWER module destroyed.
XORG module destroyed.
INTERFACE module destroyed.
USERBUS module destroyed.
BUS module destroyed.
SIGNAL module destroyed.
```

FedeDP, 2019-06-06:
Hi, thanks for your feedback.
Can you update both Clight and Clightd from -git master? 
May be update ddcutil too :)

Thanks!

SevereOverfl0w, 2019-06-07:
Updating all 3 from git master has not fixed it.

FedeDP, 2019-06-07:
Very unfortunate.
Is there any other log for Clightd? Can you post here a clightd log while running clight?
To get clightd log you should use:
`journalctl -b -u clightd.service`

SevereOverfl0w, 2019-06-07:
I'm not using systemd, so I started clightd manually in order to get the logs from stdout:

```
❯ sudo /usr/lib/clightd/clightd
Registered Als sensor.
Registered Camera sensor.
[default]|IDLE|: Creating client 0
[default]|IDLE|: Adding inotify watch as first client was started.
[default]|IDLE|: Starting Client 0
Using YUYV pixelformat.
V4L2_CID_SCENE_MODE unsupported
V4L2_CID_AUTO_WHITE_BALANCE unsupported
V4L2_CID_EXPOSURE_AUTO unsupported
V4L2_CID_AUTOGAIN unsupported
V4L2_CID_ISO_SENSITIVITY_AUTO unsupported
V4L2_CID_AUTOBRIGHTNESS unsupported
V4L2_CID_IRIS_ABSOLUTE unsupported
V4L2_CID_ISO_SENSITIVITY unsupported
[default]|SENSOR|: 10 frames captured by /dev/video0.
[default]|BACKLIGHT|: Target pct (smooth 1): 0.85
[default]|BACKLIGHT|: Reached target backlight: 0.85.
Display not found
[default]|IDLE|: Creating client 1
[default]|IDLE|: Starting Client 1
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
Display not found
^C[default]|SIGNAL|: Received signal 2. Leaving.
```

FedeDP, 2019-06-09:
> Display not found
> Display not found

Are all these "Display not found" logs consecutive (ie: very near in time)?

> When using an external monitor (only, laptop monitor is xrandr --off) when I run clight, clightd starts outputting Display not found and nothing happens.

When you use both internal and external monitors, does clightd work fine?
What if you use internal alone?
Thanks!

FedeDP, 2019-06-09:
Ops may be i've seen your issue; the log comes from [here](https://github.com/rockowitz/ddcutil/blob/83f9174005cef2bb6edcc868ddacd93ef668ab3a/src/ddc/ddc_displays.c#L846).
The issue is caused by:
>  Serial number:    Unspecified

For your monitor.
Unfortunately that field is mandatory right now for Clightd because it is used to distinguish each external monitor (as an unique name).

I'll check if another field can be used (eg: i2c bus) and answer back here.