FedeDP/Clight #83: SELinux policy package

tminor, 2019-06-09:
First, thanks for Clight! It's been great so far!

## Is your feature request related to a problem?
Maybe? It's been my habit to always set SELinux to permissive mode as soon as it disallows something that's otherwise properly configured. This time I resolved to make SELinux happy!

## Describe the solution you'd like
To facilitate SELinux's happiness, I think it would be nice to provide an SELinux type enforcement file in the wiki or in the repo. I'm not an SELinux expert, but here's an example type enforcement file that might work without further tweaking:

```

module clightd 1.0;

require {
        type init_t;
        type xserver_t;
        type xauth_home_t;
        type v4l_device_t;
        class chr_file { ioctl map open read write };
        class unix_stream_socket connectto;
        class file { getattr open read };
}

#============= init_t ==============

allow init_t v4l_device_t:chr_file { ioctl map open read write };
allow init_t xauth_home_t:file getattr;

allow init_t xauth_home_t:file { open read };

allow init_t xserver_t:unix_stream_socket connectto;
```

My example is a result of flailing until Clight worked, so it probably needs work :)

## Describe alternatives you've considered
The obvious solution is just configuring SELinux to run in permissive mode and call it a day :)

FedeDP, 2019-06-09:
Wow, nice thanks! I will surely add this to the wiki pages!
I think this should go in Clightd though :)

Thank you very much, happy to hear you like clight!

tminor, 2019-06-10:
Yes, you're right! An example AVC message makes this obvious, but reading is hard :)

```
type=AVC msg=audit(1560171357.260:245): avc:  denied  { read } for  pid=8460 comm="clightd" name="hiddev1" dev="devtmpfs" ino=2840 scontext=system_u:system_r:init_t:s0 tcontext=system_u:object_r:usb_device_t:s0 tclass=chr_file permissive=0
```

I'll go ahead and close this. Thanks again for Clight!

closed 2019-06-10 by tminor

FedeDP, 2019-06-10:
Thanks :)
I'll add it to the clightd wiki asap!